subsurface/subsurface #565: Clicking "Log > Add dive" causes window to freeze

Alexander-Wilms, 2017-09-10:
This happens with a Flatpak build of the master branch using this manifest: https://github.com/Alexander-Wilms/flathub/blob/org.subsurface_divelog.Subsurface/org.subsurface_divelog.Subsurface.json

[gdb-subsurface.txt](https://github.com/Subsurface-divelog/subsurface/files/1290844/gdb-subsurface.txt)

neolit123, 2017-09-10:
thanks for the report. i've notified the mailing list about this.

closed 2017-09-16 by dirkhh

Alexander-Wilms, 2017-09-16:
Thanks! It seems like that commit broke the build, though: https://travis-ci.org/Subsurface-divelog/subsurface/builds/276276872?utm_source=github_status&utm_medium=notification

dirkhh, 2017-09-16:
Yes and no. There is one test (TestParse) that for odd reasons doesn't appear to be 100% stable.
The build is fine, but the test failed once again with a spurious po2 attribute

dirkhh, 2017-09-16:
Restarted the travis job and it now passes

PR #581 (2017-09-16): Don't comput plan variations when not in the planner
Fixes #565
commit 58b7c6f8 (2017-09-16): Don't comput plan variations when not in the planner

Fixes #565