subsurface/subsurface #566: Program crashes trying to import dive logs

scubadoo2, 2017-09-11:
Program crashes trying to import any/all of my ZXU log files from Aeris Atmos 2 computer #1.  Does not crash importing ZXU logs files from Aeris Atmos 2 computer #2.  All exported from Aeris Computer Interface 2.x program.
[DAVE_1.ATMOS 2..ZXU.zip](https://github.com/Subsurface-divelog/subsurface/files/1291222/DAVE_1.ATMOS.2.ZXU.zip)

scubadoo2, 2017-09-11:
Version 4.6.4 on Windows 7 x64

scubadoo2, 2017-09-11:
OK, I think I sort of figured it out.  The first 17 dives were actually manually entered into the Aeris program for dives before I got the computer, not downloaded from the computer.  There was only basic information entered, no dive profile data.

mturkia, 2017-09-11:
Yeah, we crash when the input file does not contain any profile data. Easy fix.

commit f3ab2ee6 (2017-09-11): Fix crash on manually added ZXU/DL7 file

The CSV import crashed if the input file had no samples in it.

Fixes #566
closed 2017-09-11 by dirkhh