subsurface/subsurface #567: Download gps fix resu

nqdrizzt, 2017-09-12:


dirkhh, 2017-09-13:
That's odd. So just to make sure I understand - you are running Subsurface-mobile on both devices? They are both on Android? They are both connected to the same account, but during the upload to the GPS/Webservice the date part of the time stamp gets lost?

nqdrizzt, 2017-09-14:
Yes, that's absolutely correct. I also realized that there are error messages in the log returned by the server during uploading the GPS fixes via the API.

dirkhh, 2017-09-14:
INTERNAL SERVER ERROR - darn, I wish we printed a better error message...
The backend is one of the few pieces of our infrastructure that I am not all that comfortable with (the running joke is that it's in Python and Python and I don't get along). I'll take a look to try and figure out what's going on there because the data that you are sending seems legit at first glance...
I assume that you can reproduce the issue, so if you try again, you get the same result?

nqdrizzt, 2017-09-14:
Yes I can confirm that it's reproduceable any time. Du u need any further loggings?

dirkhh, 2017-09-14:
OK, so the errors are completely misleading - they are most likely just "duplicate entries". What I still don't understand is why for some of the GPS fixes the date is wrong.
When I look at the backend database there are indeed entries with 1970 as data (and others with correct looking dates). And for many fixes that I've checked, there were identical entries both for 1970 and 2017 with matching dive time, lat and long.
I literally have no idea what's going on there.
One thing I notice, though. For example the last entry in your screen shot. 1:24pm, -8.36032/116.087 there is no such matching entry - just one from 1970. Yet the log that you attached shows no attempt to send such a fix.
Any idea why that might be?

nqdrizzt, 2017-09-14:
The log I attached was done at a later point of time as the screenshot. In between I manually tried to explicitly delete all entries with a timestamp of 1970 via the app and also cleared the GPS fix cache on the device. Probably the deletion of that fixes also wiped them from database and at the time of the log file creation they were already gone.  About The duplicate entries (one from 1970 and 2017)I don't have any clue . Maybe  because I mapped the dives manually to a location before uploading them to the backend? Some of the dives I also edited in the windows version of subsurface. But these have to be March 2017 or prior  (not that ones from this week)

nqdrizzt, 2017-09-14:
By the way: if I download all the fixes to the other device, I can't see any entry with a date from 2017,all have a timestamp from 1970

PR #983 (2017-12-25): Use correct date format
### Describe the pull request:

- [x] Bug fix
- [ ] Functional change
- [ ] New feature
- [ ] Code cleanup
- [ ] Build system change
- [ ] Documentation change
- [ ] Language translation

### Pull request long description:
 Use correct date format

A very very trivial fix, for a mysterious issue. When loading GPS fix data from the server, the string date was parsed with the format "yyy-M-d". And no, the "yyy" is no typo here, but was the reason that data from the read from server got a 1/1/1970 data. And when a user decided to upload that data to the server again, we en [... cut]
commit 37e98a8d (2017-12-25): Use correct date format

A very very trival fix, for a mysterious issue. When loading
GPS fix data from the server, the string date was parsed with
the format "yyy-M-d". And no, the "yyy" is no typo here, but
was the reason that data from the read from server got a
1/1/1970 data. And when a user decided to upload that data
to the server again, we ended up with 2 copies of the
GPS fix. One with correct data (as originally saved), and
one new with the bogus date.

In order to het rid of those weird 1/1/1970 GPS fixes, users
will have to remove them by hand.

Fixes: #567
closed 2017-12-25 by janmulder