subsurface/subsurface #569: Importing from Shearwater Desktop always chooses CCR instead of OC

SeppoTakalo, 2017-09-12:
When importing from Shearwater Desktop the Subsurface seems to assume dive mode `CCR` instead of `OC`
[image]
The Shearwater seems to be showing `Dive type: OC`
[image]

Please see the attached dive log file exported from Shearwater.
[new_dives.db.zip](https://github.com/Subsurface-divelog/subsurface/files/1296981/new_dives.db.zip)

mturkia, 2017-09-12:
I wish I had a proper CCR dive log to see what is different in that when compared to OC. The current detection for CCR mode apparently is not the way to implement this. Is there any way to set CCR mode in Shearwater Desktop or is it displaying the mode the DC was during the dive?

SeppoTakalo, 2017-09-12:
I believe it is showing the value that the computer was set during the dive.
I have not set any mode within the desktop application itself.

closed 2017-09-13 by dirkhh