subsurface/subsurface #574: `atof()` used for converting values from databases without checking locales.

SeppoTakalo, 2017-09-13:
It seems that `atof()` function is used to convert floating point values coming from SQLite database.
This is a problematic because `atof()` cares about users current locale settings but SQLite does not seem to.

This causes problems where decimal separator is `,` instead of `.` when SQLite seems to be always using `.`
End result would be that all numbers converted have only their integer part.

```
~/src/subsurface/subsurface $ ag "atof\(" -l --stats
core/parse-xml.c
26 matches
1 files contained matches
```

`atof_l()` could have been used with `NULL` locale, but its not portable. I don't know any portable alternatives.

SeppoTakalo, 2017-09-13:
I wonder if it too much wrong to call `setlocale(LC_NUMERIC, "C")` before starting the conversion.

mturkia, 2017-09-13:
Smarttrak import does the following:

        // Set an european style locale to work date/time conversion
        setlocale(LC_TIME, "POSIX");

On the other hand, we have core/strtod.c due to locale reasons.

PR #580 (2017-09-15): Fix two issues from Shearwater import
* Convert all usages of `atof()` to use internal `strtod_flags()` with relaxed flags. This ignores user locales when importing data. Fixes #574 
* Use dive ID instead of log number when importing from Shearwater Desktop databases. Same number can appear many times within the same database but dive ids are unique. This was mentioned in https://github.com/Subsurface-divelog/subsurface/pull/573#issuecomment-329196154
commit f7db0c70 (2017-09-15): Convert usage of atof() to strtod_flags()

The provided strod_flags(str, 0, 0) should work as a drop in replacement
for atof() but does not care about locales which may cause atof() to fail.

strtod_flags() would allow checking of conversion result, but I did not
change the existing logic. This was just regexp search&replace change
to get rid of atof(). I use flags 0 to get more relaxed conversion.

Fixes #574
closed 2017-09-15 by dirkhh