subsurface/subsurface #584: Small spike at beginning of VPM-B based ceiling in profile

sfuchs79, 2017-09-16:
I would like to highlight a small issue again which I already discussed some time ago with @atdotde because I made an interesting new finding around it:
The issue is that for any (!?) VPM-B based ceiling displayed in the profile by Subsurface I have a small spike in this ceiling at the very beginning. Please see the image for clarification.

Now I found out that I can reproduce a situation where for the very same dive and the very same divelog the spike is there and not there. What makes the difference is the seuquence in which I select the dives. I also attached the example logfile.
If I open this file at the first moment dive 711 is displayed. This dive has no deco ceiling at all.
If I select dive 710 now for the first time the spike is *not* there.
If I now select dive 709 this dive shows the spike and if I now go back to 710 the spike is also there.

Maybe this helps to track down the issue.

Two additional things:
- Ceiling setting is VPM-B with "+2"
- Issue seen on my side under Linux and Windows

[Spike_Test.zip](https://github.com/Subsurface-divelog/subsurface/files/1308657/Spike_Test.zip)

[image]

atdotde, 2017-09-17:
Hi,

> On 16. Sep 2017, at 23:04, Stefan Fuchs <notifications@github.com> wrote:
> 
> I would like to highlight a small issue again which I already discussed some time ago with @atdotde <https://github.com/atdotde> because I made an interesting new finding around it:
> The issue is that for any (!?) VPM-B based ceiling displayed in the profile by Subsurface I have a small spike in this ceiling at the very beginning. Please see the image for clarification.
> 
> Now I found out that I can reproduce a situation where for the very same dive and the very same divelog the spike is there and not there. What makes the difference is the seuquence in which I select the dives. I also attached the example logfile.
> If I open this file at the first moment dive 711 is displayed. This dive has no deco ceiling at all.
> If I select dive 710 now for the first time the spike is not there.
> If I now select dive 709 this dive shows the spike and if I now go back to 710 the spike is also there.
> 
> Maybe this helps to track down the issue.
> 
> 

I am afraid, I cannot reproduce this on my Mac, neither with current master nor 4.6.4. But I remember having seen the spike in the past. All this sounds like some of VPN’s state variables is not properly initialized. Candidates are initial_XX_gradient and bottom_XX_gradient but max_XX_crushing_pressure could also be involved. So, to debug I would print those values around the time that the spike appears and look for differences. Could you try this?

Best
Robert

sfuchs79, 2017-09-17:
Hmmm... I have something: I added the debug output of initial_XX_gradient and bottom_XX_gradient in tissue_tolerance_calc();
I generated two files, one which shows the values calculated when there is no spike in the result and one where there is the spike. This is for sure done again for the very same dive. These are the files:
[710_with_spike.txt](https://github.com/Subsurface-divelog/subsurface/files/1309176/710_with_spike.txt)
[710_wo_spike.txt](https://github.com/Subsurface-divelog/subsurface/files/1309177/710_wo_spike.txt)

I see that already the very first call of tissue_tolerance_calc(); shows different values. But only the very first. Then there is a long sequence of iterations with identical values and then there is a sequence of different values again exactly at the position when I see the spike in the profile.

What can be the next step?

sfuchs79, 2017-09-24:
I can confirm that the spike at the beginning of the deco ceiling in the profile is fixed by #599 
Many thanks to @RickMWalsh and @atdotde !! :-)

So I close this because I think there is no doubt that #599 can be merged into master.

commit 8b5c5674 (2017-10-01): VPM-B profile: initialize first_ceiling_pressure

If we don't set first_ceiling_pressure at start of dive, a shallow ceiling can
be shown when it shouldn't be.

Fixes #584
closed 2017-09-24 by sfuchs79