subsurface/subsurface #588: Googlemaps not working when build from source in Mac OS X

SeppoTakalo, 2017-09-20:
Using Qt 5.9.1.
I cloned the master repository and followed instructions in [INSTALL](https://github.com/Subsurface-divelog/subsurface/blob/master/INSTALL)

Built using the `./subsurface/script/build.sh` script which builds succesfully but I noticed following warning:
```
cp: cannot stat '/Users/st/Qt/5.9.1/clang_64/plugins/geoservices/libqtgeoservices_googlemaps.dylib': No such file or directory
```

Then when starting the application:
```
st@st-laptop:~/src/subsurface $ ./subsurface/build/Subsurface.app/Contents/MacOS/Subsurface 
qml: MapWidget.qml: cannot find a plugin named: googlemaps
qrc:/MapWidget.qml:21: Error: Cannot assign [undefined] to QDeclarativeGeoMapType*
MapWidget ready
```

And in the UI, the map widget shows miniature map icon on top of white background.
[image]

dirkhh, 2017-09-20:
Did you happen to save a log of the build? It should have cloned the sources, built and installed that googlemaps plugin...

SeppoTakalo, 2017-09-20:
Install-root folder seem to have directory tree replicated that probably should have been targeted to `/`

```
st@st-laptop:~/src/subsurface $ ls -R install-root/Users/
install-root/Users/:
st

install-root/Users/st:
Qt

install-root/Users/st/Qt:
5.9.1

install-root/Users/st/Qt/5.9.1:
clang_64

install-root/Users/st/Qt/5.9.1/clang_64:
lib  plugins

install-root/Users/st/Qt/5.9.1/clang_64/lib:
cmake

install-root/Users/st/Qt/5.9.1/clang_64/lib/cmake:
Qt5Location

install-root/Users/st/Qt/5.9.1/clang_64/lib/cmake/Qt5Location:
Qt5Location_QGeoServiceProviderFactoryGooglemaps.cmake

install-root/Users/st/Qt/5.9.1/clang_64/plugins:
geoservices

install-root/Users/st/Qt/5.9.1/clang_64/plugins/geoservices:
libqtgeoservices_googlemaps.dylib       libqtgeoservices_googlemaps_debug.dylib
...
```

`/Users/st` is my home dir.

SeppoTakalo, 2017-09-20:
I fixed the problem manually by going into this `src/googlemaps/build` directory and called `make install`

Then I restarted the build script and everything seems to work OK. Also it fixed the freezing issue #589

neolit123, 2017-09-23:
/close

dirkhh, 2017-09-23:
I still want to fix the actual build issue - it's good that you fixed it for yourself, but the script should try to get this right.

mfbernardes, 2018-07-21:
Seems fixed by 0620c81594

closed 2020-10-03 by bstoeger