subsurface/subsurface #589: UI hangs up when build from source on Mac OS X using Qt 5.9.1

SeppoTakalo, 2017-09-20:
Using Qt 5.9.1.
I cloned the master repository and followed instructions in [INSTALL](https://github.com/Subsurface-divelog/subsurface/blob/master/INSTALL)

Built using the `./subsurface/script/build.sh` script which builds successfully. Only issue seems to be Googlemaps described in #588

**Problem:**
* When opening the UI it looks correct.
* Click the dive list view
* Click arrow key down for a few dive for view to switch dive.
* Usually freezes before 4 dives.

Does not happen on the stable 4.6.4 release when installed from the .DMG

neolit123, 2017-09-23:
/close

dirkhh, 2017-09-23:
While I'd argue that it shouldn't HANG without the Googlemaps plugin, I agree with closing it.

closed 2017-09-23 by dirkhh

neolit123, 2017-09-23:
i will create a new issue for this.
not 100% sure how well QML would let me to not crash, though.

dirkhh, 2017-09-23:
On Sat, Sep 23, 2017 at 10:23:28AM -0700, Lubomir I. Ivanov wrote:
> i will create a new issue for this.
> not 100% sure how well QML would let me to not crash, though.

I think the trick will be to simply not show the widget if we
can't find the plugin. Is that possible? Or do we notice "too
late"?

neolit123, 2017-09-23:
it might be possible to completely hide the widget, but it also provides useful feedback from users if the map plugin isn't loading - i.e. we always want the map plugin to load.

ATM the crashes are happening due to QMetaObject::invokeMethod() calls, i think.