subsurface/subsurface #590: OSTC4 download dives fails with "Error parsing the header"

antiker1, 2017-09-20:
Hi.

Trying to download the profiles from some trimix dives gives: "`Error parsing the header`" in the UI and
```
subsurface:~$ subsurface
[3.109370] ERROR: Invalid gas mix. [in hw_ostc_parser.c:843 (hw_ostc_parser_samples_foreach)]
```

Any idea for a quick fix, as this blocks downloading all further new dives and I cannot write my logs now ;-)

Thanks!

antiker1, 2017-09-20:
Maybe it's because of this

[image]

Not sure how this happened... something with the gas change didn't work as it should (I was anyway diving after RT), and the ppO2 seems screwed up.

neolit123, 2017-09-23:
@mturkia, @torvalds, @dirkhh 

hi, @antiker1. like you've said it looks like the PP02 is wrong on the dive computer itself.

questions:
1) does it work for a normal dive (e.g. without the glitchy PPO2)?
2) can you can download the dive files in some other way and send them here for a review?

antiker1, 2017-09-23:
Hey @neolit123

Thanks for looking into this :-)

1. Yes it worked just perfect for "normal" dives.
2. Given #561 is unfortunately set to not going to be resolved and thus I was actually just looking into some alternative dive log software which doesn't modify the data read out :-/
So far I found only some Windows (and I have no Windows machine here on holidays) programs which can read out the OSTC 4. I'll keep looking and if I find some I'll report back.

I'll do another deep one next Monday down to >100m and that should give me a number of TMX/EAN gas changes. So I'll see whether that PPO2 thingy happens again then and whether I can read out that dive.

Thanks.

mturkia, 2017-09-24:
Can you save subsurface logfile and dumpfile on the download dialog and send them for testing purposes? It looks like the gaschange index is more than there are available mixes, and thus the parsing is aborted.

antiker1, 2017-09-24:
With log and dump files enabled when reading out:
[log-dump_subsurface.log](https://github.com/Subsurface-divelog/subsurface/files/1327558/log-dump_subsurface.log)
Gives a "Dive data import error" and no dumpfile is created.

With log file only enabled:
[log-only_subsurface.log](https://github.com/Subsurface-divelog/subsurface/files/1327560/log-only_subsurface.log)

Dump file only enabled, gives a "Dive data import error" and no dumpfile is created.

Thanks!

neolit123, 2017-09-24:
seems to throw a:
```ERROR: Invalid gas mix. [in hw_ostc_parser.c:872 (hw_ostc_parser_samples_foreach)]```

@antiker1  would be still great if we can obtain the original dive file from the DC, somehow.

antiker1, 2017-09-24:
Thats with Diving Log 6.0,... but it also seems to download only part of the dive and then abort (without error):
[Logbook.mdb.zip](https://github.com/Subsurface-divelog/subsurface/files/1327823/Logbook.mdb.zip)

glance-, 2017-09-25:
There is no dump functionallity available for the OSTC4. The way we do memory dumps only works for the ostc3-family.

You can dump each dive as a binary with dctool. Anyway, all the data are already there in the log file.

Its quite easy to hack around the bad data comming from the OSTC4. It might be a bug in libdivecomputer or it might be a bug in the OSTC4. I haven't looked at the actual data.

antiker1, 2017-09-25:
It may very well be a bug in OSTC4 (actually quite likely already, because the ppO2 it shows on the pic is just bogus)...but during that dive, the deco stops were also completely wrong...

neolit123, 2017-10-11:
closing this issue as it seems that the bug is not in subsurface.

antiker1, 2018-08-20:
@neolit123 Could you please reopen this? The bug still persists and I cannot read out my dive computer anymore because of it.

Where else should it be than subsurface? The computer itself works fine.

neolit123, 2018-08-20:
> Where else should it be than subsurface? The computer itself works fine.

from one of the previous messages:
> Thats with Diving Log 6.0,... but it also seems to download only part of the dive and then abort (without error):

both Subsurface and Dive Log use libdivecomputer as the library for talking to DCs.
so technically the `bug-not-in-subsurface` is valid here. like Anton said:
> It might be a bug in libdivecomputer or it might be a bug in the OSTC4

i can re-open but a developer with a OSTC4 needs to reproduce this and fix it.
until then we have no solutions.

glance-, 2018-08-20:
You still haven't provided the raw data from the dc. Please dump out the raw dive log from the dc.

If you want to, its quite simple to hack libdivecomputer to not care about the corrupt data, and just skip it, so you can get the rest of the data.

I've done that more than once to handle in dc corruptions, and firmware bugs.
(Once I even corrected the logbook data in the dc, but that's overkill.)

antiker1, 2018-09-02:
@glance- How can I get the raw data? Or is it what I tried in the comment above (https://github.com/Subsurface-divelog/subsurface/issues/590#issuecomment-331729435)?

How would I hack libdivecomputer to not care about the corrupted parts?