subsurface/subsurface #591: Progress on the Flatpak packaging

Alexander-Wilms, 2017-09-20:
I've added QtWebKit and was able to log into my Facebook account. For bluetooth support, I need to build bluez, which requires some changes to the udev module: https://github.com/flathub/shared-modules/issues/8

```
checking for UDEV... yes
checking for udev_hwdb_new in -ludev... no
checking udev directory... Variable 'exec_prefix' not defined in '/app/share/pkgconfig/udev.pc'
configure: error: udev directory is required
Error: module bluez: Child process exited with code 1
```

I'm now using regular modules in the manifest instead of running the build.sh script, which makes it more readable.

neolit123, 2017-09-23:
@dirkhh @mturkia

dirkhh, 2017-09-23:
Flatpak is something I'm interested in but haven't had time to look into (given all the other pressing issues. @Alexander-Wilms - are you still making progress, is there something you need from us?

Alexander-Wilms, 2017-09-23:
I've fixed that issue and added bluez. Can you try connecting to a divecomputer vis bluetooth?

dirkhh, 2017-09-24:
I ran flatpak update and got new versions of Qt 5.9 and Subsurface.
When I run the resulting binary, it tells me that Bluetooth is not available at all.
On the console it states "qt.bluetooth: Dummy backend running. Qt Bluetooth module is non-functional"
You can test this yourself, even without having a dive computer.
Run Subsurface
Ctrl-D for Download from dive computer
On the left side of the dialog box at the bottom check the box next to "Choose Bluetooth download mode"
Another dialog will open - that should show a working Local Bluetooth device and you should be able to scan for Bluetooth devices.

Alexander-Wilms, 2017-09-25:
The qt-connectivity module didn't install its files into the /app prefix. Copying the bin, include and lib dirs to /app fixed that error message, but now I got another one:

`Cannot find a running Bluez. Please check the Bluez installation.`

There aren't any other Flatpaks using bluetooth right now, so I don't have any reference to go by. Maybe  one would need to allow qtconnctivity to talk to bluez running on the host via dbus, at least that's how I read this: https://github.com/mer-packages/qtconnectivity/blob/master/qtconnectivity/src/bluetooth/bluetooth.pro#L101

dirkhh, 2017-09-27:
Sorry for the slow responses here - busy with my day job. I see the same error that you do, but I have no idea how to fix something like that. The question is how the FlatPak isolation model is designed and how an app is supposed to connect to host platform services - and then we need to make sure that the Qt implementation knows how to do that. I assume that you could build a trivial Qt Bt sample app (http://doc.qt.io/qt-5/qtbluetooth-btscanner-example.html) to explore this issue (without having to deal with the other complexities and build time of Subsurface)

Alexander-Wilms, 2017-09-29:
No problem. I'll see if I can get the Qt BT example working with the help of the people on the flatpak IRC channel.

Alexander-Wilms, 2017-11-29:
I added a permission to allow talking to org.bluez on the system bus, and now it seems to work. I tested it with my phone, which is recognized by Subsurface.

Alexander-Wilms, 2017-11-29:
Somehow, Subsurface doesn't know that the build is a release configuration: "You are running a development version; latest version is 4.7.4.181.". No idea why, since I'm using https://github.com/Subsurface-divelog/subsurface/archive/v4.7.4.tar.gz and use the "-DCMAKE_BUILD_TYPE=Release" option.

neolit123, 2017-11-29:
that's odd,
seems like our online script (`updatecheck.html`) is returning the latest release version to be `4.7.4.181` instead of `4.7.4.0`.

@dirkhh can help as i don't know the internals of the script.

Alexander-Wilms, 2018-01-08:
Since the package seems to work fine so far, I'll close this issue, so it doesn't clog up the issue tracker.
If any of the Subsurface maintainers want push-access to the https://github.com/flathub/org.subsurface_divelog.Subsurface repository, you can ping the admins: https://github.com/orgs/flathub/people

closed 2018-01-08 by Alexander-Wilms