subsurface/subsurface #592: Planner: ascend form last deco stop - change implementation?

Findingmenno, 2017-09-20:
Hi,

first of all thanks for the planner, great part of the software.

A question, I think the ascend to surface from last deco stop should be changed. It seems the last deco stop + ascend to surface seem to have you at the surface as your ceiling has been removed (ie deco been cleared). For an example dive:
*with a 6 min/sec ascend rate to surface and 6 mtr final deco stop it would be: 10 min final stop and 1 min ascend to surface.
*with a 1 min/sec ascend rate to surface and 6 mtr final deco stop the would be: 5 min final stop and 6 min ascend to surface.

I think a better implementation would be to start (not end) the final ascend as the deco has been cleared. So in example above: 11 min deco stop + X ascend time. This gives a bit more 'padding' in the final stop and ensure divers clear their stops completely

RickMWalsh, 2017-09-21:
Hi Findingmenno,

You raise an interesting point, and one that has been discussed extensively previously.  And you are correct that in the planner (at least when using the Buhlmann deco model*), the ascent time is considered when calculating when it is ok to ascend to the next stop, or surface.  It's true that ignoring the ascent time would increase the time of the final stop, but why should the algorithm ignore the user's selected parameters, and the algorithm that calculates that ascending to the surface at the chosen rate is ok?

The only person ensuring the diver clears their stops completely is the diver, who should understand what was planned, and execute the plan accordingly.  So if the plan has a final ascent rate of 1m/min, then the diver needs to ascend to the surface at that rate in order to obey the plan.

Personally, I plan my dives with all ascent rates at 9m/min, which assumes ascending from 6m will take less than 1min, but then I at least aim to ascend at 1m/min (ok in flat water, tough in rough seas), giving myself 5min of bonus deco.  I suggest you do something similar if that's what you want and expect.

I think there's an argument for changing the default ascent rates to 9m/min (assumed by Buhlmann, and recommended limit by most agencies), but I'll let Robert and Dirk make that call.

*With the VPM-B model, Subsurface actually does what you suggest, and ignores the ascent rate when determining whether it is ok to ascend.  Why? When implementing the model, which came after the Buhlmann model, we compared our plans to those created in other software, including the original Fortan / MS-Dos VPM-B program.  For consistency with the baseline implementation we chose to ignore ascent rates for VPM-B only.  No such "standard" implementation of Buhlmann model exists, and different VPM-B software calculates slightly different plans based on slightly different assumptions and idealizations.

neolit123, 2017-09-23:
@atdotde hi, robert could you have a look at the discussion between @RickMWalsh and @Findingmenno too see if this change-request is reasonable.

atdotde, 2017-09-23:
Totally agree with @RickMWalsh . The planner plans according to model, if the user wants fudge factors she should add those.

atdotde, 2017-09-24:
No, I meant "works as is". The planner should not do any fudging on the behalf of the user (as others will not like this).

neolit123, 2017-09-24:
@atdotde ok i see, i will label and close the issue as the change-request is not accepted.

closed 2017-09-24 by neolit123

Findingmenno, 2017-09-25:
Hi all,

thanks for the quick feedback.

In response to the few point made in the replies.

I fully understand and agree that the diver is in the end responsible to plan the dive, and plan the dive. That is what I do. He/she shouldn’t rely on any padding/‘fudge’ factors to execute a dive safely and to plan.

Also I agree that the current implementation follows the model, using deco stops and entered ascent rates it brings the diver to the surface when the stops are cleared.

However I feel in practical sense the ‘accepted’ norm in tech diving is that a computer generated plan clears you from your final stop after clearing your deco, so not considering the final ascent to surface into the deco time. It has been a while since I’ve been playing in Multideco, but I believe that works in that way. What might be muddling the waters here, is that you can select a slow ascent rate in the shallows in the planner (which most divers do) and not directly realizing that this effects the final stop time as people do not expect this.

Again I’m not saying that what the planner does is not correct in theoretical sense. But when I showed the planner to my tech instructor colleagues (I work at a tech dive center in the Caribbean) who are experienced Multideco users, this behaviour surprised them. The general consensus being that this implementation was not ‘how it should be done’.

glance-, 2017-09-25:
Those minutes in the end doesn't really change anything. Temperature, current, and what you do when you hit the surface isn't in the schedule either, but they really affect the outcome.

Deco schedules isn't really "rules", they are more like guidelines. On a good day you can get away with breaking a lot of rules, on a bad day you get punished without breaking any.

If you break the deco rules generated by subsurface, you get to keep all the pieces.

Findingmenno, 2017-09-26:
Yep, fully agree. And I’m aware that all deco theory and models are just approximations of what is happening, and that many factors influence if you get bent or not. Read Mark Powell’s book on deco, very interesting.

Maybe I should try to refine the point I’m trying to make. The implementation chosen in Subsurface in all aspects correct, but is different from the way other more established programs have implemented it. (AFAIK, as I do not have experience with all programs.) This requires an extra mental ‘awareness’ of the user (not a simple copy-paste of the calculated schedule on your slate), and may result in a loss of ‘trust’ in the programs planner. If you want to get users from other programs to use the planner, that might be a limiting factor.

atdotde, 2017-09-26:
When we decided to do it this way, we compared our plans to those of a number of other deco planning programs. I do not recall which but I can tell you quite a few others do also factor in the desaturation during ascents between stops (including the final ascent). And honestly, it makes no sense to take as the endpoint of your stop the fact that you could be at some other fictitious depth (which you could not reach since ascending 3m in an instant is a bad idea in any case). The only thing that matters is if you are above or below the ceiling.

I can see two reasons why one would do it as you suggest (but then one should do it at all stop depths, only the last one is totally random): just checking if the ceiling is above the next stop is easier computationally, so if you are limited by cpu power you might consider this.

The other is that you want to add some additional stop time to increase conservatism of your plan. But then why do it this way? Why not always stay five extra minutes at the shallow stops? Or do an extra 7 minutes at 1m of depth. This is totally up to you. But your deco program should not ask you to do this.

And if you really want the other thing, Subsurface allows you to do that: Simply set very high ascent velocities!

RickMWalsh, 2017-09-27:
In my opinion, the ideal situation would be both:
a) logical - let the diver ascend at the planned ascent velocity if doing so does not break the theoretical ceiling, and
b) consistent with user expectations.
In my technical training, the course texts run through examples planning dives with V-Planner/MultiDeco (TDI), and GUE DecoPlanner (GUE).  Unfortunately, I suspect neither of these consider ascent time in determining whether it is safe to ascend to the next level, so it is a fair point by @Findingmenno to say that is the expected behaviour.  It may not be "right" but it is expected.  Based on that, it would seem not possible to satisfy both (a) and (b) above.
However, when you consider that default ascent rates in theory (Buhlmann, Navy tables) software (definitely DecoPlanner and I think V-Planner and MultiDeco), and dive computers (I have a Shearwater, but the same holds for just about any other) are all 9-10m/min.  At this rate, the time to ascend from the final stop is less than one minute, which is hardly worth worrying about.  If the default ascent rates were changed to 9m/min, then:
- *by default* the difference in duration of the final stop compared to default operation of other software due to ascent time would be negligible
- the user could still adjust the ascent rates, and Subsurface will still consider the time it takes to ascend between levels.

This would achieve the intention of treating ascent time logically, while still producing (with default settings) results consistent with @Findingmenno (and others') expectations.

atdotde, 2017-09-27:
@RickMWalsh fair enough. Will you create a PR for a changed default?