subsurface/subsurface #594: Petrel via BT in latest builds

dirkhh, 2017-09-23:
That fails for @RickMWalsh and me on Mac / Fedora / Android

marccollin, 2017-11-04:
don't work too for me on linux (suse tumbleweed)

```
Subsurface: v4.7.2, built with libdivecomputer v0.6.0-devel-Subsurface-branch (e177b28627a81815211abb742cbf97dc7e5330a9)
INFO: Open: name=LE:00:13:43:0E:6C:C2
ERROR: Failed to open the serial port. [in shearwater_common.c:46 (shearwater_common_open)]
```

dirkhh, 2017-11-04:
It would be interesting to know exactly which Shearwater dive computer you have... it shows up as a BLE device, so this could be a newer Petrel 2 or any Perdix.
I have had problems with my dual stack Petrel 2 (the one that does both BT and BLE) and I know that I had it working again on Mac prior to the 4.7 release - but I haven't tried in Linux in a while.
I think Tumbleweed is on Qt 5.9.x? Can you confirm?

marccollin, 2017-11-04:
i use petrel 2 (have it since 2 years ago)

hardware said me: 2.4.2
tumbleweed curently use 5.9.2

marccollin, 2017-12-18:
still have the issue with 5.9.3

dirkhh, 2017-12-18:
So I have one Shearwater BLE dive computer that works flawlessly and one that intermittently fails.
For that second one (a Petrel 2 at the same hardware version as yours) what helps is to remove the battery for a day to make sure it completely resets the SRAM on the BLE chip and then restart the dive computer and connect it to my computer. Frustrating, but the only workaround I have found so far.

marccollin, 2017-12-27:
same issue
tumbleweed should us soon 5.10, maybe that will fix issues

marccollin, 2017-12-30:
ok tumbleweed get qt 5.10, but it's seem like there is a build issue

[](https://build.opensuse.org/package/live_build_log/home:Subsurface-Divelog/Subsurface/openSUSE_Tumbleweed/x86_64)

problem seem to be  QGeoMapType

current release of subsurface 4.7.6 is supposing to support qt 5.10?

appimage is working fine under tumbleweed

dirkhh, 2017-12-30:
4.7.6 supports Qt 5.10 just fine and 4.7.7 will build correctly on Tumbleweed.
I just triggered a new "daily build" for all the Linux flavors - Tumbleweed is rebuilding as we speak.
This won't replace the broken 4.7.6 build but will be a 4.7.6.4 build (complicated rambling reasons... just go with it, please).
So in order to install this (once the build finishes) please switch to the Subsurface-daily repository (and remember to switch back once 4.7.7 is out in about a month or so)

dirkhh, 2017-12-30:
Actually, forget what I said, I added a patch for it via the spec file so now Tumbleweed should build 4.7.6 as well - I'm just waiting for OBS to process and post this, but it should show up soon.

marccollin, 2018-01-02:
at term, proposing only a appimage could be a solution to reduce complexity of building a package for every distribution.

dirkhh, 2018-01-02:
That was our hope - but it turns out that a) there are still some issues with the AppImages and desktop integration and b) there are many Linux users who ask us for native packages.

BTW: can you now download from your Petrel or is this still broken?

marccollin, 2018-01-02:
with apps image that worked

seem like qt 5.10, fixed issue.