subsurface/subsurface #601: Deco ceiling calculation incorrect when replanning a dive in planner

sfuchs79, 2017-09-24:
While investigating another issue @atdotde and I made the following observation. Item is really low prio but I prefere to keep track of it:

Start from an empty log. Plan a VPMB+2 dive to 30m for 20m using the planner.
Save in the logbook.
Edit dive in planner.
Save again.
Save to xml, quit subsurface
Open Subsurface with this dive.
**Edit dive in planner: much less (incorrect!) deco ceiling displayed** (see first picture)
Delete all computed waypoints ceiling normal again (see second picture)

[image]

[image]

RickMWalsh, 2017-09-24:
Other than verifying the issue, I have not had time to look into this.  I suspect the CVA calculation may be taking deco time as zero, perhaps because the final defined point is at the surface.  Maybe profile.c is treating some parts as "real dive" and some parts of the code as "in planner".  Not sure if the fix is to be found within profile.c or in the code that sets up a dive to edit in the planner.
On another point, the deco ceiling in the second picture doesn't look right to me either.  The ceiling is shown as clearing ~10min before starting the final ascent, which isn't correct.

sfuchs79, 2017-09-25:
Yes, strange. Sorry, didn't do deep enough testing.

Add here a reference picture from same dive V4.6.4 to confirm Ricks additional finding:
[image]

After some hopefully correct tests I see this strange (?!) result:

Issue | Master with #599 both commits | Master with only Roberts commit | Master with only Ricks commit | Master w/o #599 
----- | ------- |----------  | ------- |----------
Spikes | No | Yes | No | Yes
Much less ceiling before removing waypoints | Yes | Yes | Yes | Yes
Little less ceiling after removing waypoints | Yes | Yes | Yes | No

neolit123, 2017-09-26:
thanks, for the details @sfuchs79.

atdotde, 2017-09-28:
Looking a bit more at the code, I agree with @RickMWalsh, this is not a bug of Subsurface but an oddity of VPM-B: In the 'critical volume loop', the allowed gradient for the next iteration is determined by the (I am using the logic of the model justifications, not that I really believe all those) gas that left the tissue and had a chance to form bubbles. That has a surface part that is not relevant here and another part from the deco part of the dive (since for dives for which VPM-B was designed, off-gassing only happens during deco). The latter is given by the allowed gradient (except for boyle compensation assumed constant during ascent in vpmb) times the time that it acts, i.e. the deco time. So the deco time is a relevant factor for the algorithm.

In the planner, we take "deco time" to mean the part of the dive that is handled by the planner (as opposed to the manually entered part of the dive). Thus, when reediting a saved dive, all the planner waypoints are then treated as entered and thus that part of the dive is no longer considered deco time. and this gives you in the end the greater allowed gradient and thus shallower ceiling.

So, the short answer is, we are taking VPM-B out of its range of validity (planned dives with a clear distinction between bottom time with on-gassing and deco with off-gassing). So, there are several options how to proceed: The simplest one (and I am leaning towards it) is: works as designed. Possibly we could add a paragraph to the manual warning not to trust VPM-B for dive plans with significant off-gassing during the manually entered part of the dive.

Option two would be to make our file format a bit smarter so we keep the dp->entered information on disk. This would solve the problem for the case described by @sfuchs79 but actually only hide the real problem. Therefore I am not in favour of this.

Option three is to try to take off-gassing in the manually entered part of the dive into account (by computing the time integral of the gradient for each tissue as long as the gradient is positive) and use this in the computation of the next gradient. This might be morally correct (if you believe the vpm-b story to begin with) but then we would clearly deviate from the original fortran program, at least for dives with off-gassing during the prescribed part of the dive.

So my proposal how to proceed is: Take option one for the time being (and write something for the manual). If somebody is really into fixing vpm-b then she might try to implement option three (but please with a ui option to turn it of for compatibility). So I will for now close this issue but feel free to reopen it if you disagree.

RickMWalsh, 2017-09-28:
Hi Robert,

I think there is a fourth option worth investigating, and that might not be
too difficult to implement. When we iterate to find the deepest ceiling, we
save the runtime at that point, then:
Deco time = total runtime - runtime at first ceiling
Unless I'm missing something this should be consistent with the original
plan.
Now I expect you'll ask when I can investigate if this idea works and for
me to send a pull request. Don't hold your breath; I might be able to find
some time soon or it could be weeks...

atdotde, 2017-09-28:
Indeed, that shouldn't be too hard. Will try tomorrow (time for bed now).

sfuchs79, 2017-09-30:
@atdotde @RickMWalsh 

Sorry, me again:
Just would like to make you aware of the fact that we are talking about two issues here:
- The "much less ceiling before removing waypoints" issue which was first recognized by Robert and me is maybe the one with less priority now. This is the old thing that is present since a long time.
- The "little less ceiling after removing waypoints" issue (see picture below) which was recognized by Rick and which is not present in 4.6.4 but appears after merging #599 is a little bit ugly and more important to fix. This issue is now even present when planning a new dive. So it's more of "calculated ceiling completely incorrect in planner and doesn't fit with planner deco calculation".

[image]

atdotde, 2017-09-30:
OK, now I spent another evening on this (investigating many different versions and bisecting). I am still not done, but my current understanding is this:

@sfuchs79 you are correct, these are two different issues. I guess issue one is addressed by my "this is how VPMB behaves" comment above.

Issue two is indeed caused by the code that removes the spikes. Or more specifically:
undoing 850e345 and commenting out the two lines 
	deco_state->first_ceiling_pressure.mbar = 0;
	deco_state->max_bottom_ceiling_pressure.mbar = 0;
in clear_vpmb_state() in deco.c reintroduces the spikes but moves the ceiling to "a little less". 

But looking at the ceiling behaviour upon replanning in 4.6.4 it seems to me that these changes are correct as in without those we were silently using invalid old state of these variables. And if we were coming from the planner, the first ceiling would still have the correct value from the last round of the planner (which should not influence the profile). 

So the upshot is: the two patches in the present pull request are "correct" but we still need to look into the vpmb code in profile.c. A good starting point would be to implement @RickMWalsh 's option four.

Any comments?

sfuchs79, 2017-10-03:
My hope whould be that the more ugly "little less ceiling" issue could be fixed by a minor change. My guess would be that any value is still not correctly initialised when calling the profile.c function(s) when "in_planner" but really I don't know.

Two minor comment:
- This comment here is incorrect, isn't it:
https://github.com/Subsurface-divelog/subsurface/blob/master/core/profile.c#L1085
- Could you reopen this issue?

sfuchs79, 2017-10-04:
@neolit123 Please reopen this issue as well

PR #654 (2017-10-08): VPM-B profile: calculate parameters when in planner mode
Calculating parameters when in the planner mode is necessary to display the correct ceiling.

Fixes #601

### Describe the pull request:

- [x] Bug fix
- [ ] Functional change
- [ ] New feature
- [ ] Documentation change
- [ ] Language translation

### Pull request long description:

### Changes made:

### Related issues:

### Additional information:

### Mentions:
commit f9d05a80 (2017-10-08): VPM-B profile: calculate parameters when in planner mode

Calculating parameters when in the planner mode is necessary to display the correct ceiling.

Fixes #601
closed 2017-10-08 by atdotde