subsurface/subsurface #602: Double increment/decrement issue with QSpinBox elements in planner preferences

sfuchs79, 2017-09-24:
Discussed between @atdotde and me on mailing list. Most probably not so easy to fix. Put it here to track it:

Under Win10 with current master if I increase or decrease any QSpinBox value in the planner preferences by clicking on the up or down arrow the QSpinBox value is increased/decreased by two steps instead of one. This happens with any QSpinBox based value like gas options, ascent rates, VPM conservatism,...

[...]

I found a hint how this double increment/decrement issue of the QSpinBoxes could be explained:
https://github.com/MRtrix3/mrtrix3/issues/306

Found even a second reference on the web and maybe a better solution:
https://github.com/manisandro/gImageReader/issues/37

[...]

>> Maybe we should indeed disable most of the planner UI elements while the calculations are performed and enable them again after they are finished. [...]
>
> no, the planning should happen in a background thread. [...]. Plus, it shouldn’t be a lock, each new invocation of a planner thread should kill all previous ones. 
>
Yes, you are for sure right that this would be the perfect solution.

> Yesterday, I sent a pull request for a switch to turn of the computations of the variations. With that switch set to off, do you still think the responsiveness of the planner is worse than before? In that case, a bit more profiling work could improve that situation.
>
> [...]
Yes, unfortunately even with variations turned off the responsiveness of the planner is a little bit worse than before. It is not worse in a way that I would complain about the plan calculation time itself but unfortunately the difference is enough to reveal this issue with the QSpinBox UI elements (Windows10, my PC (!) - can be different elsewhere). Nevertheless I'm absolutely not blaming your change in the algorithm for anything and this has also one specific reason: I already once saw the QSpinBox issue in v4.6.4 or even before on my (slow) Notebook. So as I was saying already at the very beginning: Your change only revealed the issue.

To be more precise:
Current master release build, variations off on my desktop PC running Win10: QSpinBox issue (double increment/decrement) always present
4.6.4 on my desktop PC running Win10: No QSpinBox issue
4.6.4 (or older) on my Atom based Notebook running Win10: QSpinBox issue (double increment/decrement) sometimes present

Than means for me it's not really top priority to try to improve calculation time but it is important to fix the issue with the QSpinBox elements. The UI elements should work correctly no matter how long the calculation takes.
For me there are still two viable options:
Your solution with the background thread or my proposal with disabling the QSpinBox UI elements while calculation is running.
My solution for sure has the disadvantage that the user has to wait for the end of the calculation after each QSpinBox change. It for sure also only work together with disabling keyboard tracking for all QSpinBox elements. That means user has to type, press enter or move focus to another UI item and then calculation will start. Today calculation is triggered after each key-press that changes the value.

neolit123, 2017-09-26:
i remember seeing this issue but it no longer happens for me even with complicated planned dives. the correct way is to use a separate thread for the planner calculations, as already suggested.

mikeller, 2026-10-17:
Hi @sfuchs79,

Following up on neolit123's comment above: this hasn't reproduced for them even with complicated planned dives, once the planner moved its calculations to a background thread rather than running them synchronously on UI interaction (which was the likely cause of the double-increment).

Closing this as likely resolved per that testing. Please reopen with your current OS/build details if you still see a QSpinBox jump by two steps per click in the planner preferences.

closed 2026-10-17 by mikeller