subsurface/subsurface #605: show again the full list of tags

antiker1, 2017-09-24:
Hi.

There's the filed "Tags", which auto-completes to some apparently hard-coded standard fields and to whatever one has manually used within a log.

In earlier versions, when one wrote e.g. "salt,shore," then after a new "," a dropdown list with *all* the known tags was presented.
This seems to be no longer the case and at least one letter is required for such list e.g. "s" will give salt, short and student.

I think the previous behaviour was much nicer, as it gave me the a quick overview of all the tags in use.
So one would not accidentally forget a rarely-used tag like e.g. altitude when one adds such a dive, as one could quickly go through the list.

It would be even nicer if the list was sorted first by whether a tag has already been used or not, and within those two groups alphabetically.

Thanks for your consideration

AFulgens, 2017-09-25:
+1

Not only that, but a "tag editor" feature would be nice: a simple CRUD for tags, and _maybe_ bulk-applying tags onto multiple dives.

mturkia, 2017-09-25:
Bulk applying tags is supported. Just add the tags you wish to one dive, ctrl-c to copy them (you get a dialog to ask what info to copy), select multiple dives and hit ctrl-v to paste the data to all selected dives.

AFulgens, 2017-09-25:
@mturkia my bad, I'm still a new user to Subsurface :) But am I correct, that a tag-editor does not exist currently? (I am a bit confused, because I added a custom tag to some of my dives, then deleted it from all dives, and the dropdown still shows this tag)

dirkhh, 2017-09-25:
>@mturkia my bad, I'm still a new user to Subsurface :) But am I
>correct, that a tag-editor does not exist currently? (I am a bit
>confused, because I added a custom tag to some of my dives, then
>deleted it from all dives, and the dropdown still shows this tag)

Filter for the tag. I'm sure it's still somewhere.
Actually, let's be more precise. It was there when you opened the file.
Once you quit Subsurface and start again, only default tags and those included in your dive list will be shown.

closed 2017-11-17 by janmulder

antiker1, 2018-09-02:
Thanks for fixing this :-)

PR #820 (2017-11-17): Show all autocompletes after entry of comma in tag list
The almost 3 year old commit e1db5f21b15b2 introduced the behavior that you need to type at least 1 character for a tag to show any autocompleted data. While this went unnoticed for years, there
is always a user that notices this changed behavior.

The solution is rather simple. Do allow the 0 length string to act as seed for autocompletion instead of just returning, and doing no autocomplete at all.

As the afore mentioned commit was explicitly meant to prevent over active firing of the tag list autocompleter, this (simple) change is car [... cut]
commit c8a7712e (2017-11-17): Show all autocompletes after entry of comma in tag list

The almost 3 year old commit e1db5f21b15b2 introduced the behavior
that you need to type at least 1 character for a tag to show any
autocompleted data. While this went unnoticed for years, there
is always a user that notices this changed behavior.

The solution is rather simple. Do allow the 0 lenght string
to act as seed for autocompletion instead of just returning, and
doing no autocomplete at all.

As the afore mentioned commit was explicitly meant to prevent over
active firing of the tag list autocompleter, this (simple) change
is ca [... cut]