subsurface/subsurface #606: Calculation for dive end time - "when + duration" vs. function dive_endtime()

sfuchs79, 2017-09-24:
Stumbled across s.th. a few days ago which I would like to discuss. And btw this would be even s.th. where I can then try to fix it but better first have a discussion ;-)

At various places in the code we need to calculate the end time of a dive and this is today often done by adding dive.duration to dive.when. This may be incorrect as soon as a dive has surface intervals in the middle of the dive which is not so unusual.
Examples for code like this are:

https://github.com/Subsurface-divelog/subsurface/blob/master/core/divelist.c#L232

https://github.com/Subsurface-divelog/subsurface/blob/master/core/divelist.c#L398
https://github.com/Subsurface-divelog/subsurface/blob/master/core/divelist.c#L404

In fact we already today have a function in dive.c which imho corretly calculates the end time:
timestamp_t dive_endtime(const struct dive *dive)
https://github.com/Subsurface-divelog/subsurface/blob/master/core/dive.c#L3494

Shall we correct this and use the function dive_endtime everywhere where it is appropriate?

neolit123, 2017-09-24:
@sfuchs79 ideally we should use a helper when possible and this look like a good change to me.

but i don't understand if we should *always* use it. for instance, in the provided `divelist.c` backend examples `init_decompression()` might want to take the surface time into account, thus it uses the raw calculation.

are there other instances of not using the helper, which can we can cover in a single patch?

@dirkhh @torvalds @janmulder

closed 2017-10-02 by sfuchs79

dirkhh, 2017-10-02:
You can automagically close issues in your commit message... simply say

Closes #xxx

on its own line :-)

(and yes, I keep forgetting to do that myself)