subsurface/subsurface #609: Android beta crashes on startup on Galaxy S8

adamel, 2017-09-27:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:
Subsurface crashes on startup, after showing splash screen but before anything else is shown. I have deleted all app data and logged in to the cloud account again, and the same thing happens.

### Operating system:
Android 7.0, Samsung Galaxy S8
Works fine on Nexus 7 2013 with the same cloud account

### Subsurface version:
2.0 (4.6.4.759)
Android beta

### Steps to reproduce:
Start app.

### Additional information:
Multiple crash reports sent as feedback from device.

dirkhh, 2017-09-27:
Thanks for the report. Android crash reports tend to be not very helpful when trying to track down issues like this (and, frankly, I haven't received them, yet - I need to check the Google Play console to see what's up with that).
After the crash, could you locate the `subsurface.log` file in the root of your storage device and attach this to this issue? That's more likely to show us what's wrong.

adamel, 2017-09-27:
[subsurface.log](https://github.com/Subsurface-divelog/subsurface/files/1336124/subsurface.log)

dirkhh, 2017-09-27:
Thanks for attaching the log.
Wow. It successfully gets your paired devices from BT.
It successfully opens the local cache of your dive data with 759 dives (which means at an earlier run it successfully connected to the cloud server).
It starts the GPS location service.
So all of the 'hard' stuff it already succeeded doing.
And then it crashes. Hmm indeed.
I assume the crash shows itself as the app simply closing? Or is it hung and unresponsive?

dirkhh, 2017-09-27:
OK, finally was able to locate your crash log in the Google Play console. It only shows crashes from the production version (and there were none) by default. Duh.
This appears to be something that some other Qt based Android apps have run into with some Samsung devices. Do you have LastPass or some other serive app enabled in the Accessibility options of your phone? If yes, could you try disabling that and try Subsurface-mobile again to see if this is the same issue you are seeing?

adamel, 2017-09-27:
Ah, that was it! I was actually running the Bixby Remapper mentioned at https://bugreports.qt.io/browse/QTBUG-45855 so it's not a terrible pain for me to disable it when I want to run Subsurface. It would be nice if there was an app side workaround of course.

dirkhh, 2017-09-28:
Glad that we have a workaround. Since the crash is deep inside Qt, there's not much we can do on the app side until this is fixed in Qt. As you saw on the bug report above, there's no activity / proposed fix for this, so I have no idea if and when this will get addressed.
As far as Subsurface is concerned, I'll close this bug as there's nothing we can do.

closed 2017-09-28 by dirkhh