subsurface/subsurface #610: gas change not (fully displayed) when switching back to the previously used gases if these are cylinder 0

antiker1, 2017-09-27:
Hi.

This may overlap with the gas change/side mount/cylinder issues already mentioned in #561, but I thought it would be better to have a new issues, to avoid complete obfuscation ;-)

I have some deeper TecDives (beyond the 100m) which made it necessary to perform air breaks (in which air is not necessarily air, but simply the least O2 percentage gas available) at the shallower stops in order to keep the CNS from getting too high.

So basically, I set the DC to use back and forth between the last deco gas (e.g. some NX 90) and the "air" (e.g. some TMX 35/35).

Now on one such dive, everything is displayed nicely, and the profile contains the gas tanks at each switch of the air break (though it doesn't show the (interpolated) curve from the gas consumption on the subsequent usage of the multiply used gas - only on the first usage).

But on one other dive, it doesn't show the tank icons (though when I move the mouse over the location where it should be, that small box shows me the gas change event).

The only visible difference between these two is, the in the first case, the 35/35 (i.e. the gas which was used multiple times during the dive), wasn't the first gas (back then I had another 17/40 for the first part of the descent)... in the second dive, the 35/35 was also the first gas (and is also cylinder 0).

In both cases,... the gas consumption curve is only shown on the first usage of it.

Cheers.

mturkia, 2017-09-27:
Can you export a log with these two dives that demonstrate the issue, and provide it for testing?

atdotde, 2017-10-18:
@antiker1 could you please attach a log file that demonstrates this bug?

sfuchs79, 2017-10-19:
@atdotde @willemferguson 
I saw also a dive a few days ago where for one gas change everything was ok save the tank change icon was not there. I did some investigations why for other gaschange events in the same dive the icon is displayed and for a single one not.
Rather quickly found the cause: The gaschange events with icon displayed have either no "flags" or "flags='1'". The gaschange w/o icon has "flags='4'". This flags=4 seems to be ok in general because it says cylinder 3 and this is correct.
I then did more investigations and it seems that the gaschange even with flags='4' is "filtered away" here:
https://github.com/Subsurface-divelog/subsurface/blob/master/profile-widget/diveeventitem.cpp#L86
It seems to be considered as a "useless internal of the dive computer".
If I move up the
else if (event_is_gaschange(internalEvent))
clause before this L86 then in my case the gaschange items shows up.
Will send a PR in a few minutes to make more clear what I mean.
Maybe we can even take it as the fix and maybe it fixes even the issue reported by the user here.
On the other hand it would be also good to better understand the background of this "useless internal of the dive computer" filtering.

closed 2017-10-19 by dirkhh

antiker1, 2018-09-02:
@mturkia  Sorry for not having provided the log. I kinda lost access to my account here back then, which I basically just made for subsurface :-D

I tried this again with 4.7.6-1~zesty (4.8.0 is broken for me, see #1624) and the issues I've described above seem to be fixed.

Thanks @sfuchs79 :-)

PR #695 (2017-10-19): When selecting icons for events don't misinterpret a gaschange

### Describe the pull request:

- [x] Bug fix
- [ ] Functional change
- [ ] New feature
- [ ] Documentation change
- [ ] Language translation

### Pull request long description:

Don't misinterpret a gaschange as "useless internal of the dive computer"
when it has a higher flags value.

Please refer to the discussion here: #610 

### Changes made:

### Related issues:

Maybe fixes #610 

### Additional information:

### Mentions: