subsurface/subsurface #612: Facebook upload not uploading as expected

fezzington, 2017-09-27:
### Describe the issue:

- [X] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

1) When passing a new Facebook upload, Subsurface prompts for an album name, and passes that to the Facebook API. However, once uploaded, Facebook groups all uploads, regardless of what the user indicates, into an album called "Subsurface Photos"
2) These uploads are supposed to be set to Private (or, "SELF", in FB API speech) but are shared by default with "Friends" -- Or at least, that's how I interpret line 124 in facebookconnectwidget.cpp
### Operating system:

macOS 10.11.6 on a late 2012 rMBP. I am unable to test other OSes at this time

### Subsurface version:

4.64.839, built by dirk (I believe this is a test build)

I believe it's using Facebook Graph 2.10, the latest FB API

### Steps to reproduce:

1) Select dive to export
2) Enter Album name
3) Upload

### Current behavior:

The dives are uploaded to "Subsurface Photos", and set to "Friends"
### Expected behavior:

The dives should be uploaded to the user defined album, and initially set to private (or "SELF")
### Additional information:

### Mentions:

@tcanabrava -- I'm told you may be the one to ask about the API stuff?

ggardet, 2017-10-23:
The problem is that we are lacking 'user_photos' Facebook permission to create an album, but we are able to upload the image to Facebook, but not in a specific album. So, the image will be published on your timeline (friends accessible) and Facebook will name the album 'Subsurface Photos'.

ggardet, 2017-11-10:
@tcanabrava or @dirkhh (not sure who submitted Subsurface App to Facebook), could you add missing permissions in the Facebook admin page, please?

dirkhh, 2017-11-10:
I’ll look into that as well - this was submitted by me.

neolit123, 2019-02-01:
we are  removing the FB feature:
https://github.com/Subsurface-divelog/subsurface/pull/1958

closed 2019-02-01 by neolit123