subsurface/subsurface #615: OSTC4: inactive cylinders are still added to the dive

antiker1, 2017-09-28:
Hey.

In the OSTC4, when I have e.g. all but one cylinder set to inactive, then subsurface (on downloading the dive) still adds 5 tanks to it.
This didn't happen with the OSTC2, IIRC. So perhaps there is some bug or the "inactive" flag is ignored.

Thanks.

janmulder, 2017-09-28:
Checked my (OSTC3) logbook, and this behavior has been there as long as I have my OSTC3 (oct. 2014). Definitely not saying that this is correct behavior, but we need to be careful changing things here. @glance-: any opinion on this?

glance-, 2017-09-28:
Subsurface has a flag, display_unused_tanks , which defaults to off. If you would have left that off, subsurface shouldn't show you those tanks.

On the other end, libdivecomputer doesn't care about the "type" field, so we don't know if a certain tank is "inactive" or not. When the libdivecomputer code sends the tank list to subsurface it sends all tanks, because you could be switching to a inactive gas later in the dive.

I'm not saying that your observation is wrong, i'm just saying that it just ain't as simple as ignoring things flagged as inactive.

janmulder, 2017-09-28:
@glance-. Thanks, yes, even developers do not know all the settings :-) Indeed, I had it set to "show also unused". Not sure that I had ever set this to true, but apparently I did.

And just checked that I can switch to a cylinder that is interfaced but hidden by the setting, so that seems fine too.

Temped to close this one as no bug. @antiker1: what is your opinion.

dirkhh, 2017-09-28:
This has been there since the beginning of OSTC support.
While this is surprising when you first see it, this isn't a bug since the OSTC (like many tech computers) doesn't tell us if a cylinder was used or not. And some tech divers like to show the additional cylinders that they had on a dive, and some don't.
Closing this as "it may not be what you expected, but this is not a bug"

closed 2017-09-28 by dirkhh

antiker1, 2018-09-02:
@janmulder @dirkhh I'm fine with closing it. I think to remember that with OSTC2 it didn't add me a `<cylinder>` when it wasn't configured in the DC. But it's really no problem for me to delete it in subsurface.