subsurface/subsurface #619: Crash on close

marccollin, 2017-09-30:
I use subsurface and when i close application, it crash.

### Issue long description:

### Operating system:
Suse tumbleweed
kernel 4.13.3

### Subsurface version:
4.6.4

### Steps to reproduce:
Start subsurface
Close subsurface

### Current behavior:
Subsurface crash

### Expected behavior:
Subsurface close correctely

### Additional information:

Application: subsurface (subsurface), signal: Segmentation fault
Using host libthread_db library "/lib64/libthread_db.so.1".
[Current thread is 1 (Thread 0x7f528cacbdc0 (LWP 7850))]

Thread 8 (Thread 0x7f524c882700 (LWP 7875)):
#0  0x00007f528b8d9e28 in read () from /lib64/libpthread.so.0
#1  0x00007f52807ad360 in ?? () from /usr/lib64/libglib-2.0.so.0
#2  0x00007f5280768bfb in g_main_context_check () from /usr/lib64/libglib-2.0.so.0
#3  0x00007f52807690c0 in ?? () from /usr/lib64/libglib-2.0.so.0
#4  0x00007f528076922c in g_main_context_iteration () from /usr/lib64/libglib-2.0.so.0
#5  0x00007f52855f565b in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) () from /usr/lib64/libQt5Core.so.5
#6  0x00007f528559e30a in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () from /usr/lib64/libQt5Core.so.5
#7  0x00007f52853ce31a in QThread::exec() () from /usr/lib64/libQt5Core.so.5
#8  0x00007f52853d2d2e in ?? () from /usr/lib64/libQt5Core.so.5
#9  0x00007f528b8d0558 in start_thread () from /lib64/libpthread.so.0
#10 0x00007f528467b1bf in clone () from /lib64/libc.so.6

Thread 7 (Thread 0x7f524f19f700 (LWP 7870)):
#0  0x00007f528b8d683d in pthread_cond_wait@@GLIBC_2.3.2 () from /lib64/libpthread.so.0
#1  0x00007f52880fae13 in ?? () from /usr/lib64/libQt5WebKit.so.5
#2  0x00007f528843f531 in ?? () from /usr/lib64/libQt5WebKit.so.5
#3  0x00007f528b8d0558 in start_thread () from /lib64/libpthread.so.0
#4  0x00007f528467b1bf in clone () from /lib64/libc.so.6

Thread 6 (Thread 0x7f5255ffb700 (LWP 7869)):
#0  0x00007f528b8d683d in pthread_cond_wait@@GLIBC_2.3.2 () from /lib64/libpthread.so.0
#1  0x00007f52880fae13 in ?? () from /usr/lib64/libQt5WebKit.so.5
#2  0x00007f528843f531 in ?? () from /usr/lib64/libQt5WebKit.so.5
#3  0x00007f528b8d0558 in start_thread () from /lib64/libpthread.so.0
#4  0x00007f528467b1bf in clone () from /lib64/libc.so.6

Thread 5 (Thread 0x7f52567fc700 (LWP 7868)):
#0  0x00007f528b8d683d in pthread_cond_wait@@GLIBC_2.3.2 () from /lib64/libpthread.so.0
#1  0x00007f52880fae13 in ?? () from /usr/lib64/libQt5WebKit.so.5
#2  0x00007f528843f531 in ?? () from /usr/lib64/libQt5WebKit.so.5
#3  0x00007f528b8d0558 in start_thread () from /lib64/libpthread.so.0
#4  0x00007f528467b1bf in clone () from /lib64/libc.so.6

Thread 4 (Thread 0x7f5256ffd700 (LWP 7867)):
#0  0x00007f528b8d683d in pthread_cond_wait@@GLIBC_2.3.2 () from /lib64/libpthread.so.0
#1  0x00007f52880f9d27 in ?? () from /usr/lib64/libQt5WebKit.so.5
#2  0x00007f528843f531 in ?? () from /usr/lib64/libQt5WebKit.so.5
#3  0x00007f528b8d0558 in start_thread () from /lib64/libpthread.so.0
#4  0x00007f528467b1bf in clone () from /lib64/libc.so.6

Thread 3 (Thread 0x7f5257fff700 (LWP 7866)):
#0  0x00007f528b8d683d in pthread_cond_wait@@GLIBC_2.3.2 () from /lib64/libpthread.so.0
#1  0x00007f528840aea4 in ?? () from /usr/lib64/libQt5WebKit.so.5
#2  0x00007f528840aec9 in ?? () from /usr/lib64/libQt5WebKit.so.5
#3  0x00007f528b8d0558 in start_thread () from /lib64/libpthread.so.0
#4  0x00007f528467b1bf in clone () from /lib64/libc.so.6

Thread 2 (Thread 0x7f5268589700 (LWP 7855)):
#0  0x00007f5284670a7b in poll () from /lib64/libc.so.6
#1  0x00007f5280769119 in ?? () from /usr/lib64/libglib-2.0.so.0
#2  0x00007f528076922c in g_main_context_iteration () from /usr/lib64/libglib-2.0.so.0
#3  0x00007f52855f565b in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) () from /usr/lib64/libQt5Core.so.5
#4  0x00007f528559e30a in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () from /usr/lib64/libQt5Core.so.5
#5  0x00007f52853ce31a in QThread::exec() () from /usr/lib64/libQt5Core.so.5
#6  0x00007f527df75a45 in ?? () from /usr/lib64/libQt5DBus.so.5
#7  0x00007f52853d2d2e in ?? () from /usr/lib64/libQt5Core.so.5
#8  0x00007f528b8d0558 in start_thread () from /lib64/libpthread.so.0
#9  0x00007f528467b1bf in clone () from /lib64/libc.so.6

Thread 1 (Thread 0x7f528cacbdc0 (LWP 7850)):
[KCrash Handler]
#6  0x00007f528a74a15c in Marble::GeoDataObject::~GeoDataObject() () from /usr/lib64/libssrfmarblewidget.so.21
#7  0x00007f528a740d61 in Marble::GeoDataIconStyle::~GeoDataIconStyle() () from /usr/lib64/libssrfmarblewidget.so.21
#8  0x00007f528a729653 in Marble::GeoDataStyle::~GeoDataStyle() () from /usr/lib64/libssrfmarblewidget.so.21
#9  0x00007f528a729679 in Marble::GeoDataStyle::~GeoDataStyle() () from /usr/lib64/libssrfmarblewidget.so.21
#10 0x00007f5266cb1987 in ?? () from /usr/lib64/libmarblewidget-qt5.so.28
#11 0x00007f52845bbcd8 in __run_exit_handlers () from /lib64/libc.so.6
#12 0x00007f52845bbd2a in exit () from /lib64/libc.so.6
#13 0x00007f52845a3f51 in __libc_start_main () from /lib64/libc.so.6
#14 0x0000559a80d9b82a in _start ()

dirkhh, 2017-09-30:
Given the crash location, I'm sure this won't happen in the latest test build.

Thanks for the report. We have removed Marble for the next version.

neolit123, 2017-10-06:
closing due to "bug-already-fixed". marble no longer used in latest build.

closed 2017-10-06 by neolit123