subsurface/subsurface #625: parseGPSText isn't stable for some values

dirkhh, 2017-10-02:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

If you edit some other aspect of a dive site and leave the GPS coordinate text unchanged, you might still get a change to the coordinates.
### Operating system:

Tested on ArchLinux, but should be the same on any OS

### Subsurface version:

latest master -895

### Steps to reproduce:

1) create a dive site with coordinates '7.32542 134.217' (that's Sia's Wall, btw)

2) modify something else in the dive site dialog

3) save the dive file

### Current behavior:

site is saved with coordinates '7.325419 134.217'
if you then do the same again, the value changes to '7.325418 134.217' (and that value is then stable)
### Expected behavior:

coordinates should stay unchanged
### Additional information:

### Mentions:

glance-, 2017-10-02:
Its probably related to the fact that we treat latitude / longitude as doubles there, instead of degrees_t.

dirkhh, 2017-10-02:
yes, we have a few functions in that code that all use doubles - I filed the issue because I didn't think I'd get to fixing this today. Feel free to jump on it @glance-  :-)

neolit123, 2017-10-07:
@dirkhh 
strangely, i couldn't reproduce this with the latest master on 64bit Linux and 32bit Windows.

but i will send a PR for using lrint() consistently when rounding to `degrees_t`, because currently we use either `(int)` cast, `lrint()` or `llrint()`.

atdotde, 2017-10-18:
Can this be closed @dirkhh ?

dirkhh, 2017-10-18:
Yes, this seems stable now.

closed 2017-10-18 by dirkhh