subsurface/subsurface #630: Deco ceiling calculation for EAN32 dive with 3m steps on/off - very strange results

sfuchs79, 2017-10-03:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

Strange effect once again with deco ceiling calculation. This time visible outside the planner.
Example dive attached. If you for this example dive with EAN32 as bottom gas switch on/off 3m steps for deco ceiling calculation the overall ceiling result changes. This should not be the case because the 3m steps should only round up every ceiling to next 3m stop.
This issue is gone as soon as you change the bottom gas for the dive to s.th. other than EAN30-33.

Maybe linked to #601 

### Operating system:

Windows 10

### Subsurface version:

Latest master

### Steps to reproduce:

### Current behavior:

### Expected behavior:

### Additional information:

[EAN32_dive_deco_ceiling.zip](https://github.com/Subsurface-divelog/subsurface/files/1353675/EAN32_dive_deco_ceiling.zip)

### Mentions:

atdotde, 2017-10-03:
There are other things clearly wrong with this dive: I see spikes and what is more, the lowest ceiling does not appear at the end of the bottom time.

sfuchs79, 2017-10-03:
Yes, exactly. Very strange. Did you try to change the bottom gas to 30% and 33%? Even no need to start the planner for this, simply change in the dive view.
Then the strange issue is gone.
Does Subsurface hate Nitrox 32?

atdotde, 2017-10-03:
My suspicion is that this comes from the fact, that for VPMB the deepest ceiling matters and for that we probably also use the rounded value (and shouldn't). But so far, I have not checkt the source for that.

sfuchs79, 2017-10-03:
Ok, with the change the effect from switching on/off the 3m steps is gone.
But still if you play with O2 values between 29% and 35% for the example dive you see that there is s.th. clearly wrong.
For O2 31% you have deeper/longer ceiling compared to result for O2 30%.

atdotde, 2017-10-04:
@sfuchs79 I can confirm this.
[EAN2.ssrf](https://github.com/Subsurface-divelog/subsurface/files/1354847/EAN2.log)

I am, though, not sure if this is really a Subsurface bug or just another oddity of VPM-B. Have to investigate this further how the change comes about (guess: first ceiling is involved once more).

sfuchs79, 2017-10-04:
If you plan the very same dive in the planner and increment O2 from 28% to 36% you can see that the planner is doing everything as expected. The deco/runtime for the dive is decreasing for every O2 percentage you add. Ok, sometimes adding 1% O2 doesn't change by a full minute but at least it looks as it is continuous.

So it's for me as with most of the issues in the deco ceiling display from the last weeks:
The planner code responsible for the waypoints and the written diveplan does everything correctly but the deco ceiling calculation in profile.c is clearly buggy!

@neolit123 Can you please reopen this issue?

PR #632 (2017-10-04): When finding deepest ceiling, do not round to multiples of 3m
Fixes #630

### Describe the pull request:

- [x ] Bug fix
- [ ] Functional change
- [ ] New feature
- [ ] Documentation change
- [ ] Language translation

### Pull request long description:

### Changes made:

### Related issues:

### Additional information:

### Mentions:
commit a163bbc4 (2017-10-04): When finding deepest ceiling, do not round to multiples of 3m

Fixes #630
PR #730 (2017-10-30): VPMB profile: use bottom_time to calculate deco_time in planner
This corrects the issue where the displayed ceiling in the profile was
"broken" by the planner, especially for shorter and shallower dives.

Also fixes issue outside of planner where the deepest VPM-B ceiling was shown
too early, messing up the deco_time calculation.

VPM-B plans respond to change in O2% in gas as expected (in my testing)

Fixes: #630

### Describe the pull request:

- [ ] Bug fix
- [ ] Functional change
- [ ] New feature
- [ ] Documentation change
- [ ] Language translation

### Pull request long description:

## [... cut]
commit 04383e27 (2017-10-30): VPMB profile: use bottom_time to calculate deco_time in planner

This corrects the issue where the displayed ceiling in the profile was
"broken" by the planner, especially for shorter and shallower dives.

Also fixes issue outside of planner where the deepest VPM-B ceiling was shown
too early, messing up the deco_time calculation.

VPM-B plans respond to change in O2% in gas as expected (in my testing)

Fixes: #630
closed 2017-10-30 by atdotde