subsurface/subsurface #636: Crash on "edit dive site" for dive with empty dive location which is inside a dive trip

sfuchs79, 2017-10-04:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

The crash only happens if the dive where I do "edit dive site" is inside a
dive trip. Please see the example file attached which is good enough on my
side to reproduce the issue.
Step by step:
- Open Test.xml
- Select dive 710
- Do "edit dive site"
- Press "Discard changes" either with or w/o editing s.th.
--> Crash

### Operating system:

Windows 10

### Subsurface version:

Latest master d774b6cc499220b908a1d57909df786619d8fe60

### Steps to reproduce:

### Current behavior:

### Expected behavior:

### Additional information:

[Test.zip](https://github.com/Subsurface-divelog/subsurface/files/1357395/Test.zip)

[Dive_site_edit_crash_1-1.txt](https://github.com/Subsurface-divelog/subsurface/files/1357397/Dive_site_edit_crash_1-1.txt)

### Mentions:

neolit123, 2017-10-04:
@dirkhh @tcanabrava 
seems dive site related:
```
00507BD2 044FCB50 00000000 044FCA18  subsurface.exe!myInvalidate  [/home/stefan/Entwicklung/Subsurface/subsurface/qt-models/filtermodels.cpp @ 384]
0050804F 00000022 48C87034 044FCA38  subsurface.exe!stopFilterDiveSite  [/home/stefan/Entwicklung/Subsurface/subsurface/qt-models/filtermodels.cpp @ 456]
```

sfuchs79, 2017-10-05:
Very interesting update here:
The crash indeep has nothing really to do with the "edit dive site" function. I also see a similar crash when playing around with the dive filter while I have dives grouped in dive trips.

What seems to crash is DiveListView::clearTripSelection()
https://github.com/Subsurface-divelog/subsurface/blob/master/desktop-widgets/divelistview.cpp#L250

If I comment out
https://github.com/Subsurface-divelog/subsurface/blob/master/qt-models/filtermodels.cpp#L390
then I don't have the same crash in the same moment as initially described.
What then happens is that the dive selection is messed up and afterwards I may have a crash again if I don't fix the dive selction by hand by clicking on a dive.

neolit123, 2017-10-07:
@sfuchs79 

i think that the recent commits from @dirkhh fixed this. i can't reproduce a crash also the dives without location text are no longer editable.
please, confirm so that we can close it?

sfuchs79, 2017-10-07:
@neolit123 Unfortunately I already discussed with Dirk that there are still similar crashes. The occurence is reduced but with some use cases its still there and I would like to keep and update this issue to track this.
Could you please have a look at the conversation (~5 last comments) here #647 
I described here a step by step scenario where I still see a crash. Can you reproduce this?

neolit123, 2017-10-07:
moving the discussion here:

@sfuchs79 
```
Steps to reproduce:

New logbook
Add (manually) new dive A w/o immediately giving a dive site name
Add dive to divetrip
Enter dive site name for dive A "Test Test" (the space may be important!)
Click "edit dive site" button
--> Crash
crash_entering_location_space.txt
```

oddly, i cannot reproduce this on Windows 7 (32bit build) and on Ubuntu (64bit build) after multiple attempts.

the crash log suggests something being wrong with `MultiFilterSortModel`.
i've tried playing with the filter too, but can't get it to crash.

given the error is triggered from `moc`, @sfuchs79 have you tried cleaning your build completely and starting a new clean build to see if the crash remains?

sfuchs79, 2017-10-07:
Yes, done. Crash still happens.

dirkhh, 2017-10-08:
Try master again... forgot to mention this issue in the commit message. I certainly found some dumb errors in string handling in my code :-(

sfuchs79, 2017-10-08:
No, sorry, no success yet. Crash is still happening.

dirkhh, 2017-10-08:
OK, I don't see the crash here - but then I don't run Subsurface on Windows and don't ever debug things on Windows because, well, Windows.
@neolit123 - looks like you can't reproduce it either.
@sfuchs79  - is this only with the binaries you build from source, or also with my binaries? I'll trigger new test builds in a minute so you have the latest version from me to compare with...

sfuchs79, 2017-10-08:
Argl...
My debug build (what I always tested during the last days): Crash
Your "daily 947": No crash
My release build (same codebase as debug): No crash

Ugly - once again only one build type crashes - nothing new, nothing really helpful.

dirkhh, 2017-10-08:
Well - from my perspective, the good news is that our users are less likely to run into this!
Does it seem reasonable to close this as it doesn't show in release builds?

sfuchs79, 2017-10-08:
May I propose this:
Yes, close it in the way that mainly you don't have to spent another huge amoung of time on this at the moment. But keep this issue open because maybe I will further investigate and if s.o. else also sees the crash will (maybe) find this open issue and maybe is even able to help somehow.

dirkhh, 2017-10-08:
Ok, so let's not close this, but change the tags :-)

sfuchs79, 2017-10-16:
Ok, there are new interesting findings and I'm getting more confident that this bug can be found. Some items were already mentioned here #669 but I'm going to summarize again:
Up to today I had no tags displayed simply because I missed to setup the "dive site layout categories" in my preferences - all were set to "none" - how stupid.
Now I selected a reasonable setup and I see the following behaviour:

- I do edit dive site for dive A which has 3 tags:
[image]

--> Everything ok in edit dive site dialog:
[image]

- Then I exit the edit dive site dialog and do edit dive site for another dive B which has only 1 tag yet:
[image]

--> In edit dive site dialog: Tag 1 ok, but there is also a tag 2 and 3 which has either the info from dive A or if I exit and enter the edit dive site dialog multiple times tag 2 and 3 have random characters:
[image]

If I then do the same for another dive C also with only one tag it may happen that the same occurs again or it may even cause the crash.

Could it be that at some point the pointers for the 2nd and 3rd category are not (re)set to NULL if there is only 1 category?

sfuchs79, 2017-10-17:
Fixed the thing I mentioned last with the random characters with this change:
5ac895ac904585dbcb2e2af80855012828ee621a
But the crash for specific operations like the following is still there with the MXE Windows debug build:

New logbook
Add (manually) new dive A w/o immediately giving a dive site name
Enter dive site name for dive A "Test"
Add dive to new divetrip
Click "edit dive site" button
--> Crash

So this was not really related but two independent things. Or may there be a possible link?!

One stupid question: Which function(s) are triggered first when one presses the "edit dive site" button?

tcanabrava, 2017-10-17:
Just one, you can discover by following the signals:

on maintab.cpp: 
       connect(ui.editDiveSiteButton, SIGNAL(clicked()), MainWindow::instance(), SIGNAL(startDiveSiteEdit()));
on mainwindow.cpp 
	connect(this, SIGNAL(startDiveSiteEdit()), this, SLOT(on_actionDiveSiteEdit_triggered()));

and that calls:

void MainWindow::on_actionDiveSiteEdit_triggered() {
	setApplicationState("EditDiveSite");
}

(This can surely be cleaned up by having just one connect instead of two)

This function will only display the correct widgets for the state, nothing more. Can you provide me a backtrace of the crash?

sfuchs79, 2017-10-17:
But what does then happen next? We are looking for something related to the dive site taxonomy and there maybe s.th. writing to uninitialized memory/pointer. But it for sure may be also s.th. completely different...

Here is the backtrace:
[dive_site_edit_crash.txt](https://github.com/Subsurface-divelog/subsurface/files/1390395/dive_site_edit_crash.txt)

It shows s.th. about the dive list filter but Dirk thought that this also could come from a memory corruption that happened before.

tcanabrava, 2017-10-17:
I'll take a look today. basically "what happens next" is the population of the fields on the dive site edit with the data from the dive site, so we *could* fed wrong information there. I didn't understand your backtrace, to me it fails pointing me where the crash happened. Is this compiled in debug mode?

sfuchs79, 2017-10-17:
What comes to my mind right now:
Maybe this really has nothing to do with the country taxonomy changes. Maybe this was even already there in 4.6.4. Because it only happens for me with the debug build and maybe I never tested such scenarios before with a 4.6.4 (Windows MXE debug) build.

dirkhh, 2017-10-17:
The thing is that while I could reproduce the initial crash in Linux, this one I can no longer reproduce. So I think there were several problems piled on top of each other. I fixed one, you fixed at least one, but there is more left.

@tcanabrava confusingly Windows crash reports have the most relevant info on the bottom. Here's what's in the file that you really care about

```
AddrPC   Params
  ntdll.dll!_NtTerminateProcess@8
  ntdll.dll!RtlExitUserProcess
  KERNEL32.DLL!_ExitProcessImplementation@4
  msvcrt.dll!___crtExitProcess
  msvcrt.dll!__unlockexit
  msvcrt.dll!__exit
  msvcrt.dll!_abort
  Qt5Cored.dll!qt_message_fatal  [/home/stefan/Entwicklung/Subsurface/mxe/tmp-qtbase-i686-w64-mingw32.shared/qtbase-opensource-src-5.9.2/src/corelib/global/qlogging.cpp @ 1690]
  Qt5Cored.dll!qt_static_metacall  [/home/stefan/Entwicklung/Subsurface/mxe/tmp-qtbase-i686-w64-mingw32.shared/qtbase-opensource-src-5.9.2/src/corelib/.moc/debug/moc_qitemselectionmodel.cpp @ 270]
  Qt5Cored.dll!activate  [/home/stefan/Entwicklung/Subsurface/mxe/tmp-qtbase-i686-w64-mingw32.shared/qtbase-opensource-src-5.9.2/src/corelib/kernel/qobject.cpp @ 3766]
  Qt5Cored.dll!activate  [/home/stefan/Entwicklung/Subsurface/mxe/tmp-qtbase-i686-w64-mingw32.shared/qtbase-opensource-src-5.9.2/src/corelib/kernel/qobject.cpp @ 3628]
  subsurface.exe!myInvalidate  [/home/stefan/Entwicklung/Subsurface/subsurface/qt-models/filtermodels.cpp @ 384]
  subsurface.exe!startFilterDiveSite  [/home/stefan/Entwicklung/Subsurface/subsurface/qt-models/filtermodels.cpp @ 450]
  subsurface.exe!qt_static_metacall  [/home/stefan/Entwicklung/Subsurface/win32/subsurface/qt-models/subsurface_models_autogen/EWIEGA46WW/moc_filtermodels.cpp @ 462]
  Qt5Cored.dll!activate  [/home/stefan/Entwicklung/Subsurface/mxe/tmp-qtbase-i686-w64-mingw32.shared/qtbase-opensource-src-5.9.2/src/corelib/kernel/qobject.cpp @ 3766]
  Qt5Cored.dll!activate  [/home/stefan/Entwicklung/Subsurface/mxe/tmp-qtbase-i686-w64-mingw32.shared/qtbase-opensource-src-5.9.2/src/corelib/kernel/qobject.cpp @ 3628]
  subsurface.exe!showEvent  [/home/stefan/Entwicklung/Subsurface/subsurface/desktop-widgets/locationinformation.cpp @ 222]
  Qt5Widgetsd.dll!event  [/home/stefan/Entwicklung/Subsurface/mxe/tmp-qtbase-i686-w64-mingw32.shared/qtbase-opensource-src-5.9.2/src/widgets/kernel/qwidget.cpp @ 9036]

```
The crash is definitely in the filter code. I just don't understand that code well enough to figure out what's going wrong.

neolit123, 2017-10-17:
@sfuchs79 
can you please confirm that this bug can be reproduced with the latest windows daily build?
also, what version of Qt are you using?

dirkhh, 2017-10-17:
Looking at the stack trace, I'd say he's using Qt 5.9.2 :-)
`Qt5Cored.dll!qt_message_fatal  [/home/stefan/Entwicklung/Subsurface/mxe/tmp-qtbase-i686-w64-mingw32.shared/`**qtbase-opensource-src-5.9.2**`/src/corelib/global/qlogging.cpp @ 1690]
`

neolit123, 2017-10-17:
@dirkhh oops, clearly.

@sfuchs79 
> can you please confirm that this bug can be reproduced with the latest windows daily build?

sfuchs79, 2017-10-17:
No, can't be reproduced with latest windows daily and also not with my own release flavor build.
Only with my MXE debug build based on Qt5.9.2.
This is why Dirk and I already once decided to give up on this item. I mainly started again because I found (and fixed) this other small bug on this "pile of bugs" and first thought this is related - but no.
Meanwhile I again tend to say this could really come from a bug in the filter code any maybe is already there since a long time and I just never tested this test case with debug build.
It's up to you if we follow up on this or not. Again, it only happens for my debug build. On the other hand I personally believe in Murphy's law and fear that this could also hit other users once.

neolit123, 2017-10-17:
ok, i see. in that case it's best if @tcanabrava can see if he can find a potential problem. if all looks good to him, then it's best if we just leave this be for now.

it's true that the stack trace doesn't tell much, so it could be an odd Qt bug that only emerges in your debug build.

sfuchs79, 2017-12-30:
Let's be optimistic and assume that this is fixed by #989 
(I have to admit that I DIDN'T test it again...)

closed 2017-12-30 by sfuchs79