subsurface/subsurface #661: In planner minimum switch time applied twice after gas change

sfuchs79, 2017-10-10:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

Another small bug in the planner:
In some specific dive setups the "minumum gas switch time" is applied twice after a gas switch.
The setup is also described in the screenshot but here are the important factors:
- Two deco gases (e.g. EAN50%, Oxy), both enabled
- Option "Only switch at required stops" is set
- Minimum switch time set to >0, e.g. 5min
- Duration of dive and deco and ascend rate are set in a way that the lowest stop is 6m
- This means with the options set the EAN50% is not used for deco
--> Due to this setup for the gas switch to Oxy the minimum switch time is applied twice

[image]

This for sure comes from somewhere here:
https://github.com/Subsurface-divelog/subsurface/blob/master/core/planner.c#L972
One would need to check if we need to skip one cylinder completely (the EAN50%).
Unfortunately I was not able to fix this quickly :-(

### Operating system:

### Subsurface version:

Current master and 4.6.4

### Steps to reproduce:

### Current behavior:

### Expected behavior:

### Additional information:

### Mentions:

@atdotde @RickMWalsh

atdotde, 2017-10-16:
@sfuchs79 could you please check if my patch above addresses this problem?

sfuchs79, 2017-10-16:
Nope. Crashes when I try to test the specific test case in:
planner.c @ 980
This has also been the difficulty for me to understand where we need to skip the unnecessary gaschanges. Not at this point where you put it, I guess. But I also don't know better yet.

atdotde, 2017-10-18:
For me, this seems to work. Could you please confirm @sfuchs79 ?

sfuchs79, 2017-10-19:
Unfortunately no. Now the EAN50 is accidentally used for the 6m stop in my test case instead of the available Oxy.
May I propose s.th.: Let's try to fix this and the O2 breaks thing post 4.7. Both items seem to be somehow linked and pretty difficult with all the effects from all the possible options in the planner.

sfuchs79, 2017-11-08:
@atdotde 
I have the feeling that this issue either disappeared due to other improvements you did (o2 breaks?) or you fixed it by intention and I didn't recognize it.
Can you also confirm that the issue is gone?

sfuchs79, 2017-11-08:
Ah, no, I tricked myself. It didn't really disappear. I don't see it any more for my old example dive because we now have this special treatment for 100% oxy. If I change oxy to 99% I still see the issue.

PR #790 (2017-11-09): Planner don't add minimum gas switch time more than once

### Describe the pull request:

- [x] Bug fix
- [ ] Functional change
- [ ] New feature
- [ ] Documentation change
- [ ] Language translation

### Pull request long description:

Avoid adding the minimum gas switch time more than once even
if we skip some available deco gas.

Maybe stupid but hopefully efficient and safe way to fix this.
Closes #661 

### Changes made:

### Related issues:

### Additional information:

### Mentions:

@atdotde
closed 2017-11-09 by atdotde