subsurface/subsurface #664: Subsurface crash when importing dives from Vyper

audunlea, 2017-10-11:
### Describe the issue:

- [X] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:
Subsurface v.4.6.4 crash when selected dives are to be imported from Vyper computer. Connection to computer lists expected dive to import correctly, crash when click OK-button.
Error in /var/log/syslog: 
 [  973.200807] traps: subsurface[5788] general protection ip:7fd7c3896ee8 sp:7ffcf48e8e50 error:0 in libQt5Core.so.5.5.1[7fd7c364a000+4c5000]

### Operating system:
Ubuntu 16.04
Linux coffee2go 4.4.0-97-generic #120-Ubuntu SMP Tue Sep 19 17:28:18 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux
libqt5core5a is already the newest version (5.5.1+dfsg-16ubuntu7.5).
Vyper
Does the same happen on another OS?     Don't know

### Subsurface version:
What version of Subsurface are you running?   4.6.4
Does the same happen on another Subsurface version? unknown
Are you using official release, test build, or compiled yourself? official release

### Steps to reproduce:

1) Start Subsurface
2) Connect Vyper and set til TP - PC mode
3) Import from dive computer
4) OK-button
### Current behavior:

Subsurface crash and close completly

### Expected behavior:

Importing selected dive and stay open

### Additional information:

No more logs to provide - not any logs generated
### Mentions:

neolit123, 2017-10-11:
@audunlea 
thanks for the report.

we are going to need your help to get a better understanding of why this is happening, by running Subsurface under a debugger.

first see if GDB (the GNU debugger) is installed on Ubuntu by doing a:
`$ gdb -v`

if it's missing you can install it with:
`$ sudo apt-get install gdb`

steps:
1) find where the Subsurface binary is located and `cd` to that path
2) call `gdb ./subsurface`
/this should enter the gdb prompt/
3) type `r` + press `enter` to start subsurface and wait for the UI to load
4) reproduce the crash
5) type the following in gdb `bt 20` + press `enter` and post the output here

thanks

audunlea, 2017-11-03:
Since last time, I've got a new update of Subsurface installed. But the
same error still occores:

Starting download from  /dev/ttyUSB0
Starting the thread 0
INFO: dc_deveice_open error value of 0
[Thread 0x7fffc66f7700 (LWP 16486) exited]
Finishing the thread  dives downloaded 2
[Thread 0x7fff521c5700 (LWP 16502) exited]

Thread 1 "subsurface" received signal SIGSEGV, Segmentation fault.
0x00007ffff246eeec in ?? () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
(gdb) bt 20
#0  0x00007ffff246eeec in ?? () from
/usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#1  0x00007ffff246f1d4 in QSortFilterProxyModel::data(QModelIndex const&,
int) const ()
   from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#2  0x00007ffff39e519d in ?? () from
/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#3  0x00007fffe3ab23f6 in ?? () from
/usr/lib/x86_64-linux-gnu/libQt5XcbQpa.so.5
#4  0x00007fffe3ab2f49 in ?? () from
/usr/lib/x86_64-linux-gnu/libQt5XcbQpa.so.5
#5  0x00007fffe3ab83ac in ?? () from
/usr/lib/x86_64-linux-gnu/libQt5XcbQpa.so.5
#6  0x00007fffecb14abe in ?? () from
/usr/lib/x86_64-linux-gnu/libQt5DBus.so.5
#7  0x00007fffecb1626e in ?? () from
/usr/lib/x86_64-linux-gnu/libQt5DBus.so.5
#8  0x00007ffff24d7c01 in QObject::event(QEvent*) () from
/usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#9  0x00007ffff37a005c in QApplicationPrivate::notify_helper(QObject*,
QEvent*) ()
   from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#10 0x00007ffff37a5516 in QApplication::notify(QObject*, QEvent*) () from
/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#11 0x00007ffff24a838b in QCoreApplication::notifyInternal(QObject*,
QEvent*) () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#12 0x00007ffff24aa786 in
QCoreApplicationPrivate::sendPostedEvents(QObject*, int, QThreadData*) ()
   from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#13 0x00007ffff24fe3c3 in ?? () from
/usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#14 0x00007fffeea65197 in g_main_context_dispatch () from
/lib/x86_64-linux-gnu/libglib-2.0.so.0
#15 0x00007fffeea653f0 in ?? () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#16 0x00007fffeea6549c in g_main_context_iteration () from
/lib/x86_64-linux-gnu/libglib-2.0.so.0
#17 0x00007ffff24fe7cf in
QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>)
()
   from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#18 0x00007ffff24a5b4a in
QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) ()
   from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#19 0x00007ffff24adbec in QCoreApplication::exec() () from
/usr/lib/x86_64-linux-gnu/libQt5Core.so.5
(More stack frames follow...)
(gdb)

neolit123, 2017-11-03:
@audunlea 
can you please do a deeper backtrace in gdb - e.g:
`bt 30`

if needed, increase `30` with a number of steps so that it show some calls inside the subsurface module.
should look like:
```
#N <address> in <function-name> from <subsurface>
#N+1 <address> in <function-name> from <subsurface>
#N+2 <address> in <function-name> from <subsurface>
#N+3 <address> in <function-name> from <subsurface>
```

audunlea, 2017-11-03:
Hi, Lubomir

I seems like it doesn't give any more information when i type 'bt 30' or
'bt 50'

(gdb) bt 30
#0  0x00007ffff246eeec in ?? () from
/usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#1  0x00007ffff246f1d4 in QSortFilterProxyModel::data(QModelIndex const&,
int) const ()
   from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#2  0x00007ffff39e519d in ?? () from
/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#3  0x00007fffe3ab23f6 in ?? () from
/usr/lib/x86_64-linux-gnu/libQt5XcbQpa.so.5
#4  0x00007fffe3ab2f49 in ?? () from
/usr/lib/x86_64-linux-gnu/libQt5XcbQpa.so.5
#5  0x00007fffe3ab83ac in ?? () from
/usr/lib/x86_64-linux-gnu/libQt5XcbQpa.so.5
#6  0x00007fffecb14abe in ?? () from
/usr/lib/x86_64-linux-gnu/libQt5DBus.so.5
#7  0x00007fffecb1626e in ?? () from
/usr/lib/x86_64-linux-gnu/libQt5DBus.so.5
#8  0x00007ffff24d7c01 in QObject::event(QEvent*) () from
/usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#9  0x00007ffff37a005c in QApplicationPrivate::notify_helper(QObject*,
QEvent*) ()
   from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#10 0x00007ffff37a5516 in QApplication::notify(QObject*, QEvent*) () from
/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#11 0x00007ffff24a838b in QCoreApplication::notifyInternal(QObject*,
QEvent*) () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#12 0x00007ffff24aa786 in
QCoreApplicationPrivate::sendPostedEvents(QObject*, int, QThreadData*) ()
   from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#13 0x00007ffff24fe3c3 in ?? () from
/usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#14 0x00007fffeea65197 in g_main_context_dispatch () from
/lib/x86_64-linux-gnu/libglib-2.0.so.0
#15 0x00007fffeea653f0 in ?? () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#16 0x00007fffeea6549c in g_main_context_iteration () from
/lib/x86_64-linux-gnu/libglib-2.0.so.0
#17 0x00007ffff24fe7cf in
QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>)
()
   from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#18 0x00007ffff24a5b4a in
QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) ()
   from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#19 0x00007ffff24adbec in QCoreApplication::exec() () from
/usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#20 0x0000000000512264 in main ()
(gdb)

Seems like I've got the lib file:

root@coffee2go:~# ls -ltr  /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
lrwxrwxrwx 1 root root 19 mai   12 20:47
/usr/lib/x86_64-linux-gnu/libQt5Core.so.5 -> libQt5Core.so.5.5.1
root@coffee2go:~# ls -ltr  /usr/lib/x86_64-linux-gnu/libQt5Core.so.5.5
libQt5Core.so.5.5    libQt5Core.so.5.5.1
root@coffee2go:~# ls -ltr  /usr/lib/x86_64-linux-gnu/libQt5Core.so.5.5.1
-rw-r--r-- 1 root root 5052920 mai   12 20:49
/usr/lib/x86_64-linux-gnu/libQt5Core.so.5.5.1

root@coffee2go:~# ldd /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
        linux-vdso.so.1 =>  (0x00007fff349f8000)
        libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0
(0x00007fd29b2d5000)
        libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1 (0x00007fd29b0bb000)
[... 229 lines cut ...]
        libkrb5.so.26 => /usr/lib/x86_64-linux-gnu/libkrb5.so.26
(0x00007f71cda8f000)
        libasn1.so.8 => /usr/lib/x86_64-linux-gnu/libasn1.so.8
(0x00007f71cd7ed000)
        libhcrypto.so.4 => /usr/lib/x86_64-linux-gnu/libhcrypto.so.4
(0x00007f71cd5ba000)
        libroken.so.18 => /usr/lib/x86_64-linux-gnu/libroken.so.18
(0x00007f71cd3a4000)
        libXau.so.6 => /usr/lib/x86_64-linux-gnu/libXau.so.6
(0x00007f71cd1a0000)
        libXdmcp.so.6 => /usr/lib/x86_64-linux-gnu/libXdmcp.so.6
(0x00007f71ccf9a000)
        libsystemd.so.0 => /lib/x86_64-linux-gnu/libsystemd.so.0
(0x00007f71ccf15000)
        libwind.so.0 => /usr/lib/x86_64-linux-gnu/libwind.so.0
(0x00007f71cccec000)
        libheimbase.so.1 => /usr/lib/x86_64-linux-gnu/libheimbase.so.1
(0x00007f71ccadd000)
        libhx509.so.5 => /usr/lib/x86_64-linux-gnu/libhx509.so.5
(0x00007f71cc892000)
        libcrypt.so.1 => /lib/x86_64-linux-gnu/libcrypt.so.1
(0x00007f71cc65a000)
        libselinux.so.1 => /lib/x86_64-linux-gnu/libselinux.so.1
(0x00007f71cc438000)
root@coffee2go:~#

neolit123, 2017-11-03:
you can try this:
- `cd` to the folder where the `subsurface` binary is
- type:
`addr2line -e ./subsurface 0x512264`
(`addr2line` is part of the `binutils` package)

see if it prints out a `file:line`, but even that won't help us much as it's crashing immediately in `QCoreApplication::exec()`. this tells me that something is wrong in your setup. maybe it's the installed Qt libraries.

by the way are you running subsurface as `root`? if yes, can you try a normal user?

audunlea, 2017-11-03:
Hi

Not much help I think:

audunle@coffee2go:/usr/bin$ addr2line -e ./subsurface 0x512264
??:?
audunle@coffee2go:/usr/bin$

I'm running Subsurface as my normal user, audunle. Used root just for
debugging

But Your tip helped. Reinstall Qt as follow fixed the problem!! :-)

root@coffee2go:~# apt-get install --reinstall libqt5core5a

Subsurface didn't crash after download of dives, and I were able to save
the xml-file and continue adding information

Thanks a lot!
--
Regards
Audun

neolit123, 2017-11-03:
glad it worked.
it was probably a library with a different version or it was somehow corrupted.

closed 2017-11-03 by neolit123