subsurface/subsurface #672: Subsurface 4.6.4 for windows crash clicking "Import GPS data from Subsurface web service"

estresao, 2017-10-16:
### Describe the issue:
Using Subsurface 4.6.4 for Windows it crashes when I click on the menu option  "Import GPS data from Subsurface web service"

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:
Subsurface 4.6.4 for Windows

HI everyone,

I'm trying to use for 1st time GPS data captured using Subsurface Mobile App for Android in Subsurface. I'm using Subsurface 4.6.4 on Windows 7 Enterprise SP 1 64-bits.

When I click the menu option "Import GPS data from Subsurface web service" Subsurface crash and a Windows pop up window comes out informing "subsurface.exe has sopped working"

Windows report file:

<?xml version="1.0" encoding="UTF-16"?>
<WERReportMetadata>
    <OSVersionInformation>
        <WindowsNTVersion>6.1</WindowsNTVersion>
        <Build>7601 Service Pack 1</Build>
        <Product>(0x4): Windows 7 Enterprise</Product>
        <Edition>Enterprise</Edition>
        <BuildString>7601.23889.amd64fre.win7sp1_ldr.170810-1615</BuildString>
        <Revision>1130</Revision>
        <Flavor>Multiprocessor Free</Flavor>
        <Architecture>X64</Architecture>
        <LCID>1033</LCID>
    </OSVersionInformation>
    <ProblemSignatures>
        <EventType>APPCRASH</EventType>
        <Parameter0>subsurface.exe</Parameter0>
        <Parameter1>0.0.0.0</Parameter1>
        <Parameter2>5900de99</Parameter2>
        <Parameter3>subsurface.exe</Parameter3>
        <Parameter4>0.0.0.0</Parameter4>
        <Parameter5>5900de99</Parameter5>
        <Parameter6>c0000005</Parameter6>
        <Parameter7>001cca15</Parameter7>
    </ProblemSignatures>
    <DynamicSignatures>
        <Parameter1>6.1.7601.2.1.0.256.4</Parameter1>
        <Parameter2>2057</Parameter2>
        <Parameter22>0a9e</Parameter22>
        <Parameter23>0a9e372d3b4ad19135b953a78882e789</Parameter23>
        <Parameter24>0a9e</Parameter24>
        <Parameter25>0a9e372d3b4ad19135b953a78882e789</Parameter25>
    </DynamicSignatures>
    <SystemInformation>
        <MID>DF60B558-E8FD-4CF7-8D13-B8A391BADB7F</MID>
        <SystemManufacturer>LENOVO</SystemManufacturer>
        <SystemProductName>20BG0044UK</SystemProductName>
        <BIOSVersion>GNET81WW (2.29 )</BIOSVersion>
    </SystemInformation>
</WERReportMetadata>

### Operating system:

Windows

Windows 7 Enterprise with SP 1 64-bits

Laptop:
        System Manufacturer: LENOVO
        System Product Name: 20BG0044UK
        BIOSVersion: GNET81WW (2.29 

Haven't try

### Subsurface version:

4.6.4

haven't try

Yes

### Steps to reproduce:

1) Click menu option "Import GPS data from Subsurface web service"

### Current behavior:

Subsurface crash and Windows pops up a message with says "subsurface.exe has sopped working"

### Expected behavior:

GPS data from the cloud imported

### Additional information:

### Mentions:

sfuchs79, 2017-10-17:
This was already reported before here: #367 
And hopefully fixed here: 38af1f2ab93f2ed20552b0869a02844778138739

Could you please try again with the latest test version from here:
http://subsurface-divelog.org/downloads/test/

estresao, 2017-10-17:
Thanks Stefan,

Using the test version it "works", I mean, now I get the pop up windows but
when click on download I get a "server replied: INTERNAL SERVER ERROR".

I guess that's something different to the initial bug.

Regards,

Jose A. Zapata

dirkhh, 2017-10-17:
On Tue, Oct 17, 2017 at 10:42:27AM +0000, estresao wrote:
> Thanks Stefan,
> 
> Using the test version it "works", I mean, now I get the pop up windows but
> when click on download I get a "server replied: INTERNAL SERVER ERROR".
> 
> I guess that's something different to the initial bug.

It means that something is wrong with the GPS data that are stored on the
server - not very enlightening, I know. Are there any other messages?
If you run subsurface.exe from a cmd.exe window, it will print some more
diagnostics - anything in there?

estresao, 2017-10-17:
I guess I don't know how to do that properly (1st time using it).

By launching subsurface .exe from the command line I've got no error when click download button.

The status message after click is the following: "Download error: Error transferring http://api.hohndel.org/api/dive/get/?login=<MYUSER>@GMAIL.COM - server replied: INTERNAL SERVER ERROR"

I've copied the REST call and pasted in a browser and I get the following response: {"download": "notok", "error": "Login invalid"}

So looks like Subsurface pops up a internal server error when the login is not valid?

I supposed my user ID is the user I've set to store data in the cloud... is not? In Subsurface's Preferences - Network options I've got no default user ID in "Subsurface user ID" parameter because I din't find it in Subsurface-Mobile 2.0(4.6.4.996) for Android, so I'm assuming it's my cloud user (also tried using my PIN).

Sorry if this reply got out of the initial topic.

Jose A. Zapata

estresao, 2017-10-17:
I've found how to get the user id for the web service. Haven't seen it in the documentation, but found the page "http://api.hohndel.org/login/" and it gave me the ID to be set in preferences.

Thanks a lot!

Jose A. Zapata

dirkhh, 2017-10-17:
Clearly that's a terrible error message, but yes, you need the insanely long hex string instead of your email address (the GPS webservice long predates the Subsurface Cloud and didn't use the email address - in hindsight a poor decision on our part).
I'm glad this is resolved, sorry for making it so confusing! I'll close this issue.

closed 2017-10-17 by dirkhh