subsurface/subsurface #674: When downloading in Subsurface-mobile, respect trips

dirkhh, 2017-10-17:
### Describe the issue:

- [ ] Bug
- [ ] Change request
- [x] New feature request
- [ ] Discussion request

### Issue long description:

If you have auto-trips on in Subsurface and use both Subsurface and Subsurface-mobile during a dive trip, whenever you download from the dive computer with Subsurface-mobile, dives aren't added to the active trips but are added outside of the trip...

### Operating system:

OS independent (OK, download on Subsurface-mobile clearly means Android)
### Subsurface version:

Latest master on both
### Steps to reproduce:

### Current behavior:

### Expected behavior:

I think we should respect the autogroup setting that's already in git storage

### Additional information:

### Mentions:

janiversen, 2018-06-22:
Isn't this just a variant of #446 ?

dirkhh, 2018-06-22:
I think so. Should be ready to add.

janiversen, 2018-06-23:
I looked at it, and it is something I would like to to try and make. I am however confused dome issues are assigned to a person (I saw one I want fixed8, in my “world” it is rude to interfere with those, so please what are the rules in ssrf?  ps. as you know I am diving heavely now so please allow me a couple of weeks to complete this.

dirkhh, 2018-06-23:
If it's assigned to someone, I'd ask that person if they are working on it before I started

/D