subsurface/subsurface #675: Filter list not updated when dive site name is changed via "edit dive site" dialog

sfuchs79, 2017-10-17:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

Please assume a long logfile with many dives. I activate the dive filter. Now I for example identify some entries in the filter lists for buddy names, dive site names, tags, dive suit, whatever which I want to correct because there is a typo or similar.
If I do this directly in the main tab "Notes" and apply the change(s), immediately the filter lists are updated correctly. If e.g. an entry is removed because a typo was corrected the wrong entry disappears.
Now to the bug:
If I do change the dive site name of a dive not in the main tab but by clicking the "edit dive site" button and then edit the name there, the change is not reflected in the filter lists.

### Operating system:

Windows 10

### Subsurface version:

4.6.4.1013 which is latest master plus some WIP items.

### Steps to reproduce:

### Current behavior:

[image]

### Expected behavior:

Update filter lists also after a change of the dive site name in the "edit dive site" dialog is done and applied.

### Additional information:

### Mentions:

dirkhh, 2017-10-17:
This is the third subtle little filter bug. @tcanabrava promised to look at the other two, let's assign this one to him as well :-)
@tcanabrava - this is fine to do after 4.7 - I'd love to see the crash (the one we discussed an hour ago) to be addressed if we can... the others we can do in 4.7.x

bstoeger, 2017-11-25:
@sfuchs79: Could you perhaps test PR #857, to see if it behaves like you would expect? Thank you.

closed 2017-11-25 by dirkhh

sfuchs79, 2017-11-25:
@bstoeger 
This is the edit dive site dialog i'm talking about:
[image]

bstoeger, 2017-11-25:
Ok - I see what you mean. Please try #863, this should change the behavior.

Unfortunately, with the changes in #857, this means that the old selection in the dive list is lost. :( I'm also not sure how to fix it - somehow the filter would have to know which entry was renamed...

commit 739b2742 (2017-11-25): LocationFilter: use equality instead of substring comparison

The location filter used a substring comparison, which had
the side effect that dives with the location "Test 1" were
shown when filtering for "Test 12".

Moreover, avoid a deep copy of the location list. This is
done by looping over one item less than rowCount() instead
of removing the last item. While doing this, remove an
unnecessary if-statement.

This commit partially fixes #675.
PR #863 (2017-11-25): Refresh display if a dive site was edited
Calls MainWindow::refreshDisplay() if a dive site was edited so that
the filter lists are updated. The old call to updateDiveInfo() was
removed because it is call implicitly in MainWindow::refreshDisplay()
anyway.

Fixes #675.

Requested-by: Stefan Fuchs <sfuchs@gmx.de>

### Describe the pull request:

- [x] Bug fix
- [ ] Functional change
- [ ] New feature
- [ ] Code cleanup
- [ ] Build system change
- [ ] Documentation change
- [ ] Language translation

### Pull request long description:

Refresh filter lists if a dive site was edited. Unfortunately  [... cut]
commit 01d96108 (2017-11-25): Refresh display if a dive site was edited

Calls MainWindow::refreshDisplay() if a dive site was edited so that
the filter lists are updated. The old call to updateDiveInfo() was
removed because it is call implicitly in MainWindow::refreshDisplay()
anyway.

Fixes #675.

Requested-by: Stefan Fuchs <sfuchs@gmx.de>