subsurface/subsurface #687: Make mention bot not suggest author of PR as reviewer

atdotde, 2017-10-18:
### Describe the issue:

- [ ] Bug
- [x] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

It makes little sense to review ones own changes.

### Operating system:

### Subsurface version:

### Steps to reproduce:

### Current behavior:

### Expected behavior:

Exclude the author of PR from suggested reviewers.

### Additional information:

### Mentions:

dirkhh, 2017-10-18:
I think this is a change in mention-bot, not in Subsurface...

atdotde, 2017-10-18:
Actually, my understanding is that this is a config option, so it should be set in subsurface's config file for mention bot.

neolit123, 2017-10-29:
@atdotde hmm, is this possible?
i didn't see a way to exclude the mention if the PR is from the same user.

neolit123, 2017-11-15:
closing this as i think it's not possible to configure it as requested.
also, the mention bot is not even working most of the time...

closed 2017-11-15 by neolit123