subsurface/subsurface #692: Instructs "samples" and "plot_data" way to handle availability of information

sfuchs79, 2017-10-19:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

[post 4.7!!!]
Today in the structs "samples" and "plot_data" for some member variables we do not have a good strategy to distinguish between no data available and data available with a specific value for some special cases.
One example is (compass) bearing:
For bearing a value of '0' is a valid information. I tend to remember that 0° is straight north ;-)
But today we use the '0' as a indicator for "no info". We better should use s.th. like '-1' for no info. 
libdc/the DCs seem to do it like this - please correct me!
But if we change this to '-1' for no info we need to change also the default value for new samples and for new plot_info to -1 to have everything consistent.
I did already change the code to do it for the samples and this works pretty well but would need to do it also for plot_info.
And bearing is not the only example.
There is e.g. also NDL where '0min' is a valid information. You can see this today that there is a small issue when you move the mouse to the start of a profile of a dive from a DC. Then you get a NDL of 0min printed in the info box. This changes as soon as the NDL in the plot_info is filled with the correct info from the first available DC sample. Also here we should init NDL value in plot_data with s.th. like '-1' for no NDL info available and then not print this info based on the '-1' data.

I write this because I'm looking for feedback and proposals for a good solution. My main questions are:
- Is this correct what I'm saying here?
If yes:
- What would be the right strategy to set default values != 0 for member values of samples?
- What would be the right strategy to set default values != 0 for member values of plot_info?

### Operating system:

### Subsurface version:

### Steps to reproduce:

### Current behavior:

### Expected behavior:

### Additional information:

### Mentions:

neolit123, 2017-10-31:
@atdotde @dirkhh 
any comments about the suggestions @sfuchs79 is making for the `samples` and `plot_data` structs?
maybe that's a question for Linus?

sfuchs79, 2017-10-31:
@neolit123 
Dirk, Linus and I already had some good discussions about this and there is a related PR in WIP status: #717 
But I still identified some hurdles and will need some more time and maybe also help/advice to finalize this. I'm very sure this won't be ready for 4.7.2 and it is also far away from being important for 4.7.2
So if nobody complains let's keep it open and I will continue my work as soon as I have some more time.
BTW: After 4 days of diving and also using Subsurface and discussing Subsurface functionality with two friends of mine I collected a few (small) items I will even do with higher prio after coming home :-)

sfuchs79, 2017-11-14:
Closed via #717

closed 2017-11-14 by sfuchs79