subsurface/subsurface #696: Autogroup feature on for logfile w/o autogroup after other file with autogroup

sfuchs79, 2017-10-19:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

Question:
Why do we save the autogroup config setting only for autogroup=1 into the XML?
https://github.com/Subsurface-divelog/subsurface/blob/master/core/save-xml.c#L563

This means that if I load a file with autogroup on
--> Autogroup will be on
If I then load a file with autogroup off (no entry in XML)
--> Autogroup will stay on

Item is of sort:
Don't like this.
Can fix it :-)
But need to know if there is a rationale for today's implementation.

### Operating system:

### Subsurface version:

### Steps to reproduce:

### Current behavior:

### Expected behavior:

### Additional information:

### Mentions:

dirkhh, 2017-10-20:
this is where our 'is it the default - otherwise don't write it' logic may be hurting us. And autogroup is especially tricky because we use to store it in the local settings.
And of course it's ignored by Subsurface-mobile.
Definitely something to investigate more deeply in my infinite spare time.

dirkhh, 2017-10-20:
I'll also need to talk to @torvalds about this to make sure I'm not forgetting why we did this in the first place.

closed 2017-12-19 by sfuchs79