subsurface/subsurface #701: metric units being displayed despite preferences set to Imperial

PapaMarc, 2017-10-20:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:
units displaying improperly as metric in one case, even though units have been set to Imperial in all cases.
eg. Equipment(tab).Cylinder.Size displays in liters even though in preferences units selection is Cubic Feet

In other areas i've reviewed, it appears units ARE correctly displayed with Imperial units. In all fairness, i'm new to SubSurface, but like what i see so far!

### Operating system:
OS: Win10
Version 1703 (OS Build 15063.674)
HP/Compaq Pavilion dv6
Haven't tried another OS-- separate bug to be filed. eg. have tried loading Android client, but selection for 'No Cloud' doesn't work and is prompting for valid email addr. I'd prefer at least initially, to 'try' with my own local file, and perhaps even more 'manually' setup OneDrive cloud for sharing of a log file (i understand in that case i'll need to be careful not to leave a copy open in one, while working on another-- and have no expectation of conflict detection nor resolution).

### Subsurface version:
4.6.4 for Windows/PC
haven't tried a different version
release version
Git hash NA

### Steps to reproduce:
Load SubSuface 4.6.4. Set preferences to Imperial Units. Load a set of dives (which i imported from a WLog .log file (WLog's .add file for extra long description was ignored). 

Look at cylinder size row in the Equipment tab.  
Expected: cyl size in cu ft. (tank used was an old steel 72)
Actual: cyl size appears as 10.3 l

### Additional information:

Will also note-- as a preference i have set my preferred tank to HP80, though the dives i'm looking at were imported from a dataTrak log, and in the case above it represents an old steel 72 tank.

my first bug report in this db; apologies if i didn't quite get all the conventions right. cheers.

### Mentions:

dirkhh, 2017-10-20:
imperial units for tanks are weird and illogical. Depending on how the data was entered and where it came from, we may not be able to give you the imperial size, as that requires a wet volume (that's the 10.3l), but also a working pressure. And unless the working pressure is provided (and I am guessing it is not, and I'm guessing it should be 3000psi) there is nothing we can do.

dirkhh, 2017-10-20:
Oh, and it's Subsurface - not SubSurface

closed 2017-10-20 by dirkhh

PapaMarc, 2017-10-20:
data was imported from a .log file. ~80% of ~160 dives were downloaded from an AirZNitrox computer so start and ending PSI is included. I do have tank size entered in WLog in cuft, and it does appear effectively in subsurface, albeit appearing as liters not cuft.
BUT you are correct-- there was no data entered for working pressure. When i enter it manually this resolves the issue, and then it appears in cuft, per selected preferences.  Thanks so much!!
eg. nothing further needs to be done, it's working correctly in the code.  It just has a dependency on working pressure in the given entry, in order to display imperial. 
 
in that example of the steel 72, yes that's more/less correct...
DOT 3AA 2250 indicates a low pressure 72(cf). The 3rd row stamp reads 2 <NSymbol> 80+ meaning it was manufactured by Norris in Feb 1980; the "+" indicates it was rated for 10% overfill.
@ rated pressure of 2250psi it holds 64.7cf
@ its rated 10% overfill or 2475psi it holds 71.2cf.
@ 2152psi, ~5% underfilled it holds 61.8cf