subsurface/subsurface #703: Android client 'no cloud mode' not working

PapaMarc, 2017-10-20:
### Describe the issue:

- [x ] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:
Loaded Android client for the first time. Had moved a .ssrf file to my phone's storage. On opening client, i'm prompted for credentials. If i select the menu via the icon and select Dive List, it brings me back to the credentials prompt. The 'No Cloud Mode' button does not appear to work. I am unable to discover where i can select (or provide a path to) the ssrf file i have placed on the device.

### Operating system:
Android 5.0.2 Lollipop
Device: Xiaomi Redmi 红米Note 2 Helio X10 64bit octo-core 
Haven't tried on another platform.

### Subsurface version:
Version 2.0 (4.6.4.966)

### Steps to reproduce:
Loaded Android client for the first time. Had moved a .ssrf file to my phone's storage. On opening SubSurface for the first time, i'm presented the login/register for a Cloud Account. On selecting the 'No Cloud Mode' button to the right of the 'sign-in or register' button, i receive an error (in red text) which reads "Please enter valid cloud credentials" and am not able to proceed.

### Current behavior:
As noted, On selecting the 'No Cloud Mode' button to the right of the 'sign-in or register' button, i receive an error (in red text) which reads "Please enter valid cloud credentials" and am not able to proceed.  

### Expected behavior:
I expect that the creds dialog is dismissed, and i'm able to navigate to and open the ssrf file i have manually copied to the device.

### Additional information:

As i also noted in issue#701, I'd prefer at least initially, to 'try' with my own local file, and perhaps even more 'manually' setup OneDrive cloud for sharing of a log file, without knowing more about SubSurface's cloud storage platform, and details about its security, etc.  I understand in that case i'll need to be careful not to leave a copy open in one, while working on another-- and have no expectation of conflict detection nor resolution.

### Mentions:

janmulder, 2017-10-20:
The no cloud issue is already fixed in PR #676 

Further, it is not possible to open an ssrf file with the mobile app (and this functionality is not intended to be added), so that is why no can not find any dialog to open an ssrf file (or any other file). The data is stored on the local device as a git repository.

PapaMarc, 2017-10-20:
apologies for some ignorant questions here on your tracking/build/publish process, and file format(s) but to the extent you or others will indulge me in whole or in part...
1) i gather #676 isn't yet built into the current downloadable build from Google Play store (given the date in the store is 10/11/17 and the date of this change appears to be ~3days ago or 11/17/17).
2) is there a way of seeing from that PR #676 entry when that will appear/be targeted for a published release (on Google Play Store) build? (or, perhaps alternatively see what fixes are in a given build that is published on GPlayStore)?
3) is ssrf equivalent to xlm? eg. i imported one set of dives from WLog .log/.add format and they have an xml extension on WinPC, whereas i groomed/imported another set of dives from .csv and they have an ssrf format. I just accepted the defaults. I assumed they're all XLM (eg. SubSurface native format), but perhaps not? Would i have better luck on mobile w/the file w/the .xml extension? or is 'Git repository' yet another format?  eg. if neither is consistent w/the default Win/PC format, i'd suggest this manual 'no cloud mode' is more of a developer feature.
4) in reading more about cloud mode in other entries, and your comments... it's not clear to me if i try cloud mode, how to switch between multiple logs. or perhaps more importantly how to DELETE logs, etc. Given i'm not really sure with where this stuff sits, etc... i'd like to try/play with it, but would also want to know i could remove it entirely from the cloud when i'm done. Sounds like there may be issues w/cleanup, and perhaps reverting to using cloud mode later after selecting manual mode (if i end up on PC side, merging a few logs and using that as a default single logbook to be used for cloud mode, etc). In general, i'd really like to 'see' where my data is stored, what it looks like, and personally be the owner of the repository; just saying, and presume this group can appreciate given your born of linux origins power-users/devs v. being iphone-esk everything just magically works users. If there are better docs on how it works (other than the code which i understand is the ultimate spec) perhaps you could point me to that, though understand if it doesn't exist.
5) wrt to resolving/closing this, if it really is a 'won't fix'... is that my responsibility to do so, etc.  What is the convention of how you handle this in the db? Does the dev Resolve as' Won't Fix' (or 'By Design') and do i 'close' it as the initial filer, etc?
Many thanks!  Cheers

janmulder, 2017-10-20:
In answer:
1. Correct. The mentioned PR is in master, but not yet in a released version on Playstore. http://subsurface-divelog.org/downloads/test/ has a slightly newer version (-978) but I do not think the mentioned PR al already in there. So you will have to wait for a new test build, or a new released version on Playstore, if that is what you prefer.

2. As being an open source project, all code and commits are visible here. And any new release on Playstore have a short bullet list with the highlights. But there is no time schedule for a next mobile release.

3. ssrf is an xml style format. And, no, you cannot import any file in the mobile app, so also no xml, csv. A git repository is something entirely different. On https://git-scm.com/ you can read all about Git.

4. Let me focus on "See where your data is". In no-cloud mode it is on your device and nowhere else. Remove the app, and all is gone. In cloud mode it is on your device and on the projects cloud server. With device I mean every device you sync to the cloud sever. The project cloud server is maintained by Dirk, and a simple email to him, and he will remove your account. And we do not care who you dive with or where.

5. Not being able to load a ssrf or xml or anything from the mobile app is "by design". The no cloud failure is fixed. So you may call it as you like. You may close this, but obviously, I can close it too.

dirkhh, 2017-10-20:
This is an open source project. We do what we do because we love software. And some (but not all) of us love diving (I'm writing this from Bonaire during a break between dives).

If you don't trust us, don't use our software. It's that simple.
The cloud storage sits on a server on AWS, I pay for it out of my own pocket because I'm stupid and I like for people to have fun using the software that I maintain. At some level it annoys me that I constantly have to defend the security / privacy of the data on that server. Again, if you don't trust me. DO NOT RUN A BINARY THAT I BUILD ON YOUR DEVICE!!!! Seriously.
I don't really know what else to say to this. Our software as well as the cloud storage cost you no money, there are no ads, we don't use, sell, or even access your data. Even when people ask me to help them debug problems, I don't access their data without explicit permission. I really, really couldn't care less about your data. About where you dive, when you dive, how often you dive, with whom you dive. But you have no reason to believe me, so just don't use the software and write your own.
Whatever.
I'll update Subsurface-mobile for Android in the next few days which fill fix the bug with the "no cloud button" not always working.

closed 2017-10-20 by dirkhh

atdotde, 2017-10-20:
Let me add to Dirk's explanation (that in short if you worry about us/Dirk having your data on our/his hard drive you should not execute binary programs from us/him because those you do all kinds of things with what is on your device including sending that anywhere): 

It's a matter of maintaining our software and making sure our users (most of which are not computer nerds) have a good and not a frustrating experience with our software. It is a fact that people are not used to dealing with "files" on mobile devices (it is possible to some degree on Android with external memory cards) but at least on iOS, the internal files system is completely hidden from the users and the amount of exchange of data between applications in their respective sandboxes is restricted for security reasons to a degree that a filesystem in the traditional meaning does not really make sense.

So, in order to bring files (i.e. your dive log files) to your mobile device, you have to go some roundabout way. Either you use an external memory card or you use something like Dropbox or Owncloud or even less popular solutions and we are getting requests of users that subsurface should support their favourite way of data transfer. 

Even if we invested the time to implement that, those solutions (since in the end they tend to be all hacks) are very error prone and if you are not exactly knowing what you are doing it is very likely that things won't really work (because some file is not found or properly synced or whatever). And then the users will come to us (in much greater numbers than those that currently request those features) with comments like "your shitty piece of software is totally broken because I cannot get the file transfer to work or the doge ate my dive log".

Thus we decided not to support those hacks but rather provide one sync service (and we should have called it sync service rather than cloud since that has a bad name) that we can be pretty sure we can maintain (since we have control over the whole chain desktop-server-mobile device) plus, since it is internally simply git, is very good at handling syncing conflicts and not losing data.

And since it is git, at least on the desktop (where you can actually access your files), you will find your git repository with all your dives on your hard drive. Just run subsurface with command line option -v and it will tell you where it keeps it.

Again, since is versioning, you cannot really delete the log directly (but you could delete all your dives and save that but there would still be the memory of what was there). This is good since it saves us again from users' frustration over losing their dive data.

But if you really want your cloud log removed, I am sure by sending an email to Dirk this is done in a matter of minutes. And, again, if you don't trust him, it makes zero sense to run software provided by him!

PapaMarc, 2017-10-20:
Didn't mean to cause a flame/storm. I actually wasn't concerned with you guys. I AM running your code on my PC and my phone. Sorry for any misunderstanding. In short, i suspect we'd all agree, there are a few malicious attackers out there even if none of us know who they are, and even if we further assume none of us are the attackers.
First let me say, i love sw and diving too, and my first impressions of what you guys are doing is VERY positive. And i greatly appreciate the vibrancy of the community and the quick responses i got. I think it's really cool what you guys are doing.
Perhaps i wasn't clear (was trying to keep things brief, and i had a number of questions as it was, but that's always a risk)-- you're misinterpreting the security related inquries-- sorry for that, and didn't intend to offend in any way. 
For starters, i'm new to your project, and didn't know where the 'cloud' feature dataset was stored, anything about that platform for that storage (now i know, along w/your code and bugdb it's Git hosted on AWS), nor who had access (now i know that's you dirkhh) -- though now i understand a bit more about that based on janmulder's helpful/insightful response.
To be clear, the potential security surface area (with that data in the cloud) go faarrr beyond you guys who deeply invest selflessly in creating/sharing this seemingly killer platform-- but rather extend to any interested attacker, anywhere in the world, who might be able to attack/access/compromise my, your or any other of your user's information. And since i was dredging up data on hundreds of dives that i hadn't looked at in a couple decades, and haven't yet gone through all of it, and don't recall what i wrote in the free form text descriptions... wasn't sure i wanted to potentially 'publish' it. Personally, i don't have a FB page, as i choose to forgo those benefits while maintaining some of my privacy.  eg. From a big picture, apart from subsurface, there have been a lot of high profile hacks in the past year+ ... Equifax being the most recent high profile one in past weeks, but here's a nice visualization of many others: http://www.informationisbeautiful.net/visualizations/worlds-biggest-data-breaches-hacks/ . And arguably you can call me paranoid, but i spent the past decade working on the core antivirus, in Windows/Hotmail/OneDrive/Azure, etc, where i saw many other types of code based attacks in gory detail, which impacted hundreds of millions. So i merely was approaching this from a desire to understand some of the underpinnings of the infra used, and its potential surface area, (including potentially/initially trying a more private cloud approach, where i'm admin and don't need to know someone, let alone impose on someone to cleanup the mess i make while i play around with it-- since i didn't see mention of a 'delete' feature in addition to the add), etc.
Again, sorry if any offense was taken; i'll emphatically state, NONE/ZERO was intended.  I think what you guys are doing is REALLY cool. And given Git's and AWS's track record (though i didn't know that was what was being used before i asked), as well as use for source maintenance by some major co's including Google/FB/MS/Twitter, etc it further reinforces more than reasonable design choices and risk mitigation from a platform standpoint--- you're certainly on a solid foundation. Albeit, again, before i asked, i didn't know this...
@dirkhh -- to the extent various people ask about security of the cloud functionality, and @atdotde to the extent you're increasingly receiving requests to support manual/private use of alternate cloud storage platforms to point at a file in replicated storage  -- i would offer that perhaps rather than infer any assault by the query-er to your integrity or intentions, or the project's integrity... hope instead that there may be a groundswell of security conscious and risk aware users, living in an increasingly cloud based world, who are becoming ever more cautious (and maybe less likely to get phished, or otherwise exploited).  There are probably numerous ways to approach and deal with this, even without writing code, and if you're interested i'd be happy to discuss further (not sure this is the best forum, again as i'm not sure w/convention of your project and how it runs), but among other things, some discussion (eg. FAQ) of how that feature works with treatment from a security perspective, might assist in mitigating concerns (or leading to the next level of depth of q's), etc. 
To be clear, my concern in asking such questions, coming from the background i do , wasn't actually that you'd be concerned that i was disparaging you guys in any way, but rather w/out our having incrementally built more trust between us, that you might be concerned i was interested in attacking your entire cloud database.   That of course is the other issue w/potentially over documenting the admin side of what you do.  But some treatment of security seems reasonable, as any reliance solely on security-through-obscurity, by not sharing any details of where it's hosted (wherein it's an open source project anyway, so any attacker could simply look through the source, and see as much wrt where/how it's saved) isn't something to rely on; and clearly you guys were happy to share some details per @janmulder 's response to my intial queries.  

OK-- that's enough for now. Again, sorry for any misunderstanding. I like what you guys are doing. Really i do. Maybe i can reintroduce myself.  ;-)  Hi-- I'm Marc.