subsurface/subsurface #705: In fullscreen mode main menu bar menus don't work

sfuchs79, 2017-10-20:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

In fullscreen mode main menu bar menus don't work.

### Operating system:

Windows 10

### Subsurface version:

4.6.4.1xxx

### Steps to reproduce:

Press F11 to enter fullscreen mode. Try to open a menu.

### Current behavior:

### Expected behavior:

### Additional information:

### Mentions:

dirkhh, 2017-10-21:
I didn't know there was a Windows full screen mode.
In other words... no idea. :-(

sfuchs79, 2017-10-21:
Linux has this full screen mode as well. Just press F11.
I switched it on accidentally and then struggled to leave it again because I didn't know the hotkey and well, yes, the menus don't work. That's s.th. for @neolit123 to reproduce first.

neolit123, 2017-10-21:
@dirkhh @sfuchs79 

this BUG should only happens on Windows. i found a fix at a QT BUG report and will send a PR soon.

neolit123, 2017-10-21:
@dirkhh @sfuchs79 

after some more testing on fullscreen mode:
- doesn't work at all in Ubuntu 17.04; could be very hard to get this to work!
- works in Ubuntu 16.04 but the menu bar disappears when in fullscreen
- it's buggy on Windows and needs hacks / fixes

@dirkhh 
i don't know if we have that many users of the fullscreen mode (F11), but i suggest that we just remove this functionality. it's too unstable...the Windows bug has been present since early Qt 5.x and on Ubuntu Unity seems to be pain.

if you are going to release the new Subsurface today, the best i can do is send a patch to remove fullscreen support?

sfuchs79, 2017-10-21:
I clearly vote for (at least temporally) removing the mode. As I was saying I activated it on accident.

dirkhh, 2017-10-21:
OK, this is the first I've seen "fullscreen mode".
Since 4.7.0 is now completely screwed up, please create and merge a PR that disables it on all platforms, @neolit123 
I'll do 4.7.0 later tonight, assuming network gets back to just being bad instead of completely hosed. Otherwise I'll do it in a few days.

PR #712 (2017-10-22): mainwindow: disable fullscreen support by default

### Describe the pull request:

- [x] Bug fix
- [ ] Functional change
- [ ] New feature
- [ ] Documentation change
- [ ] Language translation

### Pull request long description:

Require the FULLSCREEN_SUPPORT macro to enable fullscreen
support.

The toggle was added 4 years ago in Subsurface, but with the
current version of Qt 5.9.x, it's very buggy on Windows and
Ubuntu. While it's possible to make this work on Windows,
it seems to behave broken in different ways on different
versiosn of Ubuntu.

Fixes #705

### Changes made:

1) add a macro [... cut]
commit 3f7900bf (2017-10-22): mainwindow: disable fullscreen support by default

Require the FULLSCREEN_SUPPORT macro to enable fullscreen
support.

The toggle was added 4 years ago in Subsurface, but with the
current version of Qt 5.9.x, it's very buggy on Windows and
Ubuntu. While it's possible to make this work on Windows,
it seems to behave broken in different ways on different
versiosn of Ubuntu.

Fixes #705
closed 2017-10-22 by dirkhh