subsurface/subsurface #710: Import from / Export to phpDiveLog

IzzySoft, 2017-10-21:
### Describe the issue:
- [ ] Bug
- [ ] Change request
- [x] New feature request
- [ ] Discussion request

### Issue long description:
In the past, I've been using an *Uwatec SmartPro* dive computer with *AquaDiveLog* on a *Palm Tungsten T5* – both devices died almost simultaneously a few years ago and no longer start up (since then, I've only kept my paper logbook). I used the Conduit of [AquaDiveLog](http://www.aquadivelog.org/) to convert all data to CSV, for use with [phpDiveLog](https://github.com/IzzySoft/phpDiveLog). With both the DC and the Palm dead, that's the only source I could use to import my dives to *Subsurface* – and I'd like very much to export again to that format for my online dive log.

To import those data with the current version of *Subsurface,* I'd have to first adjust the CSV files (reason below) – so before doing that manually (or semi-automated) I thought I'd check if *Subsurface* might add direct support for that (being the author of *phpDiveLog* I of course will try my best to help on that, but I'm just a PHP dev). While the import would be a one-time job (and I've no idea how many divers are still using it), if *Subsurface* would support it via Export would give it new life (and me a reason to continue on it) – and certainly some folks would enjoy using the combination.

Technical details below in the corresponding section.

### Operating system:
* OS: Linux Mint 18.1
* Device: currently none (donations welcome :wink:)

### Subsurface version:
Subsurface 4.6.4 (from PPA)

### Steps to reproduce:
n/a

### Current behavior:
Neither import nor export from/to phpDiveLog available.

### Expected behavior:
Hopefully, both available :wink:

### Additional information:
*phpDiveLog* uses a collection of CSV files, which are basically a representation of a "relational database,* all residing in the same directory (per user, that is – so one such directory represents a log book – example representations can be found [here](https://github.com/IzzySoft/phpDiveLog/tree/master/diver/demo/data)):

* `global.csv`: Statistics like max depth, avg dive time, etc (can be ignored for import)
* `divesites.csv`: Details on the dive sites: location, place, GPS (lat,lon,alt), depth, water (fresh/salt/..), type (open water, wreck, cavern, ..), rating (numeric; 1-5 stars), description (text).  
  think of "location" as the town (eg. "Gozo") and "place" as the divesite (eg. "Fassej Rock"). Sites are connected to dives via a numeric "id" column.
* `logbook.csv`: the logbook data, one row per dive. Has some columns separate which are single columns in Subsurface, and some columns I couldn't correlate when trying an import (eg. tank details).
* `diveNNNNN_profile.csv`: Dive profiles, one file per dive, with `NNNNN` representing the dive ID (as given in `logbook.csv` as `dive#`).

As there's no separate import for divesite details currently, to import from those CSV data one first had to "join" `divesites.csv` and `logbook.csv`. Subsurface has a mode to import profiles – but I'm not sure if one must first join all the separate files while adding a dive ID column, or import a bunch of files one by one and tell Subsurface what dive it belongs to.

### Mentions:

IzzySoft, 2017-10-21:
PS: Wrote myself a "quick'n'dirty" converter which seems to do its job, generating a CSV with combined information from logbook and divesite data for import into Subsurface. Noticed that some columns cannot be matched/imported, even if available in Subsurface as well:

* rating: in the CSV, a number between 0 and 5. In Subsurface, 0 to 5 stars. Should be possible to match. (ADL/PDL also has ratings for dive sites, but those are not present in Subsurface)
* visibility: in the CSV it's a "text field" – and was supposed to hold a distance (eg. "10 m"). In Subsurface it's again 0..5 stars, so not sure how to match that.
* current, workload: are there no corresponding fields in Subsurface?

Some other fields I had to concatenate (eg. location + place). There are also user-defined columns, which (if having data) could be concatenated and added to the notes field to not get lost – though if it shall be *exported* again, it might need some thoughts.

Another issue is importing the dive profiles. I've managed to import them and they were even merged automatically; but despite of having mapped the DC's "warnings" to `sample setpoints`, I can see them nowhere – neither in the graph, nor in the data. Content of those warnings were things like "slow warning" or "id 0xffde" – no idea what the latter means.

mturkia, 2017-10-23:
Have you looked into Subsurface XML format? Since this is a somewhat complicated conversion from multiple CSV files it might be better to convert directly to XML and import that. This way you will have access to all the fields that we save.

IzzySoft, 2017-10-28:
@mturkia for export I indeed had considered that, especially for the divesite data (as there is no other interface to that). For import I wouldn't dare to fiddle with the "real database". I'm not very used to working with more complex XML (for me it's always a pain in the … behind to deal with attributes, for example). I'd much prefer JSON :wink:

If I understand correctly, I should see to "compile" the different CSV files into a single XML dataset similar to the format used in Subsurface, so the "Subsurface importer" would simply have to merge it (after doing some "sanity checks" on it)?

mturkia, 2017-10-30:
@IzzySoft yep, that is actually how most of our importers from other logging software currently work. Just convert the data to Subsurface style XML and import that.

neolit123, 2017-11-09:
closing due to inactivity. possibly resolved.
re-open if see fit.

closed 2017-11-09 by neolit123