subsurface/subsurface #713: Photo import: time sync going havoc

IzzySoft, 2017-10-21:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:
Trying to import photos to my dives, as described in the manual. Best described using an example:

Dive start 2006-09-08 10:30, duration 41min. Timestamp of the photo, according to file browser and Exif: 2006-09-08 10:39 – so well within the dive. On the first try, I get a "time is off" warning saying the photo sits at 12:39, so I adjust by -2h – but the photo is not imported. Trying again, Subsurface suddenly says the photo is at 8:39; I say OK, +2 – but again no luck. 3rd try brings it at 12:39 again.

File didn't change all the time, so I'm very confused. If I say "import anyway" it gets imported – but of course is not shown in the dive profile.

### Operating system:
OS: Linux Mint 18.1

whether it happens on another OS, I cannot tell.

### Subsurface version:
* Subsurface version: 4.6.4
* No other version tried
* fresh install from PPA

### Steps to reproduce:
As described:

1. right-click on the dive
1. select "load images from files"
1. pick an image

### Current behavior:
Import is not possible, except when ignoring the timestamp – in which case the image is not shown in the graph.

### Expected behavior:
As the timestamps match, the photo should be imported without warning, and shown in the dive graph.

### Additional information:
To work around such issues it would be great if those data can be adjusted/edited after the import, or instead of giving a shifting vector having the option to input the time directly. Oh: just noticed that's even possible by dragging the photo and dropping it on the graph at the appropriate "time mark". Unfortunately, if one misses it's no longer adjustable, and one must remove and re-add the photo first.

### Mentions:

sfuchs79, 2017-10-22:
Can't reproduce this under windows 10.
Could you please provide a screenshot of the picture import dialog when you see the failure and also one example picture?
If it's ok for you, you can attach it here on Github zip compressed.

atdotde, 2017-10-22:
Even more important would be a pair of xml and image files that illustrates the behaviour

IzzySoft, 2017-10-22:
@sfuchs79 sure:

[image] [image]

Screenshots of subsequent trys with the very same photo (click images for larger variants). Note the timestamp subsurface says the image has: it differs exactly for the amount of the shift specified. Besides, it doesn't matter if I pick the same photo each time – it always "toggles" that way: first picture too old, second too new (or the other way round, depending where you start).

As described, the timestamp of the photos usually fits well into the dive time. Taking a look at the Exif data of the photo from the screenshots:

    Date/Time Original              : 2006:09:09 10:04:37
    Create Date                     : 2006:09:09 10:04:37

Which is about the 9th minute of a dive having lasted 39 minutes:

[image]

(as you can see, I finally forced the load of the images and placed them manually on the graph, as described above)

@atdotde Not sure if I understand you correctly – but here's an excerpt of the Subsurface XML record for that dive:

``` xml
<dive number='61' divesiteid='a37e4811' date='2006-09-09' time='09:55:00' duration='38:52 min'>
  <buddy>Udo</buddy>
  <notes>Really a nice dive! Very good visibility and an amazing location. Not only the wreck was nice, the statue is also fun - and the buddies have been fun, too!</notes>
  <suit>semi-dry suit 7mm 1 piece + booties</suit>
  <cylinder size='15.0 l' start='200.0 bar' end='30.0 bar' />
  <weightsystem weight='10.0 kg' description='imported' />
  <divetemperature air='35.0 C' water='23.0 C'/>
  <divecomputer model='Manual import' deviceid='ffffffff'>
  <depth max='40.1 m' mean='21.756 m' />
  <sample time='0:00 min' depth='0.0 m' />
  <sample time='0:00 min' depth='0.0 m' />
  <sample time='0:04 min' depth='2.1 m' />
  <sample time='0:16 min' depth='4.4 m' />
  …
  <sample time='44:14 min' depth='0.0 m' />
  </divecomputer>
  <divecomputer model='csv' deviceid='ffffffff' duration='39:00 min'>
  <depth max='40.1 m' />
  </divecomputer>
</dive>
```

After having forced in (and manually adjusted) the image, there's one more line before `</dive>`:

``` xml
  <picture filename='/home/izzy/Documents/Fotos/2006/Malta/Diving/Eagle/IMG_0103.JPG' offset='+8:14 min' hash='d2deffa1bdfcde19e8fa07cea919c89310f6c6a1'/>
</dive>
```

Dive log (and profile data) was imported via CSV, converted from my PhpDiveLog data (see #710) – just in case that's relevant.

IzzySoft, 2017-10-22:
PS: YUCK! Fixed it. Despite of the offset shown, I turned off the "shift" (setting it to 0), marked the "force" checkbox, and clicked "OK" – guess what? It got imported and positioned correctly! I simply had to ignore that warning:

[image]

So it's rather a cosmetical error. Interesting: if I adjust the shift (eg. to 1h) and then take that back (0 shift), the warning is gone. Setting the shift to ±1 min also solves that (I just figured I had to shift by 5min anyway).

But now I cannot remove photos: right-click, "delete all images" has no effect. Didn't figure how to "select" photos (so "delete selected" doesn't work either). Work-around is of course to remove them manually from the XML … Btw: From inside the photos tab, none of the context menus seem to work ("load images" neither) – so loading I only can do by right-clicking the dive in the log, and chosing "load images" from there.

sfuchs79, 2017-10-22:
Ahhh... my bad memory:
Both items should be already fixed now. The first issue you reported was already fixed end of April 2017 but V4.6.4 is even older, its from mid of April 2017.
Can you please try again both things with latest version V4.7.1 which will be available soon?

IzzySoft, 2017-10-22:
@sfuchs79 As soon as it pops up in the PPA, sure! Need to use a "dummy log" for that then – just finished transfering my old data, incl. hand-written logs, and connecting the photos :wink:

Now I just hope someone picks up my other issue (#710) so I can get the data exported to my self-hosted log. People might already wonder whether I've stopped diving in 2009 :scream:

IzzySoft, 2017-10-22:
@sfuchs79 v4.7.1 just arrived. From a quick check I can confirm most of the issues solved:

* no more "mismatches" reported where there are none
* right-click and selecting "load images" also works from the image tab now
* same for "delete all images"

what still doesn't work:

* right-click on an image to remove it (there is no "delete this image")
* marking images to act on (so no deal on "delete selected images" as no selection seems possible). I've checked the manual, but found no mention of how to achieve this. The obvious things (drawing a rectangle around with the left mouse button down, clicking while holding Ctrl, Shift or Alt) didn't yield the expected result.

As a side-note: the "toggle pictures" icon seems to have no effect on the active view – the new state only comes to effect after having switched to another dive. Most other buttons have immediate effect.

sfuchs79, 2017-10-23:
Ok, good, some improvement. Some of the other items are already mentioned here plus more Windows specific issues:
#368 
Can't reproduce the "toggle pictures" issue.

IzzySoft, 2017-10-23:
Good match indeed! Most of it also seems to be fixed – except of selecting multiple pictures or even being able to delete a single one (right-click action on a single picture). The "toggle pictures" issue is just a minor annoyance I can live with (one simply has to know it's that way – if it is, of course).

So maybe we consider this issue here closed – and I track the other one for the remains?

dirkhh, 2017-10-24:
Yes, I'll close this one, please open a new one for the remaining issue - that makes it much easier to track.

Thanks

closed 2017-10-24 by dirkhh

IzzySoft, 2017-10-24:
@dirkhh thought so. Won't open another one, though, as that would be a duplicate to #368 then. As I wrote, the "toggle pictures" working with a delay is a) something I can live with (knowing it's that way *for me)* and b) cannot be reproduced by e.g. Stefan. Too minor to justify a separate issue (unless you say so that is). Everything else is already covered by #368.

commit 3ff73d0a (2015-03-25): HTML: don't collapse the dive if hyperlink is clicked

fixes the wrong behaviour as dive details must not be closed when
hyperlinks are clicking, So check for Hyperlinks before toggling the
dive.

Fixes #713