subsurface/subsurface #718: Manually entered dives and "mean depth"

IzzySoft, 2017-10-22:
### Describe the issue:

- [ ] Bug
- [x] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:
When entering dives manually (or importing from CSV without a profile), Subsurface "fakes" a dive profile and bases the "mean depth" on that. But one might happen to know the "mean depth" from a buddy with a DC. Still, there's no way to enter that (except for manually fiddling with the XML – where the mean depth gets ignored unless one manages to adjust the "fake profile" accordingly).

Wouldn't it be possible to let the user enter a mean depth, and then automatically adjust the "fake profile" accordingly?

### Operating system:
OS: Linux Mint 18.1

I assume this is OS independent.

### Subsurface version:
* Subsurface version: 4.6.4
* No other version tried
* fresh install from PPA

### Steps to reproduce:
1. Add a dive
1. look for a field to enter the mean depth

### Current behavior:
Mean depth cannot be specified manually.

### Expected behavior:
User should have the possibility to manually enter a mean depth – at least as long there's no "real profile" attached to the dive (i.e. one imported from a DC, other divelog, CSV).

### Additional information:

### Mentions:

neolit123, 2017-10-28:
CC for a review
@atdotde @glance-

28173, 2017-12-12:
I like the same feature for the windows version.

mkborregaard, 2020-08-30:
I would also like this (I'm on Mac)

mikeller, 2026-10-18:
Hi @IzzySoft,

Revisiting this: a recent pull request (#4677) attempted exactly this, adding a manual average-depth input field, and ran into the reason this hasn't been done: Subsurface is built around the invariant that average depth is always calculated on the fly from the dive's profile, for manually-added dives too. As mikeller explained in review there, any explicitly-set average depth gets silently overwritten every time the dive is validated (on save and on load), so a simple override field doesn't actually work, it would need a new 'dive without a profile' data type to do properly, which risks breaking assumptions made across a lot of the codebase.

The thread did land on a cleaner way to get what you're after: generating a synthetic profile that matches a given duration, max depth, and average depth together (similar to how manually-entered dives already work, just extended to honour an entered average depth too), implemented once in the backend so it works on both desktop and mobile, mikeller outlined this approach and a contributor (Michael-Tanzer) offered to build it that way.

Closing this report in favour of consolidating around that more concrete, already-scoped effort, please follow #4677 for progress, since that's where the actual implementation path forward is being tracked. Reopen here only if that approach stalls entirely and this needs to be picked up as a separate, fresh effort.

closed 2026-10-18 by mikeller

IzzySoft, 2026-10-18:
Thanks @mikeller (both of you? 🙈 – "As mikeller explained")!