subsurface/subsurface #719: Fedora 25 / Subsurface 4.7.1: missing flag icons

dirkhh, 2017-10-22:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

Installing the official binary I don't see any flag icons. I'm guessing we are not bundling them correctly?

### Operating system:

Fedora 25

### Subsurface version:

4.7.1 release binaries

### Steps to reproduce:

### Current behavior:

### Expected behavior:

### Additional information:

### Mentions:

@neolit123

ggardet, 2017-10-23:
Do you mean the flags displayed on the map?

dirkhh, 2017-10-23:
YES

ggardet, 2017-10-23:
By official binaries, you mean those built on OBS, I guess?

I tested daily on F26, openSUSE TW and openSUSE Leap 42.3 with no flag problem.
4.7.1 is showing fine the flags on my openSUSE 42.3.
Very strange.

Do you have log in the console that may help?

dirkhh, 2017-10-23:
As I mentioned, this is F25
And reading the code, the little flag marker icons are included in the executable via the QRC, so this isn't a packaging problem.
I'm beginning to wonder if this is a Qt 5.7 vs. Qt 5.9 problem. Do we need a feature that isn't in 5.7 in order to show those little flags?
@neolit123 any idea?

mikeller, 2026-10-17:
Hi @dirkhh,

Revisiting your own report here: Fedora 25 reached end-of-life years ago, and current Fedora releases have long since moved past whatever icon-font/package state caused the missing flag icons back then. We're closing this as stale/moot given the OS itself is no longer supported.

Please reopen if missing flag icons turn up on a currently-supported Fedora release.

closed 2026-10-17 by mikeller