subsurface/subsurface #720: Missing dependency in Debian package (PPA)

IzzySoft, 2017-10-22:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:
Just having updated from v4.6.4 on Linux Mint 18.1 (via the PPA) I noticed the entire globe/map section is gone. Seems some dependencies are missing. Starting from command-line I see:

```
qrc:/MapWidget.qml:2:1: module "QtQuick" is not installed
QUrl("qrc:/MapWidget.qml") failed to load with status: 3
qrc:/MapWidgetError.qml:2:1: module "QtQuick" is not installed
```

Installed the `qml-module-qtquick2` package, and the map is back.

### Operating system:
Linux Mint 18.1

### Subsurface version:
Subsurface 4.7.1. Didn't happen on previous versions as they used a different implementation.

### Steps to reproduce:
Install (or update to) v4.7.1, start the application: map widget remains empty (gray). Unless that package was already installed, of course.

dirkhh, 2017-10-22:
I triggered a new build of 4.7.1 that should include that dependency. Please let us know once you get the update notification and were able to try this. Thanks for the reports, @IzzySoft

closed 2017-10-22 by dirkhh

IzzySoft, 2017-10-23:
@dirkhh thanks for the fast reaction! As it's already installed on my system now I won't notice – but good to know it's fixed (preventing reports on missing maps :wink:)

dirkhh, 2017-10-23:
@IzzySoft you should get a notification that 4.7.1-2 is available (you have 4.7.1-1 on your system).
Or you could manually check for an update

IzzySoft, 2017-10-23:
Got the notice. Uninstalled the `qml-module-qtquick2` I had installed manually, updated subsurface – module was still missing:

```
Package: subsurface
Priority: optional
Section: utils
Installed-Size: 17543
Maintainer: Dirk Hohndel <dirk@hohndel.org>
Architecture: amd64
Version: 4.7.1-2~xenial
Depends: libc6 (>= 2.17), libcurl3 (>= 7.18.2), libgcc1 (>= 1:3.0), libgrantlee-templates5 (>= 5.0.0), libqt5bluetooth5 (>= 5.5.0), libqt5concurrent5 (>= 5.0.2), libqt5core5a (>= 5.5.0), libqt5gui5 (>= 5.2.0) | libqt5gui5-gles (>= 5.2.0), libqt5location5 (>= 5.5.0) | libqt5location5-gles (>= 5.5.0), libqt5network5 (>= 5.0.2), libqt5positioning5 (>= 5.5.1-3ubuntu1), libqt5printsupport5 (>= 5.0.2), libqt5qml5 (>= 5.0.2), libqt5quick5 (>= 5.0.2) | libqt5quick5-gles (>= 5.0.2), libqt5quickwidgets5 (>= 5.5.1) | libqt5quickwidgets5-gles (>= 5.5.1), libqt5svg5 (>= 5.0.2), libqt5webkit5 (>= 5.0.2), libqt5widgets5 (>= 5.2.0), libsqlite3-0 (>= 3.5.9), libssh2-1 (>= 1.2.5), libssl1.0.0 (>= 1.0.0), libstdc++6 (>= 5.2), libusb-1.0-0 (>= 2:1.0.16), libxml2 (>= 2.7.4), libxslt1.1 (>= 1.1.27), libzip4 (>= 1.0), zlib1g (>= 1:1.1.4), qml-module-qtlocation, qml-module-qtpositioning
```

As you see, it's still not mentioned in the dependencies.

mturkia, 2017-10-24:
On Mon, Oct 23, 2017 at 3:51 PM, Izzy <notifications@github.com> wrote:

> Got the notice. Uninstalled the qml-module-qtquick2 I had installed
> manually, updated subsurface – module was still missing:
>
> Package: subsurface
> Priority: optional
> Section: utils
> Installed-Size: [...]
The debian/rules file is updated within our  sources (defines the
dependencies), but Dirk's release tooling uses a copy of it outside the
sources dir. And I suppose that file was not updated.

dirkhh, 2017-10-24:
> On Oct 24, 2017, at 6:19 AM, mturkia <notifications@github.com> wrote:
> 
> On Mon, Oct 23, 2017 at 3:51 PM, Izzy <notifications@github.com> wrote:
> > As you see, it's still not mentioned in the dependencies.
> >
> The debian/rules file is updated within our sources (defines the
> dependencies), but Dirk's release tooling uses a copy of it outside the
> sources dir. And I suppose that file was not updated

I was quite optimistic that I DID update that manually (which is why I triggered the new build in the first place), but clearly I must have messed that up.

Now that I have somewhat better internet, I'll try again.

/D

IzzySoft, 2017-11-01:
Just confirming the latest release has the dependency properly listed \o/

PR #721 (2017-10-22): Include qml-module-qtquick2
Ubuntu and Debian require package qml-module-qtquick2 to be installed.

Fixes #720

### Describe the pull request:

- [ ] Bug fix
- [ ] Functional change
- [ ] New feature
- [ ] Documentation change
- [ ] Language translation

### Pull request long description:

### Changes made:

### Related issues:

### Additional information:

### Mentions:
commit 455559f5 (2017-10-22): Include qml-module-qtquick2

Ubuntu and Debian require package qml-module-qtquick2 to be installed.

Fixes #720