subsurface/subsurface #723: 4.7.1 Crash on OS X 10.10

fstengel, 2017-10-24:
I cannot run 4.7.1 on my laptop running 10.10. Visibly the xml library subsurface requires seems to be from a later system.

### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:
On trying to run the application, I get the crash described in the Additional information section. What I see is an incompatible version of libxml. Could it be that this version is incompatible with older versions of OS X?

### Operating system:
OS X 10.10.5

### Subsurface version:
4.7.1

### Steps to reproduce:
Basically: run the application

### Current behavior:
Crash
### Expected behavior:
Application running...

### Additional information:
Crash log:
```
OS Version:            Mac OS X 10.10.5 (14F2511)
Report Version:        11
Anonymous UUID:        <<removed>>

Sleep/Wake UUID:       <<removed>>

Time Awake Since Boot: 680000 seconds
Time Since Wake:       2100 seconds

Crashed Thread:        0

Exception Type:        EXC_BREAKPOINT (SIGTRAP)
Exception Codes:       0x0000000000000002, 0x0000000000000000

Application Specific Information:
dyld: launch, loading dependent libraries

Dyld Error Message:
  Library not loaded: @loader_path/../Frameworks/libxml2.2.dylib
  Referenced from: /Users/USER/*/Subsurface.app/Contents/MacOS/Subsurface
  Reason: Incompatible library version: Subsurface requires version 12.0.0 or later, but libxml2.2.dylib provides version 10.0.0

Binary Images:
    0x7fff651cf000 -     0x7fff6520592f  dyld (353.2.4) <3C068A0D-69D5-3AD6-86D9-6261D3B1564E> /usr/lib/dyld
    0x7fff87de6000 -     0x7fff87ed8ff7  libxml2.2.dylib (26.15) <B587DA25-360B-3742-BE9A-04B171504D79> /usr/lib/libxml2.2.dylib
```
(Machine details omitted)

janmulder, 2017-10-24:
There is a bugfix for this, but it seems that the file on the website (subsurface-divelog.org/downloads/test/Subsurface-4.7.1-11-g02f4490deafc.dmg) is currently broken. We will have to wait for @dirkhh to upload it again.

fstengel, 2017-10-24:
Thanks. I was hoping that would be the case.

dirkhh, 2017-10-24:
I re-uploaded the file and verified md5 sums, so this one is correct. Please try http://subsurface-divelog.org/downloads/test/Subsurface-4.7.1-11-g02f4490deafc.dmg

janmulder, 2017-10-24:
from the mailing list: 

> 
> > On Oct 24, 2017, at 8:02 PM, Federico Masias <fede@masias.net> wrote:
> >
> > Dirk,
> >
> > I just grabbed Subsurface-4.7.1-11-g02f4490deafc.dmg  -- it works, but is not signed properly, so Apple throws the Unidentified Developer error (10.11.6, if it matters)
> 
> I am uploading 4.7.1-12 since clearly there is something about -11 that just hates me.
> 
> This should be complete, valid, and signed.
> /D

So, the new binary is: subsurface-divelog.org/downloads/test/Subsurface-4.7.1-12-g9ef9c3efc02f.dmg

fstengel, 2017-10-24:
Seems to work. Tried downloading from my iX3M: crash. I'm opening another ticket.

closed 2017-10-24 by janmulder