subsurface/subsurface #724: 4.7.1 crash when trying to import from iX3M

fstengel, 2017-10-24:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:
When I try to import from my iX3M dive computer, subsurface crashes. 

### Operating system:
OSX 10.10.5

### Subsurface version:
4.7.1-12

### Steps to reproduce:
Start subsurface
Import>Import from dive computer
Brand: Ratio
Model: iX3M Easy (same with iX3M Pro Easy or any other model)

### Current behavior:
Crash

### Expected behavior:
download of dive data

### Additional information:
Crash log:
[subsurface-crash.log](https://github.com/Subsurface-divelog/subsurface/files/1412366/subsurface-crash.log)

Crashed thread is number 20 and seen around line 209 of the crash log

neolit123, 2017-10-28:
@fstengel 

hello, which language are you using?
does the crash happen with the English version of Subsurface? please, try switching the language in settings.

fstengel, 2017-10-28:
@neolit123 

Hi,

My laptop is using French ("LANG=fr_FR.utf8" when I ask for `env` in a terminal) as the system UI language.

For Subsurfaces UI Language, I normally have "Use system default" checked and "French (France)" appears in the popup.

1. I tried with  "Use system default" unchecked and "English" selected in the popup: everything works.
2. I tried "French (France)" with or without "Use system default" checked, both times a crash.

So, it seems something is off with the French translation. I'll stick to English for the time being.

dirkhh, 2017-10-28:
Thanks for the update. We have identified a mistake in the French translation that is the likely cause of this crash. Please use Subsurface 4.7.1 in English. Once 4.7.2 is released (relatively soon - the next few days) you should be able to switch back to French.

fstengel, 2017-10-28:
Thanks a lot.

closed 2017-10-29 by neolit123