subsurface/subsurface #740: 4.7.1 crash when trying to import from Cressi Giotto

nicosans, 2017-10-28:
### Describe the issue:

- [ X] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

Subsurface crash when I'm trying to import data from a Cressi Giotto dive computer. I use a infrared Divemate Fusion interface. It was working previously (at last 3 weeks ago), but I'm not sure it was with this version. 

When I click on the download button, the progress bar increase until 96 %, and then, the application crash.

### Operating system:

Linux

Mint 18.2

Cressi Giotto + Divemate Fusion IR

I didn't try

### Subsurface version:

4.7.1

Official release

### Steps to reproduce:

1) Select import from dive computer(Ctrl+D) 
2) Click on Download

### Current behavior:

Application crash when the progress bar is full.

### Expected behavior:

Application should not crash. The data selection list should appear

### Additional information:

application log :
Starting download from  /dev/ttyUSB0
Starting the thread 0
INFO: dc_deveice_open error value of 0
Erreur de segmentation

(segmentation fault, in english)

gdb dump :
#0  0x00007ffff13ebcc0 in _IO_vfprintf_internal (s=s@entry=0x7fff7f7fb800, format=<optimized out>, 
    format@entry=0x7fff680010d8 "modèle=%s firmware=%s port série=%s", ap=ap@entry=0x7fff7f7fb970) at vfprintf.c:1632
#1  0x00007ffff14b3856 in ___vsnprintf_chk (s=0xc1ad00 "modèle=Giotto firmware=(null) port série=", maxlen=<optimized out>, flags=1, slen=<optimized out>, 
    format=0x7fff680010d8 "modèle=%s firmware=%s port série=%s", args=0x7fff7f7fb970) at vsnprintf_chk.c:63
#2  0x000000000069acbd in ?? ()
#3  0x000000000069cba7 in ?? ()
#4  0x000000000077be83 in ?? ()
#5  0x000000000069c7c2 in do_libdivecomputer_import ()
#6  0x0000000000717b70 in DownloadThread::run() ()
#7  0x00007ffff22c77be in ?? () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#8  0x00007ffff70e16ba in start_thread (arg=0x7fff7f7fc700) at pthread_create.c:333
#9  0x00007ffff14a43dd in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:109
### Mentions:

neolit123, 2017-10-28:
@nicosans can you please confirm that this happens if you run Subsurface in English too?
we had some bugs only if Subsurface is running translated in French.

a fix was made @ Transifex:
http://lists.subsurface-divelog.org/pipermail/subsurface/2017-October/030578.html

CC @dirkhh

dirkhh, 2017-10-28:
I am planning to create 4.7.2 fairly soon which will fix this issue.

closed 2017-10-28 by dirkhh

nicosans, 2017-10-28:
Thanks a lot. It works perfectly if I run the application in English. I had to run it with another profile (subsurface --user=test), because the application was always in French after restart, even if I had changed the language to English before restarting.

commit 6efb3572 (2017-10-28): Pull latest translations

Fixes #740