subsurface/subsurface #741: Subsurface doesn't start after updating from 4.6.4 to 4.7.1.

elwetritsche, 2017-10-28:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:
Subsurface doesn't start after updating from 4.6.4 to 4.7.1.
4.6.4 worked without any Problems.

### Operating system:
What OS are you running? Windows 10 64bit
What is the OS version? 1709
What device are you using? Dell Latitude D620, 2 GHz, 4 GB RAM, 120 GB SSD
Does the same happen on another OS? Don't know

### Subsurface version:
What version of Subsurface are you running? 4.6.4 again
Does the same happen on another Subsurface version? Don't know
Are you using official release, test build, or compiled yourself? Official release

### Steps to reproduce:
Provide reproduction steps separated with new lines
1) 4.6.4 runs perfectly
2) Install/Update 4.6.4 with 4.7.1 without Problems  
3) Start Subsurface and .... wait .... wait .... wait ... => Nothing happnes, just nothing!!!
4) Reinstall 4.6.4, Start Subfarce, Runs perfectly again!

### Current behavior:
What is the current behavior? Just nothing!

### Expected behavior:
What is the expected behavior? Subsurface should start

### Additional information:
n.a.

neolit123, 2017-10-28:
@elwetritsche

1) hello, do you have opengl enabled on this laptop?
you can find out with this tool:
http://www.ozone3d.net/gpu_caps_viewer/

2) we need a crash log to be able to debug this.
- go to the start menu and start "Event Viewer"
- go to `Windows Logs -> Application`
- look for Subsurface crashing in the list of events
- copy/attach the text here

elwetritsche, 2017-10-28:
1) Yes: OpenGL 1.1 (GDI Generic with 2 ext.)
2)
Name der fehlerhaften Anwendung: subsurface.exe, Version: 0.0.0.0, Zeitstempel: 0x59ec7470
Name des fehlerhaften Moduls: unknown, Version: 0.0.0.0, Zeitstempel: 0x00000000
Ausnahmecode: 0xc0000005
Fehleroffset: 0x00000000
ID des fehlerhaften Prozesses: 0x1b78
Startzeit der fehlerhaften Anwendung: 0x01d34feb0baabcfd
Pfad der fehlerhaften Anwendung: C:\Program Files (x86)\Subsurface\subsurface.exe
Pfad des fehlerhaften Moduls: unknown
Berichtskennung: 21c8d196-f34c-4c9b-abe9-f19a082ca848
Vollständiger Name des fehlerhaften Pakets: 
Anwendungs-ID, die relativ zum fehlerhaften Paket ist:

neolit123, 2017-10-28:
OpenGL 1.1 is 20 years old now and i don't think we support it via the framework we use - Qt.
something is not right here...given you run Windows 10 i don't think your OpenGL version is really 1.1.
but if it's really 1.1 you might not be able to run Subsurface on this Windows setup.

1) what GPU does your laptop come with?
2) have you installed all the required GPU drivers on this setup?

neolit123, 2017-10-29:
the crash log is telling us this:
`Pfad des fehlerhaften Moduls: unknown`
which doesn't help.

the old version of Subsurface has a map in the form of a globe - uses the Marble library.
Marble appears to be more portable in terms of OpenGL, but we had different issues with it.

from my understanding of OpenGL, version 1.1 might be a bit too old to run our most recent version of Subsurface. on the other hand, the Qt framework should fall back to software rendering if there is an OpenGL problem, but it seems that it doesn't.

closed 2017-10-29 by neolit123

elwetritsche, 2017-10-29:
Dear Ivanov,

 

thanks for handling my issue and the explanation. I am happy with 4.6.4 on my old Win10-Laptop which I use for editing my dive log.

 

Because I was curious what would happen to an even older Dell Laptop which I use for playing around with Linux (Mint Linux 18.2 32 bit), I installed the newest updates. I had Subsurface 4.6.4 installed which worked properly, now 4.7.1 got installed and … guess what … it starts but with a black, empty screen. So it seems to have the same problem with Qt.

 

I think you should have an eye on that issue or at least put it in an release note, that 4.7.1 is not running on older machines, regardless ot the OS :)

 

Best regards

Felix

neolit123, 2017-10-29:
thanks for the further testing @elwetritsche

@dirkhh i will send a Pull request to update the README with this info:
http://doc.qt.io/qt-5/windows-requirements.html#graphics-drivers

> For Qt Quick 2 to work, a graphics driver that provides OpenGL 2.1 or higher is required. The default driver from Windows is OpenGL 1.1.

neolit123, 2017-10-29:
@elwetritsche

what happens if you do the following:
- open the start menu and type `cmd`
(this should open the windows console)
- use the `cd` command to navigate to the Subsurface directory
- type `set QT_OPENGL=software` and hit enter
- type `subsurface` to start subsurface 

are you still observing a crash?

elwetritsche, 2017-10-30:
Dear Ivanov,

 

thanks for the further tips. But 4.7.1 still crashes. I’m fine with 4.6.4

 

Best regards

Felix

neolit123, 2017-10-30:
@elwetritsche

given you are on Windows (we use a more recent Qt version there) we might have a fix for you in 4.7.2.

elwetritsche, 2017-10-31:
Dear Ivanov,

 

that are good new. I’m looking forward to updating to 4.7.2 :)

 

Best regards

Felix