subsurface/subsurface #747: Time to climb to X altitude missing!!

hhassey, 2017-10-30:
### Describe the issue:

- [ ] Bug
- [ ] Change request
- [x] New feature request
- [ ] Discussion request

### Issue long description:

The no fly or time to desaturate is primitive the way  I See it. Sometimes after a few dives one has to drive back home and that drive may take you high in the mountains, sometimes higher (lower ata pressure than a pressurised aircraft cabin), so in order to take the desicion as to when to hit the road, it would be awesome to have a table in Sub Surface that says time before you can safely climb to  X altitude according to the gradient factors chosen!!!

### Operating system:

### Subsurface version:

### Steps to reproduce:

### Current behavior:

### Expected behavior:

### Additional information:

### Mentions:

dirkhh, 2017-10-30:
We have actually played with this. Turns out that trying to use the existing algorithms to decide on "no-fly" or "no-altitude" will drive you nuts. Because even with the most extreme diving and the most conservative assumptions, we could never justify than more than a couple of hours of no fly.

So quite simply put: we have no idea what advise to give. And apparently neither do the "experts", because different dive computers give VASTLY different no-fly times (I frequently dive with three or more dive computers on a trip and the no-fly times will be off by double digit hours at times).

hhassey, 2017-10-31:
Dirk Thanks for your quick answer!! this is one of the things I love about
this project/community. I guess you are right, now it makes more sense why
the shearwater computers don't include a time to fly calculation, they told
me the same thing when I asked.

Best Regards

Humberto

neolit123, 2017-10-31:
thanks for the suggested enhancement.
i will close this issue now, but please re-open if see fit.

closed 2017-10-31 by neolit123