subsurface/subsurface #754: Map widget: Improve init. position for new dive site marker and update marker based on coord

sfuchs79, 2017-10-31:
### Describe the issue:

- [x] Bug
- [x] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

I see two more improvements for the new map widget based on the latest changes already done there:
- Today we have the following behavior: Map zoomed in to dive site A. One creates/imports a new dive with empty dive site. Enter the dive site B for the new dive which is a new one. Edit the new dive site. Map stays zoomed in at dive site A. You have to zoom far out and grab the new marker at '0' position somewhere south of Ghana and drag it to the desired position for dive site B.
Minimum two possible ways to improve this: 1. Zoom out automatically in this case to "whole world" or 2. Create the new marker at dive site A location.
- If you create a new dive site or edit an existing one you can manually enter coordinates. But the marker in the map is today not moved to the entered coordinates immediately while you are still inside the edit dive site dialog. It's updated only after pressing "apply changes". If you need some fine tuning to the coordinates by moving the marker in the map after entering the coordinates you today need to reenter the edit dive site dialog.
Can we update the marker immediately after entering valid coordinates? One remark: Please NO keyboard tracking for the edit field for this feature! Update only after coordinates edit looses focus (pressing enter or clicking in other field).

### Operating system:

### Subsurface version:

### Steps to reproduce:

### Current behavior:

### Expected behavior:

### Additional information:

### Mentions:

neolit123, 2017-11-01:
@sfuchs79 

so, these are some good suggestions. thanks.
from experience, it's usually best if we open a discussion to get the ideas more solid. i will post post this on the mailing list to see if we get any feedback from there too.

I.
for the first point, i would prefer to go with **1)** because for **2)** can end up being trouble.
for 2) the user might have already clicked a marker, so the marker is now centered on the screen. then by adding a new marker for the new dive site, the new marker will be placed exactly over the previous marker of the centered dive location. this can end up being a bit confusing, so i'm not sure about it.
i mean...the new marker will still be highlighted, so hopefully the user now understand that he has to move it. but if he doesn't move it, now these 2 markers will be stuck on top of each other and BTW we have code [***] that merges dive sites too close to each other, so one of them will be gone and these two markers will be merged into the same dive site! maybe that's OK as the user can just edit it again...

so 1) seems safer (view the whole globe and place a new marker at 0, 0), but 2) seems like the nicer behavior.

II.
for the second point, this is simply a missing feature (signal / slot connection). shouldn't it be more intuitive to also add a [UPDATE] button next to the text field with the coordinates + also update on losing focus? in terms of the signal / slot linkage, it seems doable but hopefully it doesn't end being difficult (keep [***] in mind).

sfuchs79, 2017-11-11:
@neolit123 
Took me some time until I was able to test the improvements. Great work! 
But... \<g>
I immediately have more suggestions:
- What is a great "side effect" of the new strategy is the following: After creating a new dive site for a new dive if the user "accidentally" already moves the map to the right location before pressing the "edit dive site" button this work is not lost when pressing the dive site button because the marker is created at the current map center. Thats great. But currently there is the strong need to once grab the marker then to have the gps coordinates copied to the edit field and to have it saved. Why not immediately copy the initial coordinates once. Maybe the marker is already at the 100% exact position and the user just wants to press "apply".
- If you now enter coordinates directly by keyboard this is reflected immediately in the marker position. BUT the map is not following the marker. So very often the marker will be out of sight and you have to find it by zooming out. The map should follow the marker in this case while keeping the current zoom level.

sfuchs79, 2017-11-11:
@neolit123 
Shall we reopen this issue to continue the discussion here?

neolit123, 2017-11-11:
@sfuchs79 i will just re-open the issue.
but notify the ML again about your suggestions.

neolit123, 2017-11-11:
@sfuchs79 @dirkhh 

i do agree with both improvements, but not sure i will have time to implement these before the next MINOR release; which Dirk is planning for later today.

also, i like seeing discussions about such things, therefore i pinged the ML.

sfuchs79, 2017-11-11:
@neolit123 
Relax - it was already clear to me that this will be post 4.7.3 stuff ;-)
And yes, let the others comment like the last time!

dirkhh, 2017-11-11:
By the way - the goal is to maintain a reasonably frequent update schedule. Linus and I agreed that we should aim at once a month or more frequently. This way there really is never a deadline that you might miss... and as you can see with the AppImage, the more I do to automate my tooling, the easier it will become to make releases...

neolit123, 2017-11-11:
sounds good.

PR #862 (2017-11-27): map-widget: interaction improvements

### Describe the pull request:

- [x] Bug fix
- [x] Functional change
- [ ] New feature
- [x] Code cleanup
- [ ] Build system change
- [ ] Documentation change
- [ ] Language translation

### Pull request long description:

These is a PR to make the `PART 2` additions from #754.

### Changes made:

1) move the map with the edited dive site
2) start dive site editing with coords pre-entered
3) use 'getter' function coordinate() for MapLocation
4) interrupt map animations on new interactions

### Related issues:

Fixes #754

### Additional information:

@sfu [... cut]
closed 2017-11-27 by neolit123