subsurface/subsurface #757: Cloud sync no more working

francobam, 2017-11-01:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

Cloud synchro not working properly

### Operating system:

Win 10/64 German

Android Mobile 6.01 Samsung J5

### Subsurface version:

4.7.1

[image]

### Steps to reproduce:

Cloud Speicher öffnen
Ergebnis: Keine Synchronisation
Mobile Version 69 logged dives
Desktop Version 67 (last 2 not synchronized)

### Current behavior:

### Expected behavior:

### Additional information:

### Mentions:

dirkhh, 2017-11-01:
We just released 4.7.2 which addresses a couple of possible problems. Frankly, I'm not sure it will address your issue, but it seems reasonable to ask you to try. :-)

francobam, 2017-11-02:
Thank you Dirk,
I will try next days!
Br Frank

janmulder, 2017-11-28:
Doing the following workflow (daily, as I am on a dive trip). Sync the OSTC3 to mobile after the dive, fill in some data, back at a wifi point, sync to cloud (manual), back on the desktop, load cloud (from online). All working perfectly. Pushing edits back to cloud from desktop, re-syncing mobile, also working as it should.

All with relatively new (but not current master) 4.7.4 on both desktop and mobile.

Propose: close this issue.

closed 2017-12-16 by janmulder