subsurface/subsurface #762: VPM-B ceiling on drugs for NDL dives in 4.7.2

adamel, 2017-11-02:
### Describe the issue:
- [X] Bug

### Issue long description:
Dives that were not even close to exceeding NDL suddenly indicates a violated ceiling in the profiles, like this:
[image]

### Operating system:
Windows 10, Swedish
PC

### Subsurface version:
4.7.2
Not seen on earlier versions

### Offending dive:
[dive_171102_01.zip](https://github.com/Subsurface-divelog/subsurface/files/1439507/dive_171102_01.zip)

adamel, 2017-11-02:
Dives with actual deco on the other hand looks just like on earlier versions:
[image]

adamel, 2017-11-02:
Played around some more, and it seems that if i plan and save a similar dive the ceiling looks ok, but if I select another dive and then select the planned dive again the ceiling suddenly has moved down so that it is violated when surfacing.

sfuchs79, 2017-11-03:
@RickMWalsh 
The things described above are also my main findings after the latest changes. What is described above also still happens after merging commits from #761 
It started to happen after the second part of #730 , as far as I remember correctly.
I can provide a few more examples from my side as well later.

RickMWalsh, 2017-11-03:
Thanks @adamel and @sfuchs79.  I can reproduce the issue.  Clearly I hadn't done enough testing with no-deco dives.  I've just added a commit, which I think fixes this issue VPMB: time_clear_ceiling is only valid after time_deep_ceiling.  Can you re-test my pull request?

sfuchs79, 2017-11-03:
@RickMWalsh 
Ok, now we are really getting very very close as you were saying. But it wouldn't be me if I didn't find anything else to complain about... ;-)

I did my usual test cases plus I compared the displayed ceiling (outside the planner) for ~30 of my logged dives and compared the results with V4.6.4. V4.6.4 btw was the version which still had these small spikes at the beginning of the profile. The spikes were all the discussions started...

On the positive side we have:
- Latest master with your latest fixes shows almost always identical results as V4.6.4 save the fact that the spikes are removed
- No odd behavior with shallow non deco dives any more (latest issue reported here)
- No more odd behavior when changing O2 percentage outside the planner

I still saw some small deviations (attached the related dives):
- 3 logged dives (# 659, 676, 717) show different ceiling now compared to V4.6.4. I tell you this mainly for information. I tend to believe that the new results may be the correct ones.
- For one example planned dive (# 718) I still have a different ceiling between inside the planner and outside. Meanwhile my guess is that for this setup there is still s.th. wrong with the calculation inside the planner.
- When replanning a planned dive we now have the effect again that before removing the planner generated waypoints the calculated ceiling is wrong. This time in this case there is a little bit too much ceiling. I know there were already some discussions also with Robert about this and I agree that this could be accepted. I just mention it because with some intermediate state of the code this had been already completely fixed.

[Ceiling_Test.zip](https://github.com/Subsurface-divelog/subsurface/files/1441194/Ceiling_Test.zip)

Thanks again for your effort!

atdotde, 2017-11-03:
Regarding your three points: 4.6.4 definitely was wrong as besides the spikes the ceiling also depended on the order of selecting dives in the dive list (and we were able to attribute that to the fact that we used some not properly initialised variables). So the sheer fact that there are differences to 4.6.4 by itself is a feature rather than a bug.

I have to look at #718. We definitely do things differently in the planner (where the model is actually defined via the Fortran program and where there is a clear distinction between bottom time and ascent) and when just showing a profile (where that distinction is ill defined but unfortunately the model depends on it).

Point three above is (very likely) for the same reason: When replanning, we use the planner way of distinguishing bottom time and deco. But upon reediting a saved dive, we don't have the information which waypoints were manually entered. So we treat the whole dive (including deco) as bottom time. This is wrong but I would still argue this is the best we can do (and if you really want to replan, you will delete those waypoints anyway). The only way I see around this problem is that we change the file format and explicitly store the bottom time for a planned dive.

RickMWalsh, 2017-11-03:
@sfuchs79 the ceiling displayed in your three dives should be slightly greater than in 4.6.4 (with spikes) and earlier versions (no spikes, but displayed ceiling outside of planner was less than when planning dives).  My patches in pull request #761 have deliberately modified the code to calculate deco time (used in the VPM-B CVA) to better represent the behaviour of the planner (and hence the original VPM-B implemenation/definition in Eric Baker's Fortran program, and V-Planner, etc.).
For planned planned dive 718 in your file, I get the same ceiling inside and outside of the planner once the deco stop/ascent points are deleted so the ascent is planned from the end of bottom time at 38min.  The way to fix this (and point three) would be to change the planner code and the assumption that the end of bottom time is the last defined point.  This would be a pain to code and would break consistency with other VPM-B implementations for multi-level dives (but arguably be better than the "standard").  Or alternatively, as @atdotde says, save the last defined point in the saved dive structure.
@adamel I doubt you can test my pull request if you're using Windows, but your dive shows no ceiling following the last patch I sent for any VPM-B conservatism level.

sfuchs79, 2017-11-04:
@RickMWalsh 
For logged dives and situation in planner before deleting planned waypoints: Absolutely ok and understood.

Regarding the planned dive # 718:
Are you sure that you see the same ceiling? On my side it really looks different:

Outside planner:
[image]

In planner (planned waypoints removed!):
[image]

RickMWalsh, 2017-11-06:
Yes, with dive 718, the ceiling inside and outside the planner are a bit different.  But editing the dive in the planner, then deleting the last few points, produce the same ceiling as planning the dive originally.  So the issue is actually how the profile is displayed outside the planner.  I think the issue is that the planner calculates deco_time (end of bottom to to surfacing, or with commit d15779, when surfacing would happen if the final ascent rate were 9m/min) as being 11min 20s, whereas outside the planner, it is inferred (and I say inferred because bottom time isn't well defined for an actual dive) to be 10min20s.  The difference is only one minute, but it messes with the CVA, and this can have an obvious effect for relatively short deco dives.  My latest pull request includes a number of tweaks to improve our inferred "deco time" outside of the planner, but finding something that is perfect in all cases it tough.

For the related issue of needing to delete planned points from the end of the dive before re-planning works, I have an idea that should be able to guess when the deco/ascent phase of the dive starts using the entry->percentages of the selected that we already calculate.  On selecting 'edit dive in planner' (so in or called from MainWindow::on_actionReplanDive_triggered()), we should step backwards through the selected dive data points (starting from the end), deleting points until we find entry->percentages with the maximum gradient factor <10 (or thereabouts).  For example, based on the heatmap, this should get us back to 38min in dive 718, which is the last manually entered point, and ~33min in @adamel 's second posted profile. @atdotde do you think this is worth considering?

atdotde, 2017-11-06:
We could use all kinds of heuristics to guess which waypoints were inserted by the planner. In addition to what you mention, we could use the information that the planner stops are only at known depths or that the stop length is always multiples of full minutes.

But all this is guesswork. We could probably get this to such precision that one would have to be very clever to outsmart it.

But wouldn't it be much simpler to actually save the time of the last manually entered waypoint in struct dive as well as xml and git?

RickMWalsh, 2017-11-06:
Yes, @atdotde, saving the time of tha last manually entered point would be much simpler. You've convinced me. It wouldn't help for actual dives, but I'm yet to see the hoards demanding that feature.

sfuchs79, 2017-11-08:
@RickMWalsh 
Isnt't it that your last commit you added today also clearly improved the result for my example dive # 718 (see above)? I have the feeling YES. That would be great.
I think we then can finally close all the ceiling related issues here :-)

RickMWalsh, 2017-11-08:
@sfuchs79 yes, that was the idea. I was going to mention when I submitted the last commit but dinner was ready. Please let me know if you can find any more bugs.
@adamel the latest vpmb changes have been pulled into master. Could you please test the latest Windows binary in http://subsurface-divelog.org/downloads/test/ and confirm all is as expected?

adamel, 2017-11-08:
Yes, confirmed that there are no unexpected ceilings on NDL dives. Thanks!

closed 2017-11-09 by atdotde