subsurface/subsurface #769: Provide continuous builds from Travis CI as AppImage

probonopd, 2017-11-04:
### Describe the issue:

- [ ] Bug
- [ ] Change request
- [x] New feature request
- [ ] Discussion request

### Issue long description:
Currently each each `git push` triggers a [build on Travis CI](https://travis-ci.org/Subsurface-divelog/subsurface), but the build artifacts are thrown away. Instead, we could bundle them up as an [AppImage](https://appimage.org/) and provide them for download, hence allowing for easy testing of the very latest continuous builds on local machines.

### Operating system:
The builds done on Travis CI in an Ubuntu trusty environments will run on most Linux distributions since 2014 when bundled as an AppImage.

### Subsurface version:
master

### Additional information:
https://plus.google.com/+DirkHohndel/posts/Vj2r74g3hBM
> [image]

__I am working on a pull request to implement this.__

### Mentions:
@dirkhh

dirkhh, 2017-11-04:
Yes, I'd love to see that. Have you done this on Travis CI before, @probonopd - you know the AppImage build process a LOT better than I do... on G+ you indicated that you'd give it a try which I take to mean that you might be willing to donate your time to get this working. I'd really appreciate it. I'm still struggling with just the simple (yeah, right) process of getting a working AppImage for 4.7.2 - the complexity of the googlemaps plugin, the need to have Qt-5.9.2 (so that BLE has a chance to work) still have me stuck (but admittedly... I have way too many things on my plate so I haven't put as much effort into fixing this as it deserves)

probonopd, 2017-11-04:
I have a first build ready but I don't know how to best exercise it, e.g., the googlemaps plugin to make sure everything works - maybe you want to have a look?

https://transfer.sh/Yza1O/Subsurface-git.43cf422-x86_64.AppImage

mturkia, 2017-11-04:
The map is not showing for me, there is only white/blank area where the map
should be. And I get the following complaints on console when starting this
AppImage (these missing packages are required for the map):
---8<---
QApplication: invalid style override passed, ignoring it.
can't find Subsurface localization for locale "en-US"
qrc:/MapWidget.qml:4:1: module "QtPositioning" is not installed
qrc:/MapWidget.qml:2:1: module "QtQuick" is not installed
qrc:/MapWidget.qml:3:1: module "QtLocation" is not installed
qrc:/MapWidget.qml:4:1: module "QtPositioning" is not installed
qrc:/MapWidget.qml:2:1: module "QtQuick" is not installed
qrc:/MapWidget.qml:3:1: module "QtLocation" is not installed
qrc:/MapWidget.qml:4:1: module "QtPositioning" is not installed
qrc:/MapWidget.qml:2:1: module "QtQuick" is not installed
qrc:/MapWidget.qml:3:1: module "QtLocation" is not installed
QUrl("qrc:/MapWidget.qml") failed to load with status:
QQuickWidget::Status(Error)
qrc:/MapWidgetError.qml:2:1: module "QtQuick" is not installed
qt.network.ssl: QSslSocket: cannot resolve SSLv2_client_method
qt.network.ssl: QSslSocket: cannot resolve SSLv2_server_method
---8<---

miika

probonopd, 2017-11-04:
OK, that's a start. So Subsurface is now using QML?

probonopd, 2017-11-04:
In `.travis.yml` there is:

```
      bash -e ./subsurface/scripts/build.sh -both -no-bt ;
      bash -e ./subsurface/scripts/build.sh -both
    - env CTEST_OUTPUT_ON_FAILURE=1 make -C subsurface/build check
    - # env CTEST_OUTPUT_ON_FAILURE=1 make -C subsurface/build-mobile check
```

Currently I need to do manually:

```
    - # FIXME: The next 5 lines would ideally be covered by ./subsurface/scripts/build.sh
    - mkdir -p appdir/usr/bin ; cp $(find . -type f -name subsurface) appdir/usr/bin/
    - mkdir -p appdir/ ; cp -r ./install-root/* ./appdir/usr
    - mkdir -p appdir/usr/share/metainfo ; cp subsurface/appdata/subsurface.appdata.xml appdir/usr/share/metainfo/
    - mkdir -p appdir/usr/share/applications ; cp subsurface/subsurface.desktop appdir/usr/share/applications/
    - mkdir -p appdir/usr/share/icons/hicolor/256x256/apps ; cp subsurface/icons/subsurface-icon.png appdir/usr/share/icons/hicolor/256x256/apps/
```

Is there a reason why `./subsurface/scripts/build.sh` is not doing these things, or can it be enabled to do these things so that I don't have to do them in `.travis.yml`?

mturkia, 2017-11-04:
I guess the googlemaps plugin uses the following deb packages:

qml-module-qtpositioning
qml-module-qtlocation
qml-module-qtquick2

And you might want to check out packaging/ubuntu/debian/control just to
make sure you have them all..

miika

dirkhh, 2017-11-04:
@probonopd 
build.sh is used on Linux and Mac and used for a number of different purposes, as part of my process to build releases as well as for people trying to build Subsurface from source on either of these platforms.
Yes, we could add the 5 lines to it, but they'd have to be inside a `-createAppImage` command line option or something like that in order to not mess with the other uses of this script

dirkhh, 2017-11-04:
and to answer your other question, yes, Subsurface is now using QML in order to bring up the maps

probonopd, 2017-11-04:
> Subsurface is now using QML in order to bring up the maps

I am a bit puzzled here: Where are the relevant `.qml` files for the desktop build? I see plenty in `subsuface-mobile` but not for the desktop.

[`linuxdeployqt`](https://github.com/probonopd/linuxdeployqt/blob/74580fa63e0ee66e2f4c437a1805f2bed69d9a83/tools/linuxdeployqt/shared.cpp#L1395-L1416) works by running ` qmlimportsanner: "-rootPath foo/ -rootPath bar/ -importPath path/to/qt/qml` but I can't seem to figure out what `path/to/qt/qml` must be in this case...

dirkhh, 2017-11-04:
For some reasons @neolit123 decided to add the desktop QML files to the mobile-widgets folder. I don't think that is the right thing to do and want to fix that, but given that they are bundled into the executable via QRC, I don't think you need to worry about them...
Anyway, the three relevant QML files are 
`MapWidget.qml
MapWidgetContextMenu.qml
MapWidgetError.qml`

probonopd, 2017-11-04:
Since `qmlimportscanner` wants directories, not files (I think, may be wrong though) it would be neat to have them in a separate `desktop-widgets` directory.

neolit123, 2017-11-04:
@dirkhh what folder do you want me to move the map widget stuff to?

probonopd, 2017-11-04:
This is what I am trying for now: https://github.com/probonopd/subsurface/commit/c4a3c83c7296ea044d24d5981eda4fa00a2f236d

neolit123, 2017-11-04:
@probonopd 
the map widget also uses some icons.

probonopd, 2017-11-04:
> Yes, we could add the 5 lines to it, but they'd have to be inside a -createAppImage command line option or something like that in order to not mess with the other uses of this script

Somehow this doesn't seem to happen (which would already make me happy on that front):

https://github.com/Subsurface-divelog/subsurface/blob/76286c4ece36baff3aaabb8a4e1252997d8ad7c4/CMakeLists.txt#L477-L485

probonopd, 2017-11-04:
Somehow `linuxdeployqt` doesn't seem to pick up the need for `geoservices/libqtgeoservices_googlemaps.so`. Can you point me to the mechanism by which it gets loaded?

neolit123, 2017-11-04:
on Windows it's loaded on runtime from a folder `plugins\geoservices\<plugin_name>.so` which is a folder next to the subsurface binary. i believe this should be a standard search path on Linux too.

probonopd, 2017-11-04:
I know. But how does a tool (like `linuxdeployqt`) figure out that this very plugin is needed, in order to put it there? (We do not bundle __all__ plugins but only those we know are needed by a [set of rules](https://github.com/probonopd/linuxdeployqt/blob/05eb73a516a92d4216fdca78aaf54d230a97134b/tools/linuxdeployqt/shared.cpp#L1099-L1257), similar to macdeployqt). 

I guess for now I just copy it in manually?

neolit123, 2017-11-04:
from my experience the `*deployqt` tools on Linux and Windows are usually broken in respect to plugins. haven't tried macOS.

neolit123, 2017-11-04:
> @dirkhh what folder do you want me to move the map widget stuff to?

@dirkhh i have moved the map widget to `<subsurface-root>/map-widget`.
any comments before the PR?

probonopd, 2017-11-04:
@neolit123 why not `desktop-widgets`, analog to `mobile-widgets`?

neolit123, 2017-11-04:
@probonopd the map-widget is QML based and we are about to use it for both the desktop and mobile versions of subsurface. the mobile support will come eventually...

neolit123, 2017-11-04:
https://github.com/Subsurface-divelog/subsurface/pull/773

probonopd, 2017-11-04:
https://transfer.sh/NZXmD/Subsurface-git.7d21b42-x86_64.AppImage has Google Maps working. 

Remaining issues identified so far - please add to the list:
* Get translations recognized
* "Share on" menu is empty - what do I need to bundle for this to work?
* Print command is gone in "File" menu - is that normal?

neolit123, 2017-11-04:
> Print command is gone in "File" menu - is that normal?

this can happen if you have enabled the `NO_PRINTING` option for cmake.

> "Share on" menu is empty - what do I need to bundle for this to work?

`FBSUPPORT` has to be enabled from cmake.

probonopd, 2017-11-04:
@neolit123 I am using the normal `build.sh` from the existing `.travis.yml`. I guess it would need to be enabled there.

neolit123, 2017-11-04:
@probonopd 

Qt 5.9.1 does not have `QWebKit`, as it's deprecated. we depend on `QWebKit` for both the printing and the facebook share (integrated browser).

unless the `build.sh` in the YML is extended with `-build-with-webkit` these two options won't be available in the generated AppImage. the bigger question would be where to find pre-built `QWebKit` binaries and how to install them via the YML.

@dirkhh can elaborate more.

probonopd, 2017-11-04:
Does anyone have a clue why translations do not get loaded?

dirkhh, 2017-11-05:
My apologies for disappearing. I spent four hours completing a Taekwondo belt test, and now I'll hang out with my kids for their birthday.
I'll look into the remaining issues if I find time later today, but it might not happen until tomorrow.

dirkhh, 2017-11-05:
OK, starting to look at it...
First comment (running this on ArchLinux)
```
qt.network.ssl: QSslSocket: cannot resolve CRYPTO_num_locks
qt.network.ssl: QSslSocket: cannot resolve CRYPTO_set_id_callback
qt.network.ssl: QSslSocket: cannot resolve CRYPTO_set_locking_callback
qt.network.ssl: QSslSocket: cannot resolve ERR_free_strings
...
qt.network.ssl: QSslSocket: cannot resolve OPENSSL_add_all_algorithms_noconf
qt.network.ssl: QSslSocket: cannot resolve OPENSSL_add_all_algorithms_conf
qt.network.ssl: QSslSocket: cannot resolve SSLeay
qt.network.ssl: QSslSocket: cannot resolve SSLeay_version
qt.network.ssl: Incompatible version of OpenSSL
```
So that's an interesting issue... and it prevents using the Subsurface cloud (as that is an https connection to our server)

As for translations, here's some helpful output for you:
```
./LANG=de_DE ./Subsurface-git.f25a10d-x86_64.AppImage -v -v
validateGL(): created OpenGLContext.
validateGL(): obtained QOpenGLFunctions.
validateGL(): detected OpenGL version 3.0.
can't find Qt localization for locale "de-DE" searching in "/tmp/.mount_SubsurRGewD6/usr/translations"
can't find Subsurface localization for locale "de-DE"
```

So it would automatically pick up the translations in ```usr/translations``` - that might be the easiest way to address this.

Bluetooth seems to work (YAY).

So what's missing is
- install QtWebKit (it's not included in the official distro that we are using, but we should be able to build against that distro - or we could try using the pre-compiled binaries from here: https://github.com/annulen/webkit/releases/tag/qtwebkit-5.212.0-alpha2 ) and call ```build.sh -build-with-webkit```
- move the translations (both Qt and Subsurface) into ```usr/translations```
- figure out how to deal with that OpenSSL issue

@probonopd - instead of me trying to mess with what you've already started, would you want to continue and work on these three issues? The first two should be pretty straight forward, not sure about the third one...

dirkhh, 2017-11-05:
OK, tested on Ubuntu 17.04, consistent results, except for the OpenSSL issue.

probonopd, 2017-11-05:
The OpenSSL issue is giving me some headache. Not sure yet what causes it, possibly related to https://github.com/probonopd/linuxdeployqt/issues/110. What happens if you unpack the AppImage and manually delete the SSL libs, then run AppRun like this:

```
chmod +x Downloads/Subsurface-git.f25a10d-x86_64.AppImage 
./Downloads/Subsurface-git.f25a10d-x86_64.AppImage --appimage-extract
rm squashfs-root/usr/lib/libssl.so.1.0.0 squashfs-root/usr/lib/libcrypto.so.1.0.0 
./squashfs-root/AppRun 
```

On a related note, I'd be _very_ interested in your opinion about our list of [Desktop Linux Platform Issues](https://github.com/AppImage/AppImageKit/wiki/Desktop-Linux-Platform-Issues) in which OpenSSL gets a [honorable mention](https://github.com/AppImage/AppImageKit/wiki/Desktop-Linux-Platform-Issues#openssl).

probonopd, 2017-11-05:
@dirkhh can you please comment on https://github.com/Subsurface-divelog/subsurface/issues/769#issuecomment-341921078, thanks.

probonopd, 2017-11-05:
> unless the build.sh in the YML is extended with -build-with-webkit these two options won't be available in the generated AppImage. 

I'll leave `build.sh` up to you guys @dirkhh @neolit123 (since it is supposed to be multi-purpose and I don't want to break it) and pick up whatever it outputs.

dirkhh, 2017-11-05:
@probonopd please look at my comment https://github.com/Subsurface-divelog/subsurface/issues/769#issuecomment-341945171 regarding the `-build-with-webkit` option. `build.sh` already knows how to do this, but so far on Travis we haven't installed WebKit (and it no longer is bundled as part of Qt binaries). I'll try to look into this this morning.
I'll also look into why not everything appears to get installed when running on Travis.

dirkhh, 2017-11-05:
OK, making progress, running a test right now. I'm skeptical how easy it will be to get QtWebKit integrated into our old build, we may have to build it from source (I'm doing a test with the prebuilt binaries right now - building from source takes a LONG time).
I'm curious - why did you disable the cache?
And also, I notice that all of your commits are without SOB - are you planning to collapse this all into one commit that you'll submit and sign off on?

probonopd, 2017-11-05:
> I'm curious - why did you disable the cache?

It was not giving speed-ups for me but threw errors. Something to be looked into once we have the rest working, I guess.

> And also, I notice that all of your commits are without SOB - are you planning to collapse this all into one commit that you'll submit and sign off on?

Yes.

neolit123, 2017-11-05:
if we stay on that `trusty` VM for a while, we can potentially build QWebKit our selfs, upload the binaries somewhere (even Google Drive might work, to safe bandwidth) and download and install it on each CI job. i've seen similar YMLs, but i'm not sure how well that would work on the Qt / Linux side.

dirkhh, 2017-11-05:
Yes, that's my current plan. I have a Trusty machine and I'm trying to figure out the best way to build and package just that. Building WebKit takes forever, so it may be a while (especially as I have another family thing that I'm heading to in a minute).
In the meantime, @probonopd I have the missing directories for you... build.sh for historical reasons didn't do a make install on Linux - it now does so and puts everything in the install-root. My branch is a mess right now as it brings in a few random commits from master, but you'll be able to find what you need in https://github.com/Subsurface-divelog/subsurface/commits/travisAppImage

probonopd, 2017-11-05:
Thanks @dirkhh that will simplify things a lot. I am assuming you are folding the changed `build.sh` into master anytime soon?

neolit123, 2017-11-05:
recently, i read that people are stripping (`-s`) the debug binary of QWebKit to reduce it's size and it still works. but that was in a thread about Win32. a ~1.25GB binary becomes ~80MB.

from my experience this might not always work, but it's worth a shot to reduce network traffic.

probonopd, 2017-11-05:
How are you handling QWebKit for macOS and Windows?

neolit123, 2017-11-05:
i think Dirk is building it from source on all platforms.

dirkhh, 2017-11-05:
@neolit123 is correct. It's built from source for the other platforms as well.
@probonopd I will fold my changes into master today. I just ran out of time this morning.

dirkhh, 2017-11-05:
@probonopd I see no new commits in your 'patch-1' branch, so I assume that you aren't working on this today? Just to make sure I benefit from anything you might have figured out. I expect to work on this on and off for the next few hours... I have a few "real life" things that I have to do as well (like, get a flu shot), but I hope to make progress. The last build https://travis-ci.org/Subsurface-divelog/subsurface/builds/297608345?utm_source=github_status&utm_medium=notification failed in unexpected ways, I need to understand what happened there as I first want to get the translations fixed before diving again into QtWebKit...

probonopd, 2017-11-05:
@dirkhh nu hurries. I will resume working on this once `build.sh` in master outputs all files, including the translations, icons, and such.

dirkhh, 2017-11-05:
I think I have the right `build.sh` changes, but right now the Travis build fails and it makes no sense to me why it does. What am I missing?
https://travis-ci.org/Subsurface-divelog/subsurface/builds/297680607?utm_source=github_status&utm_medium=notification
In there you'll see in line 4604 that `libgit2.26.so` is in `/usr/local/lib`, yet in line 4669 it complains that `ERROR: ldd outputLine: "libgit2.so.26 => not found"`

probonopd, 2017-11-05:
`linuxdeployqt` uses `ldd` under the hood. Perhaps you need to `export LD_LIBRARY_PATH=/usr/local/lib/` before you invoke it?

dirkhh, 2017-11-05:
you were explicitly unsetting it, so I thought there was a reason for that... OK, next attempt just pushed :-)

probonopd, 2017-11-05:
Unsetting it so that Qt doesn't get into the way...

dirkhh, 2017-11-05:
Not sure that makes sense, @probonopd - you set it explicitly to `/usr/local/lib` with a comment that this is to work around a bug - and then a couple lines later you unset it. I don't think Qt has anything to do with it. But either way, regardless why it was there, I removed the `unset` and am running a new test... this just all takes so long :-(

dirkhh, 2017-11-05:
and in other news... two hours into the QtWebKit build on my Trusty machine I get a build error (at that point the progress showed 90%). That's why I hate every time I have to touch Qt as a dependency...

probonopd, 2017-11-05:
This built: 
https://travis-ci.org/probonopd/subsurface/builds/297687635
https://github.com/Subsurface-divelog/subsurface/compare/travisAppImage...probonopd:patch-2

dirkhh, 2017-11-05:
mine just completed as well, testing the result... after that I have to run some errands :-(
But first I'll push my other changes into master so you can work on top of that.

probonopd, 2017-11-05:
Have a look at https://github.com/probonopd/uploadtool if you'd like to have each build uploaded to GitHub Releases rather than transfer.sh. This would also enable us to do efficient binary delta updates (no more re-downloading of Qt which each build).

dirkhh, 2017-11-05:
my changes to master are pushed. I did NOT push my changes to .travis.yml - those you can see in my `travisAppImage` branch.
Let me know which branch you will be working in (not sure how long I'll be out, but I will continue to work on this later)

dirkhh, 2017-11-05:
oh, and my latest test build switches translations just fine :-)
https://transfer.sh/phopf/Subsurface-git.2fa90ed-x86_64.AppImage
build output at https://travis-ci.org/Subsurface-divelog/subsurface/builds/297688156?utm_source=github_status&utm_medium=notification

probonopd, 2017-11-05:
> Let me know which branch you will be working in

https://github.com/Subsurface-divelog/subsurface/pull/777

__777__, my lucky number!

dirkhh, 2017-11-06:
Your PR didn't include QtWebKit. I'm betting that these are all results of me adding QtWebKit to the mix (which is needed for printing and Facebook integration).
I appreciate any help to clear those up :-)

probonopd, 2017-11-06:
Wow, so the dependency on QtWebKit is for Facebook? Isn't that a hefty price to pay? Wouldn't it be possible to open a certain URL using the system's browser instead? (I may be totally wrong here.)

[19 more replies]

closed 2017-11-06 by dirkhh