subsurface/subsurface #770: Implement update mechanism for continuous builds

probonopd, 2017-11-04:
### Describe the issue:

- [ ] Bug
- [ ] Change request
- [ ] New feature request
- [x] Discussion request

### Issue long description:
This is what Subsurface currently does (independently of AppImage):

[image]

This works well for _releases_, but not so much for _continuous builds_. For continuous builds, maybe we can have it check GitHub Releases for the latest AppImage continuous build instead, and have it update itself using zsync-based binary delta updates using [`libappimageupdate`](https://github.com/AppImage/AppImageUpdate/commit/61c7ae44a358e46798b40035a860e0446ed3b3e2) - which would mean that users don't have the bits that have not changed between builds (e.g., Qt) but only the few pieces that have actually changed.

### Operating system:
Linux

### Subsurface version:
master

### Current behavior:
A continuous build checks whether there is a new _release_

### Expected behavior:
A continuous build checks whether there is a new _continuous build_

### Additional information:
n/a

### Mentions:
@dirkhh @TheAssassin

dirkhh, 2017-11-04:
A few comments:
We should be very clear that most users really want us to do more frequent releases so they don't need to use test builds - our test builds break things in weird ways, they have on rare occasions caused data loss, we have played with file formats that had unintended side effects, etc. So in general I think we should keep the notion that for **most** users all they want is a notification that a new ***release*** version is available.
That said, I love the idea of having build artifacts of the continuous build process available for people to download and use if they so choose. What I'm a bit unclear about is the "update itself in place" - I think I'd want to understand better how that works. And I would never want this to be an automatic process without user intervention.

probonopd, 2017-11-04:
The way I see this, there should be (at least) two channels:

* Releases should update to the latest release
* Continuous builds should update to the latest continuous build

> And I would never want this to be an automatic process without user intervention.

Me too...

probonopd, 2017-11-06:
Currently blocked by https://github.com/AppImage/AppImageKit/issues/518

antony-jr, 2018-01-15:
> What I'm a bit unclear about is the "update itself in place" - I think I'd want to understand better how that works. And I would never want this to be an automatic process without user intervention.

@dirkhh this is how **AppImage** AutoUpdate works -> https://github.com/antony-jr/AppImageUpdater , You can use [this library](https://github.com/antony-jr/AppImageUpdaterBridge) to do the same for **subsurface**.  ( The example app can update itself if the user wants to... )

**NOTE** : The library does not have a stable release , **do not use this for production** but just for hacking arround! and testing the AutoUpdate for AppImages.

### How AutoUpdate Works ?
By **John Von Neumann** stored program concept ( **I bet everyone knows this** ) , The application is read from the disk to the memory for execution , now the program running for the user and the one stored in the disk are **detached** , i.e we could do anything to that from the **program running**. ( I learned this the hard way when I deleted my entire disk on ubuntu with **dd** while watching a video in youtube ).

Thus all we have to do is to run the normal updater with **argv[0]** on program runtime , the library takes care of the the updating part for any AppImage  , you just need to provide the app name or path.

In addition , it has **delta** compression , so users don't have to donwload the Qt Libs everytime. using zsync.

## This is how it works!
**CHECKING FOR UPDATES**
[image]

**GET CONFIRMATION FROM USER**
[image]

**DO THE DELTA UPDATE IF THE USER ACCEPTED**
[image]

**NOTIFY THE USER OR START YOUR APP**
[image]

atdotde, 2018-01-15:
> By John Von Neumann stored program concept ( I bet everyone knows this ) , The application is read from the disk to the memory for execution , now the program running for the user and the one stored in the disk are detached , i.e we could do anything to that from the program running. 

I am not sure this statement is correct. In my experience a program tends to crash when you modify the underlying binary file on disk. This is because the kernel does not load the whole binary into RAM and then keeps it there but rather relies on the fact that it can load parts later that are not needed initially (and also frees memory during the run which then has to be read back from disk later).

probonopd, 2018-01-15:
I'd say the new version should be created in the same directory as the old version (by downloading the binary delta and by copying the rest, that is the unchanged parts, from the original). The old version should be left intact. So that the user can run the new version and if it works properly, can decide to delete (or keep) the old version.

antony-jr, 2018-01-15:
> The old version should be left intact. So that the user can run the new version and if it works properly, can decide to delete (or keep) the old version.

@probonopd Yes the old version is kept intact but if the programmer wants to he can deleted when the same name exist!  @atdotde This proves my point that we can do anything to the file in disk.

> In my experience a program tends to crash when you modify the underlying binary file on disk. This is because the kernel does not load the whole binary into RAM and then keeps it there but rather [...]

Yes you are right , but this is already loaded in the memory so there will be no problem .  Already tested in some modern linux os , We just need the updater code to be in RAM and not the entire program , calling it in runtime can solve this.

dirkhh, 2018-01-16:
@antony-jr no, this isn't how it works. You may think that you removed the old file, but the OS keeps it around because it's still open (the app is running). Only after the last user of the disk file has terminated will the OS release the disk storage.
As @atdotde said, no modern OS loads a complete binary into RAM when an app is started. It maps the memory and pages it in from disk when needed. And in low memory situations it will happily drop pages that it already loaded into memory earlier (and simply load them back from disk if the code jumps back there again).

antony-jr, 2018-01-16:
@dirkhh But as @probonopd said it will not delete the file (Just a option that I created , works somehow ) , so the old AppImage is intact and suffixed with **.zs-old**

TheAssassin, 2018-01-25:
>> By John Von Neumann stored program concept ( I bet everyone knows this ) , The application is read from the disk to the memory for execution , now the program running for the user and the one stored in the disk are detached , i.e we could do anything to that from the program running.

> I am not sure this statement is correct. In my experience a program tends to crash when you modify the underlying binary file on disk. This is because the kernel does not load the whole binary into [...]

Thanks @atdotde for clearing up some of this.

About computer architectures: I am pretty sure @antony-jr is trying to prove some point based on the strict separation of architectures compsci professors teach in computer architecture classes. Yes, there's the Harvard architecture, which requires two different hardware level storages for program and data, and the von Neumann architecture, which expects program and data to be put into the same storage, and so on. Both systems have advantages and disadvantages, but that isn't really interesting for this discussion.

There are good reasons von Neumann's architecture has won this "battle of the architectures", and was the one early computers have been designed with. But that's only really interesting if you're interested in the history of compsci.

One reason is that operating systems implement techniques to prevent software from reading from and/or writing into other processes' data, be it the program or data. Trying to do so produces segfaults. This has been the case for decades, and prevents attacks like they were possible with early versions of e.g., DOS, Windows, Linux etc.), where malicious programs could inject code by finding out about another program's location in RAM and replacing its instructions with some other ones.

But, more importantly, modern architectures are not strictly following either of those models, they're more like a best-of of both worlds. Modern CPUs are internally using a Harvard-like setup, where program and data are managed separately to speed up things, they have their own caches for both, own MMUs, etc. Externally, the data is on a shared medium, the RAM, which is von-Neumann-like.

(Sorry, couldn't help myself, had to write some words about this, call it "background" for my next argument.)

What @antony-jr talked about (the "is the program fully loaded into RAM so modifications on disk won't hurt" or something like that) can't be explained with von Neumann or the von Neumann architecture, because that's an entirely different issue on a much higher level. It's an operating system responsibility to decide what to load into RAM, and the OS is a few levels above the hardware architecture. And @atdotde explained how modern OSes load programs.

That's also why you occasionally get some "access denied" messages trying to write to files of running programs, as @dirkhh explained, the operating system protects those files.

What can be done however, is "deleting" the original file (which doesn't remove data from the storage medium, it only removes the file system entry, (well, actually, the OS will wait until all file handles have been closed before taking any actions on disk, the files will only become "invisible" sort of)) and create a new file with the same filename. That's how AppImageUpdate's self-update feature works - it deletes the original file, creates a new one with the same filename, and then executes that new file which "looks" like it was updated but technically it's an entirely new file.

I guess @antony-jr was just trying to advertise his project.

**Please note that @antony-jr's AppImageUpdater[Bridge] are not official AppImage projects.**

probonopd, 2019-02-26:
[AppImageUpdaterBridge has a Qt plugin](https://github.com/antony-jr/AppImageUpdaterBridge/issues/7) which can be added to the Subsurface AppImage and would result in this:

[image]

There is a Subsurface AppImage that has the AppImageUpdaterBridge Qt plugin bundled for testing purposes:

https://github.com/antony-jr/AppImageUpdaterBridgePlugin/releases/download/continuous/Subsurface-x86_64.AppImage

It still has some quirks (described in https://github.com/antony-jr/AppImageUpdaterBridge/issues/7) but maybe others want to test and give feedback already. Thanks @antony-jr for making this. :+1:

probonopd, 2019-07-02:
Any opinion on this? This has been refined:
https://github.com/TheFutureShell/QtUpdateBridges/issues/2#issuecomment-499597102

glance-, 2019-11-10:
I think its a cool feature, and I'm all for it, as long as the user interaction for the update is roughly the same as for our continious and release builds, to not confuse the users.

mikeller, 2026-10-18:
Hi @probonopd,

Following up on this and the related #1113: given it's been a few years since the working prototype (the Qt plugin demonstrating binary-delta self-updates via libappimageupdate) was built and shared here, is that approach still current/viable against today's AppImage ecosystem? We'd treat integrating it as mostly packaging/build-pipeline work rather than fresh design if so, let us know if you're still interested in this landing in Subsurface's own AppImage build.

probonopd, 2026-10-18:
Hi @mikeller, yes, that is still the suggsted approach. Please let me know if you need any support. Thanks!